OpenCPN/OpenCPN #3247: AIS_sart-mob-epirb-distress-sarAircraft.txt  -Pass

rgleason, 2023-05-15:
**Describe the bug**
AIS_sart-mob-epirb-distress-sarAircraft.txt  Testing
This appears to pass.

**To Reproduce**
Steps to reproduce the behavior:
1. Download  
[AIS_sart-mob-epirb-distress-sarAircraft.txt](https://github.com/OpenCPN/OpenCPN/files/11473314/AIS_sart-mob-epirb-distress-sarAircraft.txt)

2. Use VDR to play
3. These three EPIRB, SART , and MOB all appear at the very top of the AIS Target List. This is the correct thing to have happen.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

nohal, 2023-05-15:
What problem is this issue trying to describe?

rgleason, 2023-05-15:
None, just shows that it is working. Closing

closed 2023-05-15 by rgleason