OpenCPN/OpenCPN #3248: AIS-SART-MOB-Firebar-Real-9m.txt  -Pass

rgleason, 2023-05-15:
**Describe the bug**
This file has SART which appear properly in the AIS List, at the top

**To Reproduce**
Steps to reproduce the behavior:
1. Download  
[AIS-SART-MOB-Firebar-Real-9m.txt](https://github.com/OpenCPN/OpenCPN/files/11473398/AIS-SART-MOB-Firebar-Real-9m.txt)

2. Run with VDR

**Screenshots**
[image]

[image]

[image]

[image]

nohal, 2023-05-15:
What problem is this issue trying to describe?

rgleason, 2023-05-15:
Shows that it works. Closing