OpenCPN/OpenCPN #3250: serial input no longer working

RooieDirk, 2023-05-15:
Current master, the serial usb input is no longer working on my linux box.
a git-bisect found

dirk@dirk-MINI:~/Development/OpenCPN$ git bisect bad
d159f1bdb8dff4cd05a4613b29b0afde11ad4c09 is the first bad commit
commit d159f1bdb8dff4cd05a4613b29b0afde11ad4c09
Author: dsr <bdbcat@yahoo.com>
Date:   Fri May 12 11:15:21 2023 -0400

    Android, disable comm serial thread model, WIP, #3214

 src/comm_drv_n0183_serial.cpp | 71 ++++++++++++++++++++++---------------------
 src/comm_drv_n2k_serial.cpp   |  7 ++++-
 2 files changed, 43 insertions(+), 35 deletions(-)

Not sure what is going wrong.
Dirk

bdbcat, 2023-05-15:
Corrected by [6ac1d14](https://github.com/OpenCPN/OpenCPN/commit/6ac1d14fe28e4b64dc906554e45351e08546e566)

RooieDirk, 2023-05-15:
Fixed ;)

closed 2023-05-15 by RooieDirk