OpenCPN/OpenCPN #3251: Display the warm AIS status at the start

Ptizef, 2023-05-15:
The 5.8 version introduced the possibility to easily stop AIS warming, This is very useful, but security could be at risk, because after have shut down the warming, closing O then reopen it (for example the next day for a new navigation), you are not remembered that the warming is still off, and it's easy to forget ...
The most easy way should be to display the warming status at start.
It would be better, but still the users who don't show the status bar have no AIS status information.
So it would be great to have somewhere an icon showing the status (maybe in the GPS/Compass window, or the muibar?)
JP

Hakansv, 2023-05-28:
JP..
Since I was partly involved in this I may claim a philosophy for the present function.
The reason for not showing this info at start up is that most users don't normally have the CPA alert enabled or ever use it. Than the status info would be kind of unnecessary and maybe disturbing info.
I think those with a AIS transponder use the CPA alert only on open water with few ships around or while crossing a separation zone. 
The alert will be unusable close to or inside harbors. When entering areas where the general alert may be useful it's easy to toggle the alert and have actual info available. (On devices using the status bar)
I also think we can't have warnings for everything not activated. That's kind of a false security instead of being all time aware of the surroundings and actual status of a navigational aid.

Ptizef, 2023-05-31:
Hakan,
I am not sure  "most users don't normally have the CPA alarm enabled"
The logic for me is certainly more "use by default the CPA alarm and disable it when it becomes unusable in very busy zones" but off course it's only my point of view.
But if we follow your philosophy, why not keep the original user choice and don't save the 'temporary' state. In this case, everybody would be satisfied. Those who don't use the CPA alarm could activate it when necessary, but at each start retrieve their primary choice and the contrary for the others.
At last, I don't agree with your last statement that warning would be a kind of false security. It should be (not for everything off course) a good help to know at a glance the situation of the critical parameters, avoiding having to go inside the parameters' window which is a pain in rough sea. It could also prevent an accidental oversight.
Anyway, this function is welcome and was awaited. Thanks
JP

Hakansv, 2023-05-31:
JP..
I think there's a misunderstanding. Probably due to my not so good English..
Status for the "Show AIS warnings" is saved to config when O is shut down. (That function I never touched.)
What's not viewed at O start up is the info text in the status bar. 
The status can also be checked in the menu bar. (Use Alt key to show).
The AIS warning info text in the status bar is shown only when you toggle the function first time after O start. Then it's relevant to show since you're using it. So if you press "W" first time after O start and the warning was enabled it will toggle to "Off" and be shown in the status bar.

Ptizef, 2023-06-01:
HaKan 
Right English's difficult for me too.
Nevertheless, I'll try to simplify.
My only point is that the user who generally use the warm is not always informed of the state, whereas it could be deactivated. Maybe it's not so important. For me, it is, but I can be wrong.
JP

Hakansv, 2023-06-22:
JP..
Out sailing now and then -  delays..
One thought: 
For system using the "Status bar", e.g. selected in "User.." or not Android.., we could possibly show the CPA warning status ("CPA alarm ON/OFF") from OCPN start when a AIS target is first received? Then the CPA status wouldn't "disturb" those not using AIS.
If possible to develop would that be reasonable compromise?

Ptizef, 2023-06-22:
Hakan

I hope this will not disturb your sailing ...
Yes, it's a good idea. If possible, it would satisfy everyone
many thanks and good sail
JP

Hakansv, 2024-02-15:
@Ptizef Time to close this as completed with #3345

bc38, 2024-02-15:
Graphic indication of alarms in the form of pictograms would be more informative. There is a standard for alarms

RooieDirk, 2024-02-18:
Not sure if this is a good idea. There are much more settings that could be changed without remembering the next day.
The tool to start with a predefined setting are  the 'templates'.
I think a better way is add a startup option to use a specific template. Something like 'OpenCPN.exe --template default'

Ptizef, 2024-03-03:
Fixed

closed 2024-03-03 by Ptizef