OpenCPN/OpenCPN #3267: Linux, plugins are potentially loaded from both /usr/lib/opencpn and /usr/local/lib/opencpn

bdbcat, 2023-05-18:
Should use the correct lib directory based on install {prefix}

leamas, 2023-05-18:
Indeed. The current scheme represents a transition  state from the use of system-wide plugin installations to personal installations in  $HOME and should be retired.

 A linux installation in _some-path/bin_ should load plugins from _some-path/lib/opencpn_ besides the these days default path _$HOME/.local/lib/opencpn_.  

The primary use case for _some-path/lib/opencpn_ is the system plugins (grib, wmm,, etc.). A related question is if we should load anything but these from _some-path/lib/opencpn_ . If we only load system plugins from this path  it would make for a much simpler system.  But then again, it would perhaps require that we also resolved the discussion in #3223 so there is some sort of opening for system-wide installations.

My gut feeling is that there are some interesting paths for system-wide plugins including properly packaging some of them.  So, there should some path available for this to keep doors open.

Practically, this would be about modifying `PluginPaths::Libdirs()`. As of now, it's basically a hardcoded list configured bt CMAKE around line 520. However, it would be complicated for cmake to figure out the installation path due to the use of $DESTDIR in linux installations. Probably simpler to include something like @install-path@ in the load path used and substitute it in runtime with the installation directory which we can find out.

bdbcat, 2023-05-18:
A more pragmatic approach for now, to fix this bug in O582, requires a simple change to plugin_paths.cpp.

```
diff --git a/src/plugin_paths.cpp b/src/plugin_paths.cpp
index fe8196bd9..73a83a28a 100644
--- a/src/plugin_paths.cpp
+++ b/src/plugin_paths.cpp
@@ -120,7 +120,7 @@ void PluginPaths::initLinuxPaths() {
 
   const string platform_dir = g_BasePlatform->GetPluginDir().ToStdString();
   const char* const envdirs = getenv("OPENCPN_PLUGIN_DIRS");
-  string dirlist = envdirs ? envdirs : OCPN_LINUX_LOAD_PATH;
+  string dirlist = envdirs ? envdirs : "~/.local/lib";
   m_libdirs = split(dirlist, ':');
   for (auto& dir : m_libdirs) {
     dir += "/opencpn";
d

```
Tested, works.
Thoughts?

nohal, 2023-05-18:
Wouldn't it be better to handle it in https://github.com/OpenCPN/OpenCPN/blob/v5.8.x/CMakeLists.txt#L513-L525 ?
Perhaps based on the installation prefix remove either `/usr/lib` or `/usr/local/lib`, which would disallow the duplication mentioned here and allowed both packaged and local builds to work pretty much as expected?

bdbcat, 2023-05-18:
In my testing,  g_BasePlatform->GetPluginDir() uses wxWidgets ideas of runtime dirs, and produces the correct answer for both packaged and local builds.  So no mucking with CMakeLists, which I prefer.
In both cases, we preserve the environment override for special cases.

leamas, 2023-05-18:
> Perhaps based on the installation prefix remove either /usr/lib or /usr/local/lib,

This is problematic if a developer for one reason or another installs in some other location, for example /opt. One should still expect that the plugins are loaded for the right place.

More: It's customary to install in temporary build locations and then move the complete installation someplace else.  In the end, I think using the installation path determined at runtime is the right way to go.

leamas, 2023-05-18:
> In my testing, g_BasePlatform->GetPluginDir() uses wxWidgets ideas of runtime dirs, 

We have had problems with this recently, trying to get the path to the executable. It's now solved using native linux calls which are both simple and stable.

nohal, 2023-05-18:
I have no problem with that, but we should then get rid of everything related to `OCPN_LINUX_LOAD_PATH` everywhere.

bdbcat, 2023-05-18:
get rid of everything related to OCPN_LINUX_LOAD_PATH..
Agree, will do.

bdbcat, 2023-05-18:
It's now solved using native linux calls which are both simple and stable.
How/where?  Not using wxWidgets methods?

leamas, 2023-05-19:
@bdbcat : you did that: 113c26cf74 ;)

leamas, 2023-05-19:
> get rid of everything related to OCPN_LINUX_LOAD_PATH.

Indeed. As I said, it was a transition state when we had to support the existing plugns typically in /usr/local while introducing the new path in $HOME. It's time to retire it.

Should end up in exactly two paths:
  - _@install-path@/../lib/opencpn_, where @install-path@ is the path to the opencpn executable
  - _$HOME/.local/lib/opencpn_
  
There is also the portable support, with it's own set of paths. IMHO, it's important to keep the OPENCPN_PLUGIN_DIRS environment var support, partly for development reasons.

nohal, 2023-05-19:
I'm not sure I like that `opencpn` is added to the path as a developer

eg. I want to load the WMM plugin from my build dir and do `OPENCPN_PLUGIN_DIRS="/home/nohal/source/OpenCPN/build/plugins/wmm_pi" ./opencpn`, but the plugin is not found as we do

```
16:45:06.066 MESSAGE plugin_loader.cpp:251 Loading plugins from dir: /home/nohal/source/OpenCPN/build/plugins/wmm_pi/opencpn
16:45:06.066 MESSAGE plugin_loader.cpp:444 PluginLoader searching for PlugIns in location /home/nohal/source/OpenCPN/build/plugins/wmm_pi/opencpn
16:45:06.066 MESSAGE plugin_loader.cpp:458    Directory /home/nohal/source/OpenCPN/build/plugins/wmm_pi/opencpn does not exist.
```

Which makes it effectively impossible (without some symlink trickery)

Same problem will be with all the 3rd party plugins unless the build dir is called `opencpn`, then we could point one level above it...

Do we really need/want to do https://github.com/OpenCPN/OpenCPN/blob/master/src/plugin_paths.cpp#L133 in case the source of the paths is the environment variable? And if we do, what about extending the list of the paths to include **also** the original value provided?

bdbcat, 2023-05-19:
I'm not sure what to do about the first half of your comment.  It is a problem, I agree.
In my workflow, when I need to debug a base plugin, I do "$sudo make install", which puts the base plugins in /usr/local/lib.  After the patch discussed here, I set the IDE environment variable OPENCPN_PLUGIN_DIRS to suit.   And all works well.
If you have another idea, I'm interested.

As for the second half, I think of the environment variable as overriding the hard-coded logic.  I could be persuaded that making the environment variable "supplement" the runtime path detection is OK, too.

nohal, 2023-05-19:
I would probably consider the environment variable as the source of real truth provided by the user who does know what he wants and why, so adding anything to it seems wrong to me.
Is there any reason why, when setting it, it really needs to `/usr/local/lib` and not `/usr/local/lib/opencpn`? I can't think of any, but sure can be missing some reason behind it.

nohal, 2023-05-19:
Yes, this is exactly what I had on mind. AFAICT this makes absolutely everybody happy, even the devs who will be able to debug plugins directly from the build directory

bdbcat, 2023-05-19:
OK, coming right up.

nohal, 2023-05-19:
Yes, now it does exactly what I would expect it to do

leamas, 2023-05-20:
This seems basically fine to me.  Things to consider:
1. pluginLoader.cpp:  setLoadPath(). At a quick glance, all I have time for, it looks OK. But it could use some more review and thoughts, for sure.
2. We need to update documentation, both the opencpn.1 manpage and the manual. You could consider this assigned to me.

leamas, 2023-05-20:
Reopening until documentation is fixed

PR #3287 (2023-05-24): opencpn.1: Update, notably new plugin load paths. (#3267)
Update manpage with new plugin load paths.

Closes: #3267
commit 9e0f44b0 (2023-05-24): opencpn.1: Update, notably new plugin load paths.

Closes: #3267
closed 2023-05-24 by bdbcat