OpenCPN/OpenCPN #3271: OpenCPN build CMAKE error in Windows 10 environment

BaekSophie, 2023-05-19:
I want to build and use OpenCPN in win10 environment. Visudal studio 2017 was used and the process suggested in the manual was performed.
Unfortunately, the following CMAKE error occurs while building OpenCPN. how can i solve it?

-- Found wxWidgets_VERSION_STRING: 3.1.2
--
-- *** Staging to build opencpn  ***
-- *** Build type:
-- *** Will install to C:/Program Files (x86)/OpenCPN  ***
-- GarminHost is enabled.
-- PackageName debug  opencpn C:/OpenCPN
-- Found Gettext: C:/Program Files (x86)/Poedit/GettextTools/bin/msgmerge.exe (found version "0.21.1")
-- Using downloaded rapidjson at C:/OpenCPN/build/_deps/rapidjson_src-src/include
-- Applying patch 0010-rapidjson-1.1.0-c++20.patch
-- Applying patch 0011-do_not_include_gtest_src_dir.patch
-- Applying patch 0012-Fix-Wclass-memaccess-warnings-errors.patch
-- Applying patch 0013-gcc7.patch
-- Could NOT find LIBUDEV (missing: UDEV_LIBRARY UDEV_INCLUDE_DIR)
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Could NOT find LIBUSB (missing: LIBUSB_INCLUDE_DIR LIBUSB_LIBRARIES)
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Could NOT find PANGO (missing: PANGO_INCLUDE_DIR PANGO_LIBRARIES)
-- Found OpenGL....
--     GL Lib:
--     GL Include:
-- Crash reporting enabled
-- Using Windows CrashRpt
-- Could NOT find Sqlite (missing: SQLITE_INCLUDE_DIR SQLITE_LIBRARIES)
-- Using bundled sqlite3 library.
-- Could NOT find LibLZMA (missing: LIBLZMA_LIBRARY LIBLZMA_INCLUDE_DIR LIBLZMA_HAS_AUTO_DECODER LIBLZMA_HAS_EASY_ENCODER LIBLZMA_HAS_LZMA_PRESET)
-- lzma library not found
CMake Error at cmake/Curl.cmake:23 (message):
  Cannot find bundled windows files.
Call Stack (most recent call first):
  CMakeLists.txt:1514 (include)

**Screenshots**
[image]

nohal, 2024-01-04:
You are missing dependencies. The complete instructions are available at https://opencpn-manuals.github.io/main/opencpn-dev/windows.html
The batch scripts mentioned there for either of the workflows will take care of populating the dependencies for you.

closed 2024-01-04 by nohal

zhangyu8730, 2024-05-06:
Hey, may I ask you how to solve this problem? I have the same problem now. Thanks so much!!!

leamas, 2024-05-06:
You are using outdated docs, see correct link in @nohal 's reply above.

You are also using an outdated build environment, building these days is documented for Visual Studio 2019.

@rgleason The old, unsupported development parts in dokuwiki is causing problems, not the first time. What is the reason to not remove it?

rgleason, 2024-05-06:
No reason. Will remove it however It is clearly labeled "OLD 2021-Dev Manual"  and should not be a problem.
I think the issue is some link that refers to the page he is using.

leamas, 2024-05-06:
No, that page has no such label. The same is true for each and every page in the old manual except the index page.

And as you see, users find these pages without using the index page using google. This means you need to add PROMINENT notes to each and every old page.

rgleason, 2024-05-06:
NO, I am not going to do that Alec.  See https://github.com/OpenCPN/OpenCPN/discussions/3890#discussioncomment-9328468