OpenCPN/OpenCPN #3273: NMEA2000 output by NGT-1 not working from plugin

douwefokkema, 2023-05-20:
I am busy converting the AutotrackRaymarine plugin for the integrated N2k support in OpenCPN 5.9, testing with commit 05e60639 and a NGT-1.

On Windows: Reading the the N2k messages was no problem, using the examples from dashboard_pi it was all straightforward. Remark, the received message is called payload, but is in fact the complete N2k message, which is fine.

Writing the messages seems OK, get the handle, write with WriteCommDriverN2k, returns no error, but nothing happens on the serial interface as seen by the sniffer. I looked into the code of O, the Windows code that should read the queue is not implemented. So I finished that code, output on the serial interface, but no PGN's on the bus. The NGT-1 was getting the wrong sequence of bytes. To function, NGT_1 should receive:

prio, pgn(3 bytes), dest, length of payload, payload

When cutting off 4 bytes at the beginning and 2 bytes at the end of the message output worked fine. So this is my code in coo_drv_serial_n2k.cpp (including some logging), I will not make a PR as this is merely a hack, though working fine:

//      Check for any pending output message

    bool b_qdata = !out_que.empty();

    while (b_qdata) {
      //  Take a copy of message
     
      //const unsigned char * xx = out_que.front().data();

      auto qmsg = out_que.front();
      unsigned char msg[MAX_OUT_QUEUE_MESSAGE_LENGTH];
      size_t size = out_que.front().size();
      for (int i = 0; i < size; i++) {
        msg[i] = out_que.front()[i];
      }
      out_que.pop();

      //free(xx);       ??

      wxLogMessage(wxT("$$$x send message size=%i"), size);

      wxLogMessage(wxT("%0x, %0x, %0x, %0x, %0x, %0x, %0x, %0x, %0x, %0x, %0x, %0x, %0x, %0x, %0x, %0x, %0x, %0x, %0x, %0x, %0x, %0x, %0x, %0x, %0x end of 25 \n"),
        msg[0], msg[1], msg[2], msg[3], msg[4], msg[5], msg[6], msg[7], msg[8], msg[9], msg[10], msg[11], msg[12], msg[13], msg[14], msg[15],
        msg[16], msg[17], msg[18], msg[19], msg[20], msg[21], msg[22], msg[23], msg[24]);

      // For NGT-1 send: prio, pgn(3 bytes), dest, length of payload, payload

      if (static_cast<size_t>(-1) == WriteComPortPhysical(&msg[4], size-6) &&
          10 < retries++) {
        wxLogMessage(wxT("$$$x send retry end size=%i"), size);
        // We failed to write the port 10 times, let's close the port so that
        // the reconnection logic kicks in and tries to fix our connection.
        retries = 0;
        CloseComPortPhysical();
      }

      b_qdata = !out_que.empty();
    }  // while b_qdata

Under Linux nothing is received. The WriteComPortPhysical is called, with a message that should be shorted 4 at the beginning and 3 at the end. But even then no PGN's are coming out.

bdbcat, 2023-05-20:
Thanks for the detail, and the hack .
For sure this API needed exercise for a real purpose.  Proves the maxim:  Testing by the original coder is not always effective.
I'm on it.

douwefokkema, 2023-05-20:
The code snippet is not really OK, the loop at the end sends out each message 10 times.

bdbcat, 2023-05-21:
douwe...
Now I am getting conflicting information on the format of messages sent to NGT-1, for transmission onto the bus.

From Timo, I understand that the message must be "framed" by 4 bytes at the front, and two at the end.
This framing is accomplished by the method BufferToActisenseFormat().  You will see there the initial 4 bytes, and trailing two bytes.
From your description, it seems that the "framing" is not necessary.  One should start immediately with the PGN...

Now this could be a difference between device control messages digested by the NGT-1, as discussed with Timo, versus actual messages intended for the bus, as in your use case.
Do you have any insight on this?
I will continue to read what little documentation there is on NGT-1.

bdbcat, 2023-05-21:
You might like to take a look at:
https://www.yachtd.com/downloads/ydnu02.pdf
Appendix "F".

bdbcat, 2023-05-21:
TBH, I cannot see how sending "unframed" messages to the NGT-1 could actually work.  The gateway needs to gather all bytes of the message, parse it into CAN frames, and negotiate it onto the bus.  Without knowing when a message starts and ends, (e.g.by framing the message)  how could this be done?
But I believe you, so I'm temporarily stumped.

bdbcat, 2023-05-21:
Without seeing the debug output of your patch, is it possible that a logic error is causing framing of the message twice?

douwefokkema, 2023-05-21:
I do not know of any special operating mode  in my NGT-1. I have used it on the AutotrackRaymarine plugin, and use it for bus monitoring with Actisense NMEA reader.

I never thought about it before, but it is strange that these messages are not framed, one would indeed expect a start of message and end of message indicators. I have copied my way of handling the NGT-1 from Canboat. In my original AutotrackRaymarine plugin, in serialport.cpp, in SerialPort::parseAndWriteIn exactly the same happens. From the Canboat format a message is made to write to the serial port. No framing. This method is copied straight from the Canboat stuff.

My debug output:

5:15:02.026 MESSAGE comm_drv_n2k_serial.cpp:1086 $$$x send message size=21
15:15:02.026 MESSAGE comm_drv_n2k_serial.cpp:1088 10, 2, 94, e, 6, 12, f1, 1, ff, 8, 1, a9, 5d, ff, 7f, ff, 7f, fd, 4d, 10, 3, 0, 0, 0, 0 end of 25

The total message is 21 length, so the real message is:

10, 2, 94, e, 6, 12, f1, 1, ff, 8, 1, a9, 5d, ff, 7f, ff, 7f, fd, 4d, 10, 3

Take of 4 at the beginning and 3 at the end (earlier I wrote 2, should be 3) and you get:

6, 12, f1, 1, ff, 8, 1, a9, 5d, ff, 7f, ff, 7f, fd

Which is the message the NGT-1 needs to send a PGN 127250 heading. Prio = 6, PGN 0x1f112, destination 0xff, payload length=8. And it actually does send out this PGN as seen by my NGW-1 on the same network (I don't have a second NGT-1, would have been better).

Test were done with https://github.com/douwefokkema/AutoTrackRaymarine_pi branch n2k, which is a unfinished converted version. Only thing it does is receiving PGN127250 and display the heading in the plugin window, and transmit a PGN127250 (fixed heading) when hitting the Standby button.

I hope this helps. I will be happy to test any new version from your side. Off to the boat now, so this is all for today.

bdbcat, 2023-05-22:
Douwe...
Some progress here.
1.  I can replicate your results on Anctisense NGT-1.  Unframed messages are sent  directly onto the bus.
2.  Despite the documentation, the Yacht Devices YDNU-02 also sends unframed messages directly to the bus.
3. I can make both devices work with framed messages, but there is a small difference in the format required.  Actisense want to see the N2K payload verbatim.  Yacht devices wants the message payload to omit the "source" and "time" fields,  since presumably they are maintained by the gateway itself.
4. I can detect which of the two supported devices is present, so we could use framed message TX if we want.  Modify the message content based on the type of device in use.

In absence of reliable documentation of both devices, we will need to simple decide whether to use framed messages, or not, for OCPN TX.  There are competing precedence.
1. canBoat is using un-framed, as you mention. Popular package.
2.   Timo's N2K support library uses framed messages.  Also popular package.

Thougthts on this?

bdbcat, 2023-05-22:
I'm beginning to prefer framed mode.  After some more fiddling with both supported devices, looking for common ground, etc.  Framing seems to be more robust from a setup and timing sense.  Especially seems so for messages requiring fast packet assembly.  This makes sense, since the raw messages need to be split across multiple packets.
Still fiddling, I expect to push a  patch tonight.

bdbcat, 2023-05-23:
Douwe...
I have pushed a patch to attempt to correct NGT-1 output operations.
747c78ebef6571e0b5188ed14c25e868daac7a8a
Tested on Win10 and linux.
Good luck.

douwefokkema, 2023-05-23:
Tested on Win10, all fine, both receiving and transmitting N2k messages. But this was only a limited test on a minimal network with only PGN127250. Beginning of next week I will be on the boat in real life conditions. But I see no reason it would not work there.

Also on Linux (Ubuntu 22) all fine, same tests as above.

On both platforms there is no Tx output for N2k in the NMEA debug window for N2k. Input is shown in the debug window.

Issue on Linux (OK on Win10):  When closing the Options / Connections window with the OK button while the NMEA debug window is open causes a crash. No crash when closing with Cancel.
gdb output: Thread 1 "opencpn" received signal SIGSEGV, Segmentation fault.
0x0000555555df9474 in s57chart::UpdateLUPs(s57chart*) ()
[gdb.txt](https://github.com/OpenCPN/OpenCPN/files/11544355/gdb.txt)

bdbcat, 2023-05-23:
Good news on N2K output.
"When closing the Options / Connections window with the OK button while the NMEA debug window is open causes a crash.'
Not seeing this crash.  Anyone else?  Any more info?

douwefokkema, 2023-05-23:
No crash when my plugin is removed, sorry for that.

douwefokkema, 2023-05-31:
So the change for the initialization is in SetTXPGN(), there I changed:
// Activate message
-  unsigned char request_activate[] = { 0x4B };
+  unsigned char request_activate[] = { 0x11, 0x02, 0 };

bdbcat, 2023-06-02:
douwe...
I made the change for honoring the programmed destination address.
I will shortly update for the message priority, as well.
But I did not change the "request_activate[]" string.  TBH, I do not understand this change, and where it comes from.  Did you do some reverse engineering to arrive at this string change?
The original code comes from some RE work that Timo did.  I followed his suggestions, but had no good way to test.

douwefokkema, 2023-06-02:
Dave,
This is what I did. Windows computer with O master and the newly adapted version of my RaymarineAutoTrack_pi. Connected though a NGT-1 to my boat N2k network with an EV-1 autopilot on it. I can only monitor the N2k network based on the response of the autopilot as I do not have a second NGT-1 with NMEAreader. So I issue an "Auto" command through my pi, nothing happens. I log the full command at the interface in O. Then I go back to the original version of the pi (direct control of the NGT-1). I issue the Auto command, pilot reacts by switching to Auto, and I also log the command message at the interface. I compare the messages logged. Different only in destination and prio. I change destination to 0xff in my pi. Autopilot no longer reacts. So I patch O to issue exactly the same sequence of bytes at the interface as the functioning command from my old pi. No reaction from the pilot. Mystery, what else could be different. The initialization of the NGT-1 is the only thing left. The only initialization command of the device used in Canboat and (copied) in my original pi is { 0x11, 0x02, 0 }. So I issue that command from O, just by replacing the last init string in O by { 0x11, 0x02, 0 }. Bingo, my pi functions! I do not think the other init strings in O are required, but I did not try without.

Douwe

bdbcat, 2023-06-03:
Douwe,
Thanks for the explanation.
I have implemented all.  In master now.
Tested by using Actisense NGT-1 and ActisenseReader on Win PC, and sending test messages via YachtDevices YDNU-02 from linux.
Priority field seems OK now.  But I may be having some trouble with destination address.  It seems OK in the transmitted byte stream, but ActisenseReader always reports destination is 0xff.
Will you post a simple message byte stream (wrapped) that you know works for AutoTrack_pi, as comparison?

bdbcat, 2023-06-03:
Douwe,
For reference, here is the wrapped byte stream sent for initialization:
'
10 02 A1 01 42 1C 10 03 
'

Dave

douwefokkema, 2023-06-03:
Dave,
The above byte stream including the wrapping is exactly the same as the one that I was sending in the successful test.
I can not test your new master now, I am home again and the N2k stuff is at the boat, two hours drive from here.
Douwe

douwefokkema, 2023-06-03:
But here is a working PGN126208 (0x1ed00) (Raymarine prop., set Auto) recorded at the interface:
10, 2, 94, 17, 3, 0, ed, 1, cc, 11, 1, 63, ff, 0, f8, 4, 1, 3b, 7, 3, 4, 4, 40, 0, 5, ff, ff, 97, 10, 3
Notice the payload length 0x11 after the destination 0xcc.

bdbcat, 2023-06-03:
Comparing to a simple test string from me:
10 02 94 0E 06 11 F0 01 85 08 D0 07 00 FF FF FF FF FF F7 10 03 

Priority:6
PGN:  0x1f011
Destination: 0x85
Length: 8

Structure looks identical.

bdbcat, 2023-06-04:
Douwe...
I have a little plugin test framework for N2K TX test. 
Just for an independent functional test, I executed the following, hardcoded from your sample string, from a plugin context
All seems well.  My bus monitor reports good results.

```
      // Raymarine autopilot PGN126208

  uint8_t ap_payload[] = {
     0x1, 0x63, 0xff, 0x0,
     0xf8, 0x4, 0x1, 0x3b,
     0x7, 0x3, 0x4, 0x4,
     0x40, 0x0, 0x5, 0xff,
     0xff };

  std::vector<uint8_t> vpayload;
  for (size_t i=0; i < sizeof(ap_payload); i++)
    vpayload.push_back(ap_payload[i]);

  auto bpayload = std::make_shared<std::vector<uint8_t>>(vpayload);
  CommDriverResult bresult126208 = WriteCommDriverN2K( n2k_driver_handle, 126208, 133, 3, bpayload);

```

Just FYI.
Thanks

douwefokkema, 2023-06-04:
Dave,
Sounds good. I will test on board at the next occasion, sometime middle of the month I expect. But I have no doubt it will be OK now, as initialization and destination have been corrected, and the rest was fine already. Priority was also corrected, but was not critical.
Douwe

douwefokkema, 2023-06-20:
I tested the NMEA2000 input and output with the interim version of AutotrackRaymarine_pi (branch n2k) and O master, 70ef2009. Input and output to EV-1 autopilot works fine. I also tried with 5.8.4 released version, there I did not get a nmea2000 handle for output, only nmea0183 handles were found. I did not debug this as master is fine.

closed 2023-06-21 by douwefokkema

bdbcat, 2023-06-21:
Better bring this as a new issue?
"also tried with 5.8.4 released version, I did not get a nmea2000 handle for output, only nmea0183 handles were found. "