OpenCPN/OpenCPN #3282: Serial n0183 Position Watchdog is frequently timed out. >> Position lost for some seconds.

Hakansv, 2023-05-23:
**Describe the bug**
While navigation using serial n0183 where RMC is prioritized for position the position is lost for a second or two. The period is random but can be from every two minutes to every about 15 minutes.
I've seen it the last weeks but also refer to [this CF post](https://www.cruisersforum.com/forums/f134/o5-8-2-lose-contact-with-gps-276055.html#post3780326)

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a n0183 GNSS receiver where RMC is included.
2. Be sure to have RMC as prio 0 for position (and COG/SOG)
3. Study the ship symbol and if cruising using look ahead mode the loss of position and SOG is obvious.

**Expected behavior**
A stable position not disturbing "look ahead" mode.

**Desktop (please complete the following information if applicable):**
 - OS: Reported on Win10
 - Version O >5.8.2 and including present master code. 

**Additional context**
I've tried to add debug messages to present master code (until commit 747c78ebe) to follow what's happening. See below.
The message sources will point to where I put the log-prints.
The first five are example of normal behavior 
Then an error trap from m_NMEA0183.Parse().
The log message from com_bridge shows the "offending" RMC sentence. But what I can see it's a valid RMC nmea version >=2.3
So why the parser reports false on this I can't figure out

```
22:18:18.454 MESSAGE comm_bridge.cpp:686 ***RMC n0183 payload
22:18:18.454 MESSAGE comm_bridge.cpp:706 ***RMC update watchdog
22:18:19.442 MESSAGE comm_bridge.cpp:686 ***RMC n0183 payload
22:18:19.442 MESSAGE comm_bridge.cpp:706 ***RMC update watchdog
22:18:23.451 MESSAGE comm_bridge.cpp:686 ***RMC n0183 payload
22:18:23.451 MESSAGE comm_decoder.cpp:85 ***RMC >> if (!m_NMEA0183.Parse()
22:18:23.451 MESSAGE comm_bridge.cpp:695 ***RMC if (!m_decoder.DecodeRMC:str: $GNRMC201824.000,A,5813.57617,N,01145.74416,E,0.17,145.49,230523,,,A,V*05 temp_data: 0
22:18:29.060 MESSAGE comm_bridge.cpp:186    ***GPS Watchdog timeout at Lat:58.2263   Lon: 11.7624
```

bdbcat, 2023-05-24:
At first glance, it looks like the RMC message is ill formed.  The should be a "," after $GNRMC, like maybe this:
"$GPRMC,014933,A,3505.8412,N,07638.0506,W,00.0,293,240523,012.1,W*59\r\n"

Do you see this in the raw data stream on the Debug window?

Hakansv, 2023-05-24:
Hmm.. That missed comma must be a copy-past error or similar. Sorry.
Here are other examples:
```
17:31:34.737 MESSAGE comm_bridge.cpp:706 ***RMC update watchdog
17:31:36.007 MESSAGE comm_bridge.cpp:686 ***RMC n0183 payload
17:31:36.007 MESSAGE comm_bridge.cpp:695 ***RMC if (!m_decoder.DecodeRMC:str: $GNRMC,153136.000,A,5813.57385,N,01145.74012,E,0.05,109.77,230523,,,A,V*0   temp_data: 0
17:31:37.015 MESSAGE comm_bridge.cpp:686 ***RMC n0183 payload
17:31:37.015 MESSAGE comm_bridge.cpp:695 ***RMC if (!m_decoder.DecodeRMC:str: $GNRMC,153137.000,A,5813.57378,N,01145.74017,E0.07,109.77,230523,,,A,V*07  temp_data: 0
17:31:38.028 MESSAGE comm_bridge.cpp:686 ***RMC n0183 payload
17:31:38.028 MESSAGE comm_bridge.cpp:695 ***RMC if (!m_decoder.DecodeRMC:str: $GNRMC,153138.000,A,5813.57374,N,0145.74024,E,0.09,109.77,230523,,,A,V*0A  temp_data: 0
17:31:39.031 MESSAGE comm_bridge.cpp:686 ***RMC n0183 payload
17:31:39.031 MESSAGE comm_bridge.cpp:695 ***RMC if (!m_decoder.DecodeRMC:str: $GNRMC,153139.000,A,5813.57376,N,0145.74030,E,0.08,109.77,230523,,,A,V*0D  temp_data: 0
17:31:40.011 MESSAGE comm_bridge.cpp:686 ***RMC n0183 payload
17:31:40.012 MESSAGE comm_bridge.cpp:695 ***RMC if (!m_decoder.DecodeRMC:str: $GNRMC,153140.000,A,581357377,N,01145.74033,E,0.07,109.77,230523,,,A,V*0E  temp_data: 0
17:31:41.018 MESSAGE comm_bridge.cpp:686 ***RMC n0183 payload
17:31:41.018 MESSAGE comm_bridge.cpp:695 ***RMC if (!m_decoder.DecodeRMC:str: $GNRMC,153141.000,A,5813.57378,N,0145.74034,E,0.06,109.77,230523,,,A,V*06  temp_data: 0
17:31:42.019 MESSAGE comm_bridge.cpp:686 ***RMC n0183 payload
17:31:42.019 MESSAGE comm_bridge.cpp:695 ***RMC if (!m_decoder.DecodeRMC:str: $GNRMC,153142.000,A,5813.57379,N,0145.74031,E,0.04,109.77,230523,,,A,V*03  temp_data: 0
```
A "good", no error, example
`06:44:28 (COM10) $GNRMC,044428.000,A,581.57544,N,01145.73987,E,0.14,350.86,240523,,,A,V*00<0x0D><0x0A>`

There are three traps in "DecodeRMC". The second is the only one returning "false".
1. if (!m_NMEA0183.PreParse()) {
2. if (!m_NMEA0183.Parse()) {
3. if (m_NMEA0183.Rmc.IsDataValid == NTrue) {

Hakansv, 2023-05-24:
This is all strange... What I know nothing has changed in \libs\nmea0183\src\nmea0183.cpp?

Hakansv, 2023-05-24:
A VDR attached where some "good" and some "bad" RMC are collected.

[RMC_test.txt](https://github.com/OpenCPN/OpenCPN/files/11550925/RMC_test.txt)

Hakansv, 2023-05-25:
Dave..
For info I tested on a RPi4 and sees the same like: if (!m_decoder.DecodeRMC...  is hit.
It's unlikely sentences are sometimes created with a bad checksum from the device? So the assumption some characters are missed while reading the serial buffer is still valid?

closed 2023-05-26 by Hakansv