OpenCPN/OpenCPN #3285: Linux PPA package: use of non-optimal SystemCmdSound

leamas, 2023-05-24:
**Describe the bug**
The PPA packages for non-amd64 architectures are configured using SystemCmdSound. Since this means using an external binary to run sounds, it means less control and performance compared to using libraries linked to the main executable.

It also means that the amd64 packages are linked differently than armhf/arm64.

**Expected behavior**
The PPA package should use the default  configuration which uses the portaudio libraru

**Additional context**
The official package has been using portaudio as long as it has existed.  Furthermore, the build dependencies  already contains portaudio19-dev .

BTW: A debian architecture is never aarch64, it is always using arm64. aarch64 is what is used outside of Debian, like the NDC tools, Fedora, the ARM company, etc. Hence the "aarch64" branch in debian/rules is nto really useful.

The only change required would be:
```
iff --git a/debian/rules b/debian/rules
index e2bb60d09..e3bca713b 100755
--- a/debian/rules
+++ b/debian/rules
@@ -12,7 +12,6 @@ override_dh_auto_configure:
 ifeq ($(DEB_BUILD_ARCH_CPU), arm)
        dh_auto_configure -- \
            -DCMAKE_LIBRARY_PATH=$(DEB_HOST_MULTIARCH) \
-            -DOCPN_ENABLE_SYSTEM_CMD_SOUND:BOOL=ON     \
             -DCMAKE_INSTALL_LIBDIR=lib                 \
             -DCMAKE_BUILD_TYPE=RelWithDebInfo          \
             -DDEB_VERSION=$(DEB_VERSION)               \
@@ -21,7 +20,6 @@ ifeq ($(DEB_BUILD_ARCH_CPU), arm)
 else ifeq ($(DEB_BUILD_ARCH_CPU), aarch64)
        dh_auto_configure -- \
            -DCMAKE_LIBRARY_PATH=$(DEB_HOST_MULTIARCH) \
-            -DOCPN_ENABLE_SYSTEM_CMD_SOUND:BOOL=ON     \
             -DCMAKE_INSTALL_LIBDIR=lib                 \
             -DCMAKE_BUILD_TYPE=RelWithDebInfo          \
             -DDEB_VERSION=$(DEB_VERSION)               \
@@ -30,7 +28,6 @@ else ifeq ($(DEB_BUILD_ARCH_CPU), aarch64)
 else ifeq ($(DEB_BUILD_ARCH_CPU), arm64)
        dh_auto_configure -- \
            -DCMAKE_LIBRARY_PATH=$(DEB_HOST_MULTIARCH) \
-            -DOCPN_ENABLE_SYSTEM_CMD_SOUND:BOOL=ON     \
             -DCMAKE_INSTALL_LIBDIR=lib                 \
             -DCMAKE_BUILD_TYPE=RelWithDebInfo          \
             -DDEB_VERSION=$(DEB_VERSION)               \

```

leamas, 2024-04-11:
Note: 801f81d6398af, SystemCmdSound disabled.

bdbcat, 2024-04-11:
TODO:
All is not perfect here.
Quick test on linux with multiple overlapping sound requests produces "choppy" output.
To reproduce:  Add Hakan's signalk test data feed, enable CPA warnings and audio alert.  Sounds are produced, but not very nice.

leamas, 2024-04-11:
> Quick test on linux with multiple overlapping sound requests produces "choppy" output

This was not part of the design. Question is just how this should be handled. Obviously, some kind of mutex. But besides that, should parallel play requests be queued, dropped or what?

bdbcat, 2024-04-11:
"Calling code should wait for the callback before issuing next request"
Yes.  Should be easy to debug.

bdbcat, 2024-04-12:
I think we are done here.

leamas, 2024-04-12:
Great to hear! 

The bug is about the Debian package configuration,  we should be able to close it when we abandon the current Debian package build procedures. Marking as fixed for now

Hakansv, 2024-04-13:
Today's master code
One observation on a Rpi Bullseye. The same using any of available sound driver types, hdmi/default)
Using 2 bells sound.
Click any Test AIS sound button:
First click - nothing.
Second click - "Dong". (Only one)
Next clicks - "Click"-"Dong"- "Dong".  (The "Click" is a very short distinct sound)

Activated CPA Alert:
"Click"-"Dong"- "Dong".  "Click"-"Dong"- "Dong".  "Click"-"Dong"- "Dong"

leamas, 2024-04-13:
Removing Done label for now.

leamas, 2024-04-14:
@bdbcat  ^

bdbcat, 2024-04-15:
Tested on local rPI5.  Could not reproduce Hakan's trouble.
Hakan suspects local configuration problems.
So we declare "OK" for Beta test.  
Done now.

leamas, 2024-07-06:
Fixed in  5.9.2-beta2. Closing

closed 2024-07-06 by leamas