OpenCPN/OpenCPN #3295: Crash while zooming ENC charts

Hakansv, 2023-05-26:
**Describe the bug**
Using ENC charts O crash at zoom in to large scale.
No crash for o-charts, CM93 or RNC.

In debug brake in s52plib.cpp at:  (see also shot below)
```
TriPrim *p_tp = ppg->tri_prim_head;
    while (p_tp) {
      LLBBox box;
      if (!rzRules->obj->m_chart_context->chart) {  // This is a PlugIn Chart
```

**Desktop:**
 - OS: Tested on Win10
 - Version: O master after commit 1bd1a078b "Correct ENC Safety Depth contour detection and rendering. #3291"

**Screenshots**
Debug shot:
[image]

**Call stack**
```
>	opencpn.exe!s52plib::RenderToBufferFilledPolygon(_ObjRazRules * rzRules, S57Obj * obj, _S52color * c, render_canvas_parms * pb_spec, render_canvas_parms * pPatt_spec) Line 7854	C++
 	opencpn.exe!s52plib::RenderToBufferAC(_ObjRazRules * rzRules, _Rules * rules, render_canvas_parms * pb_spec) Line 9088	C++
 	opencpn.exe!s52plib::RenderAreaToDC(wxDC * pdcin, _ObjRazRules * rzRules, render_canvas_parms * pb_spec) Line 9156	C++
 	opencpn.exe!s57chart::DCRenderRect(wxMemoryDC & dcinput, const ViewPort & vp, wxRect * rect) Line 2367	C++
 	opencpn.exe!s57chart::DoRenderViewOnDC(wxMemoryDC & dc, const ViewPort & VPoint, RenderTypeEnum option, bool force_new_view) Line 2284	C++
 	opencpn.exe!s57chart::BuildThumbnail(const wxString & bmpname) Line 3113	C++
 	opencpn.exe!s57chart::PostInit(ChartInitFlag flags, ColorScheme cs) Line 2857	C++
 	opencpn.exe!s57chart::Init(const wxString & name, ChartInitFlag flags) Line 2631	C++
 	opencpn.exe!ChartDB::OpenChartUsingCache(int dbindex, ChartInitFlag init_flag) Line 1290	C++
 	opencpn.exe!ChartCanvas::DoZoomCanvas(double factor, bool can_zoom_to_cursor) Line 4502	C++

```

bdbcat, 2023-05-27:
This fault happens when building chart thumbnails, due to some refactoring in https://github.com/OpenCPN/OpenCPN/issues/3291
This is only done during the SENC conversion step, when visiting a new area.
However, chart preview thumbnails have not been used since long, probably back to OCPN v4.x.
Sp the simple solution is to now disable all thumbnail creation logic, by #if 0  preprocessor directive.

Hakansv, 2023-05-27:
Solved, thanks

closed 2023-05-27 by Hakansv