OpenCPN/OpenCPN #3307: Canbus  blocked in Flatpak version

verkerkbr, 2023-05-29:
Some time ago I have signalled the problem with the canbus in the flatpak version.

The solution was to unlock the flatpak can0 blockage with starting :

 flatpak override --user --allow=canbus org.opencpn.OpenCPN

Then it works and can0 appears.
 
 Some users run in the same problem. Is there allready a permanent solution ? In the Flatpak enviroment or in OpenCPN ?

leamas, 2023-05-29:
Good catch, and thanks for investigating!

This should be easy to fix in the Flatpak manifest to get a more permanent solution.

leamas, 2023-05-29:
Test build is ready. Could you please verify if this works without the `override --user --allow=canbus`  option? To test: 
```
flatpak install --user https://dl.flathub.org/build-repo/26542/org.opencpn.OpenCPN.flatpakref
```

verkerkbr, 2023-05-29:
Hi Leamas,

just tested your release of OpenCPN with only the Canable canbus connected. 

It seems to work. Now Can0 is appearing.

Tomorrow I will do a full test with a GPS NMEA2000 antenna connected.

Regards,

Bram

verkerkbr, 2023-05-30:
Hi leamas,

now tested the version with an GPS nmea2000 antenna connected. Works well on my Linux Mint Vera amd64, On the RPI4 Debian Bullseye, Debian Bookworm, Bullseye and Ubuntu Lunar.
The problem is solved. And all running on one version of  Flatpak OpenCPN !

Thanks for your quick action.

Regards,

Bram

closed 2023-05-30 by verkerkbr

rgleason, 2023-05-30:
@verkerkbr  Bram, It would be very helpful if can you please "Import Plugin" one of these [climatology_pi  Debian 12 Bookworm tarballs](https://www.cruisersforum.com/forums/f134/climatology-102281.html#post3782785) to see if it installs?  We need to check this build before continuing to build TP plugins.   Thanks.