OpenCPN/OpenCPN #3313: Preserve time stamps on navobj.xml backup files

antipole2, 2023-05-31:
**Is your feature request related to a problem? Please describe.**
When OCPN quits, it rotates the navobj.xml files so that navobj.xml.2 becomes navobj.xml.3 et seq.
On MacOS at least, the timestamp for these files becomes the time this was done.
So, knowing how far back you need to go to,  it is not possible to see which backup file to restore from.
Each attempt rotates again and it is easy to rotate out the one you want. 

**Describe the solution you'd like**
The timestamp should be noted before the file name is changed and the timestamp then restored.

**Describe alternatives you've considered**
If no changes have been made to the information in the navobj files, they should not be rotated.
It would then be possible to start up with a previous navobj file to see if it was the desired one and quit to try another without rotating them all.  This would be easier than relying on force quitting to avoid rotation. 

**Additional context**
Not sure what permissions are required to do this.  Likely OS dependent.

nohal, 2024-02-20:
Backups are now not rotated if there are no changes, if https://en.cppreference.com/w/cpp/filesystem/last_write_time is available on all our supported platforms, keeping the timestamp should be easily doable

antipole2, 2024-07-22:
This is milestoned for 5.10 but not yet implemented.

Since the much improved saving of data when untidy end to OCPN this not so pressing - but would be nice.

closed 2025-05-24 by antipole2