OpenCPN/OpenCPN #3316: OCPN crashing or silently closing with multiple nav inputs

antipole2, 2023-05-31:
**Describe the bug**
Until OCPN 5.6, my set-up received NMEA0183 data from my NMEA2000 network via an Actisense W2K converter that sent the data over WiFi.  So OCPN receives the data as NMEA0183.
Now under OCPN 5.6.2 I have reconnected an Actisense NGT-1-USB adaptor that sends the NMEA2000 data direct to a serial input.  This avoids relying on the WiFI router.
However, for resilience, I have the NMEA2000 serial input as priority 1 and keep the NMEA0183 via WiFi with priority 2.

However, I am getting OCPN crashes after an interval.  Very unfortunate when on passage.
Mostly, OCPN just quits without rotating the navobj.xml files and on restart says it did not close cleanly.
I have had one crash report - see below.

**To Reproduce**
Set up both inputs as described above and run for some hours.

**Expected behaviour**
As I understand it, OCPN will prioritise the data from the NMEA2000 but if that were to fail, it would fall back on the NMEA0183 input.

**Screenshots**
See crash report below

**Desktop (please complete the following information if applicable):**
 - OS MacOS Ventura 13.3.1
 - OCPN 5.8.2

**Additional context**
My hunch is that there is a memory leak in this set-up, based on the time delay before the crash.

For now, I am avoiding the NMEA2000 input.

[image]

CarCode, 2023-06-01:
It is known that new versions have always had bugs in functions that were still OK in the old version. But on 5.8.x it's particularly bad. In order to get stable versions of OpenCPN, I recommend splitting it into one version each for the main operating systems: Linux, Windows, macOS, Android and finally a new version for iOS.

antipole2, 2023-06-01:
I only had one crash report, so that may not be the real issue.
What I was seeing was a shut-down after an hour or so into a passage.  I restarted and got a shut-own again after a while.
I stopped using the NMEA2000/USB input (hence falling back on NMEA0183 over WiFi) and had no problems again for the rest of a 2 day passage.
Here in harbour, I tried using both again and got a shut-down.  But there was no crash report.
I think either work OK individually but not the combination.

bdbcat, 2023-06-02:
If convenient, I think it would be useful to test the NGT-1 N2K adapter by itself, without WiFi N0183 input enabled.
If the N2K input runs indefinitely without fail, then the problem seems more likely to be related to the combination.
If convenient....

antipole2, 2023-06-04:
Dave...  trying to nail this as best I can...
The crash report above showed problem with tracks on exit so wonder if link with track recording.
1) N2K only + track gave crash after 35 mins.
However
2) N2K + NMEA0183 and no track crashed after 5 hours
Crash report attached here suggests N2K problem wrapping up after exit?

Both crash reports seen seem to have problem _after_ exit - but why is it doing exit?

Without NMEA0183 I miss feed to iPad at wheel station (including AIS) and also to my passage log so reluctant to test for long periods.

[image]

bdbcat, 2023-06-07:
Tony...
How did you install the NGT-1 on MacOS?  Which FTDI driver did you use, if any?
Dave

bdbcat, 2023-06-08:
Tony...
I ran OCPN 5.8.2 on Macbook 11.7 (BigSur) catching live NMEA2000 GPS input via Actisense NGT-1/USB, with tracking enabled.
Ran fine for 4 hours in fixed location, with no apparent memory or performance problems.
Anything else significantly different in your setup?

antipole2, 2023-06-08:
> Tony...
> How did you install the NGT-1 on MacOS? Which FTDI driver did you use, if any?
> Dave

Email from Actsense last April when I had an issue with driver on their site.  I installed from the link he gives.

> Good Morning,
> 
> Thank you for your email and your interest in Actisense products,
> 
> Fortunately you are not the first person to bring this up, quite recently another customer contacted us with this issue so I updated my test laptop to the latest mac OS version at the time which was Ventura 13.2.1 and I found success using the FTDI drivers on their website which I will link below,
> 
> https://ftdichip.com/wp-content/uploads/2022/06/FTDIUSBSerialDextInstaller_1_5_0.dmg 
> 
> Best Regards,
> 
> Steve Drane
> Technical Support Engineer
> Actisense, 21 Harwell Road, Poole, BH17 0GE, UK

antipole2, 2023-06-08:
```
Tony@Laing ~ % kextstat | grep -v com.apple; systemextensionsctl list
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
Index Refs Address            Size       Wired      Name (Version) UUID <Linked Against>
1 extension(s)
--- com.apple.system_extension.driver_extension
enabled	active	teamID	bundleID (version)	name	[state]
*	*	658CPPCMJJ	com.ftdi.vcp.dext (1.1/010)	NullDriver	[activated enabled]
Tony@Laing ~ % 
```

antipole2, 2023-06-08:
In port today and ran a test while not moving.
N2K + 0183 + track + some plugins.  Crashed after 4:28 hrs.  Crashed handing track during onExit.
I think it does an exit from a funny place and crashes during cleaning up the track log..

Now I can reproduce it will try with simplified set up.

NB Am on M2 Mac.

bdbcat, 2023-06-09:
I set up a test on Mac M1, MacOS 12.1.
Actisense NGT-1 USB
N2K data from static GPS receiver.
Tracking enabled.
Running for 4 hours so far, no crash.  Nothing unusual in Activity Monitor, memory stable.
Some position jitter, related to a buffer problem identified and resolved in github master for next release.

antipole2, 2023-06-11:
**Perplexed**.
Being in port for a few days and having been able to re-create it when static in port, I have run a series of trials starting with just N2K input and gradually adding back other components, (0183 in, out, tracking, plugins etc.).  Now worked up to everything in and no error over 6-7 hrs.  Sigh!
It must be a very fluky set of circumstances.

antipole2, 2023-06-28:
Having avoided N2K input for the above reasons, I tried it again with the new OCPN v5.8.4.  OCPN quit after about 40 minutes - no crash report.
N2K & 0183 input, 0183 output, track recording and active route.
MacOS Ventura 13.3.1
So the problem still exists IMHO

antipole2, 2024-05-12:
An update... I am still seeing this with OCPN v5.9.  Attached is a recent dump.

Additional observation:
When the crash happens, it interrupts route recording, of course.  I restart OCPN and start recording the route again.
Now the new part B of the route behaves normally but the interrupted Part A does not.  I cannot 'pick it up' - when I hover the cursor over it, no route pop-up appears, like it does for Part B.  The interrupted route has been left in some state whereby it is not recognised as a rout.

If I restart OCPN again, then all is well and Part A is recognised.

Note in the crash report that it is both reading the serial input (NMEA2000) and creating a track segment.  Perhap it happens when the two activities co-incide?
[image]

antipole2, 2024-06-03:
Another crash with OCPN5.9 while receiving N2K input and recording track.
Having to not use N2K input - relying on NMEA0183.
[image]

Hakansv, 2024-06-03:
If relevant:
On Win10 I can use a N2k source and enabled own ship track. But: My track is stable, not moving, so without a lot points.

antipole2, 2024-06-03:
I stress that this crash is quite rare - sometimes after 5-6 hours, sometimes sooner.
I suspect some coincidence of processes, such as when receiving N2K coincides with updating a track segment.

antipole2, 2024-07-02:
I think this still occurs with OCPN 5.9.2 (Beta2).
I am in port so ran a test using the driver.js script to drive the ship while recording the track.
(In this simulation, there is a conflict between the simulated position and the true position also being received.  However, the ship position was driven across the chart as per the simulation.)

After six hours I tried to open the Options panel and got a crash.

The dump (attached) looks like the others I have submitted.  Active threads for adding segment to track and for reading the serial (NMEA2000) input.  I think this crash happens when these two actions coincide.
[My NMEA0183 input is Position 2 and the serial NMEA2000 Position 1.]

[MultiinputCrash.txt.zip](https://github.com/user-attachments/files/16072811/MultiinputCrash.txt.zip)

bdbcat, 2024-07-03:
Tony...
I found a code exposure that could explain the occasional crash.
If you could install and test one of these fresh builds, we could confirm the fix before the next Beta.

https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/OpenCPN_5.9.0-legacyintel+10076.bebefc8a9-12.pkg

https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/OpenCPN_5.9.0-universal+10080.bebefc8a9-14.pkg

Thanks
Dave

antipole2, 2024-07-03:
Sorry - tried this and got a crash after just over one hour without me doing anything - was away from the boat..  Crash report attached.

Incidentally, this also showed what I have seen before - that after this crash, the track up to the date is not pick-usable. -no pop-up when hovering over it.  You need to restart again and then it is pick-usable.  I guess the crash while updating the track leaves it not identified as a track?  Might this be a clue?

Thanks, Tony

￼

bdbcat, 2024-07-03:
That is a clue. 
But I don't see the crash report.

antipole2, 2024-07-03:
> That is a clue. But I don't see the crash report.

I replied by email - maybe it did not get included that way.  Here it is
[MultiinputCrash.txt.zip](https://github.com/user-attachments/files/16087500/MultiinputCrash.txt.zip)

bdbcat, 2024-07-03:
It is a different crash signature.
Same effect, though.
Looking into it.

bdbcat, 2024-07-04:
Trying some tests here, with real MacOS hardware.
Do you think the error is associated with the N2K serial input from the Actisense gateway?
Or any other intuition?

bdbcat, 2024-07-04:
Tony...
Trying to test Actisense NGT-1 on MacOS Monterey, M1 chip.
Which drivers do I need?
What baud rate do you use?

antipole2, 2024-07-04:
Crash after 30 mins or so of dual input - both NMEA0173 & N2K. This time track lost apart from one point.
Crash report
[multi-input2.txt.zip](https://github.com/user-attachments/files/16100508/multi-input2.txt.zip)

Will now try dual input and no track recording.

bdbcat, 2024-07-04:
OK, I did finally succeed in getting a similar crash after 6 hours.
Setup:
Two GPS sources:
1. NGT-1 serial active, live N2K data
2. Actisense Wifi, same live N2K data.

Tracking enabled.
Digging in....

antipole2, 2024-07-04:
Just ran for 5 hours with both inputs and no problems.  A difference was that the actual position was trumping the simulator positions, so the boat did not move (unless I disconnected the inputs).   I do not know why it was different this time.

By the way, [#3237](https://github.com/OpenCPN/OpenCPN/issues/3237#issue-1708916856) still applies.  I had to enable/disable to get the NMEA0183 data through.  Perhaps the order of connection affects the precedence?

antipole2, 2024-07-05:
> Tony...
> Some information. There seems to be a small memory leak in OCPN, related to the default background chart rendereing.
> My question is this: Are you getting your crash with any part of the background chart visible, or do you always have a legitimate installed chart visible?

Always have o-charts chart - Norway this season and last.

antipole2, 2024-07-30:
Still seeing problems with Beta 3.
NMEA2000 input list position 1
NMEA0183 list position 2
This time I am getting rogue incorrect positions so my ship icon was jumping around wildley.
Example jumped to position
56° 37.501'N 047° 26.790'W
when actual position taken a few moments later was
56° 37.380'N 012° 49.283'E

Followed by crash - crash report attached.
I was also logging with VDR, so VDR file also attached. ( I have cleaned it up by inserting new lines in front of $)
[crashReport.txt.zip](https://github.com/user-attachments/files/16430332/crashReport.txt.zip)
[VDRrecord.txt.zip](https://github.com/user-attachments/files/16430335/VDRrecord.txt.zip)

To emphasise: if I disconnect the NMEA2000 input, there is no issue
 Since I do not really need NMEA2000, I can manage without.

ocornu, 2025-01-16:
I seem to have the same issue on linux (ubuntu noble) with an Actisense NGT-1-USB, when using two opencpn connections (NMEA0183 & N2K, both USB).

I hadn't used this setup since early April 2023 (it was working fine at the time). I updated the sources to the latest github/master (same OpenCpn config) a couple days ago and the bug started showing up. Compiled 5.10.0 in an attempt to bisect, but bug was there too. Fell on this ticket while investigating and, indeed, removing the NMEA0183 opencpn connection seemed to have solved the issue.

Here's the kind of feedback i get from the command line when opencpn crashes:
```
$ opencpn
*** Caught unhandled unknown exception; terminating
10:26:27: Closing NMEA NetworkDataStream 2947
10:26:27: Closing N2K Driver Serial:/dev/ttyUSB0
10:26:27: Stopping Secondary Thread
10:26:28: Stopped in 1 sec.
10:26:28: Closing NMEA Driver Serial:/dev/ttyUSB0
10:26:28: Stopping Secondary Thread
10:26:29: Stopped in 1 sec.
```

Notice that i've had variations, like:
```
$ opencpn
*** Caught unhandled unknown exception; terminating
Erreur de segmentation (core dumped)

$ opencpn
*** Caught unhandled unknown exception; terminating
09:49:06: Closing N2K Driver Serial:/dev/ttyUSB0
09:49:06: Stopping Secondary Thread
Fatal glibc error: tpp.c:83 (__pthread_tpp_change_priority): assertion failed: new_prio == -1 || (new_prio >= fifo_min_prio && new_prio <= fifo_max_prio)
Abandon (core dumped)
```
In any case, it always starts with an uncaught exception somewhere.

Not sure if that's related, but I've also noticed the dashboard had regular hick-ups (displayed values would randomly switch to "---"). This issue has disappeared since using N2K only.

bdbcat, 2025-01-16:
You might get some useful information by running OCPN under gdb.  At least maybe a stack trace...

ocornu, 2025-01-16:
Will try that later if i find the time (i'm not familiar with gdb)…

Another option would be to bissect it, considering master from March 2023 was ok, and this bug report dates back from June 1st. It's just a couple months worths of commits to go through.

leamas, 2025-01-16:
> Will try that later if i find the time (i'm not familiar with gdb).

It's not that complicated: https://opencpn-manuals.github.io/main/opencpn-dev/stacktrace.html

ocornu, 2025-01-17:
I guess it's gonna be less useful than you guys expected… :-D
```
*** Caught unhandled unknown exception; terminating
[Thread 0x7fffcd2006c0 (LWP 167972) exited]
[Thread 0x7fffcdc006c0 (LWP 167971) exited]
[Thread 0x7fffabe006c0 (LWP 167977) exited]
[Thread 0x7fffe20006c0 (LWP 167970) exited]
[Thread 0x7fffe2a006c0 (LWP 167969) exited]
08:50:35: Closing N2K Driver Serial:/dev/ttyUSB0
08:50:35: Stopping Secondary Thread
[Thread 0x7fffaa0006c0 (LWP 167980) exited]
08:50:36: Stopped in 1 sec.
08:50:36: Closing NMEA Driver Serial:/dev/ttyUSB0
08:50:36: Stopping Secondary Thread
[Thread 0x7fffa96006c0 (LWP 167981) exited]
[Thread 0x7fffab4006c0 (LWP 167978) exited]
[Thread 0x7fffe0c006c0 (LWP 167943) exited]
[Thread 0x7fffe34006c0 (LWP 167938) exited]
[Thread 0x7fffee0006c0 (LWP 167865) exited]
[Thread 0x7fffeea006c0 (LWP 167864) exited]
[Thread 0x7fffef4006c0 (LWP 167863) exited]
[Thread 0x7ffff3190d80 (LWP 167838) exited]
[Thread 0x7fffc0a006c0 (LWP 167976) exited]
[New process 167838]
[Inferior 1 (process 167838) exited with code 0377]
(gdb) bt
No stack.
```
Or am i doing something wrong?

leamas, 2025-01-17:
Nothing is wrong with what you do and this is actually some progress: we now at least what's going on,  an unhandled exception.

With some luck, you should be able to get a stacktrace with a breakpoint on abort i. e.:
```
$ gdb /usr/bin/opencpn
.... (lot's of output)
(gdb) break abort
(gdb) run
 ...Make it crash
(gdb) bt
```

Background: https://stackoverflow.com/questions/5718944

ocornu, 2025-01-18:
> With some luck, you should be able to get a stacktrace with a breakpoint on abort

Unfortunately, breaking on `abort()` does not yield any stacktrace (tried 3 times). Anything else you want me to try?

ocornu, 2025-01-18:
However, i did manage to get stacktraces on random app freezes (no unhandled exception), like this one:
```
(gdb) bt
#0  0x00007fffcccfb809 in GRIBUICtrlBar::~GRIBUICtrlBar
    (this=0x55555fc83cc0, __in_chrg=<optimized out>)
    at /home/piou/dev/OpenCPN/plugins/grib_pi/src/GribUIDialog.cpp:286
#1  0x00007fffcccfc486 in GRIBUICtrlBar::~GRIBUICtrlBar
    (this=0x55555fc83cc0, __in_chrg=<optimized out>)
    at /home/piou/dev/OpenCPN/plugins/grib_pi/src/GribUIDialog.cpp:350
#2  0x00007ffff7617fdc in wxWindowBase::Destroy() ()
    at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#3  0x00007ffff7618024 in wxWindowBase::DestroyChildren() ()
    at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
(gdb) 
```
Or that one:
```
Thread 1 "opencpn" received signal SIGSEGV, Segmentation fault.
malloc_consolidate (av=0x7ffff5803ac0 <main_arena>) at ./malloc/malloc.c:4866
warning: 4866	./malloc/malloc.c: Aucun fichier ou dossier de ce nom
(gdb) bt
#0  malloc_consolidate (av=0x7ffff5803ac0 <main_arena>) at ./malloc/malloc.c:4866
#1  0x00007ffff56aad95 in _int_free_maybe_consolidate
    (av=<error reading variable: Cannot access memory at address 0x7fffffffdc30>, size=<optimized out>) at ./malloc/malloc.c:4779
#2  0x00007ffff56ab42a in _int_free
    (av=0x7ffff5803ac0 <main_arena>, p=<optimized out>, have_lock=<optimized out>)
    at ./malloc/malloc.c:4646
(gdb)
```

leamas, 2025-01-18:
I don't think the other stacktraces are relevant, sorry. We know there is  an unhandled exception and we need to track it down.

After the first, failed attempt we IMHO need to start from the beginning. The first question is if there are plugins involved. That is, can you reproduce the crash if you start opencpn with the `--safe_mode` command line option? If you cannot, just report back and disregard the rest.

Next step is to run using gdb, but instead of putting a breakpoint on `abort` put it on `OnExceptionInMainLoop`, trying to get a stacktrace.  If this succeeds please report back, disregard the rest.

Next step is to use the  `--loglevel debug` command line option. This will create a lot of output into the logfile _~/.opencpn/opencpn.log_. Please attach that logfile here when you eventually report  back.

If there is nothing obvious in the end of the logfile, the question becomes if the problem can be reproduced on a debug build  
of current master. Do the build as described in the [manual](https://opencpn-manuals.github.io/main/opencpn-dev/linux.html#build_debian), but add the build type to the cmake command: `cmake -DCMAKE_BUILD_TYPE=Debug ..` 

Assuming the crash is still there, please report back attaching the `--loglevel debug` logfile (github requires that the logfile is compressed)