OpenCPN/OpenCPN #3325: WMM variation value not used

RooieDirk, 2023-06-09:
The global variable gVar (Magnetic variation) is set from instrument input even if the WMM plugin is available and enabled.
If WMM is disabled and a user variation entered this value is also not used but instead still the instrument sentence value.

Only tested for nmea, but the code looks the same for SignalK and nmea2k.

I see 2 solutions, (might be more)
  1) Use an "if (b_haveWMM) {...) in every input with possible variaton data in it,
  2) Add WMM_plugin to the priority list.

Dirk

Hakansv, 2023-06-11:
Dirk..
I think the philosophy so far is to; 
1. Thrust connected instruments to be relevant.
2. If you know the variation table in the instrument is outdated set its variation to 0.0 and WMM will "take over"

The "Add WMM_plugin to the priority list." sounds like a good idea. Maybe easier than to convince an old instrument to send variation = 0.0

RooieDirk, 2023-06-11:
If that is the philosophy, I think is is wrong. The WMM plugin is way more accurate then the algorithm build in a GPS. In my setup the difference is now  3ᵒ, a bit to much for comfort.
  I can't change the variation in my GPS, but possible could use the nmea-converter to add a 0.0 or ,,, for variation.

RooieDirk, 2023-06-11:
As a philosophy wouldn't it be more logical: If a user enables the WMM then use it!

Hakansv, 2023-06-11:
The same philosophy is valid for HDT. If present from a sensor use it. If only HDM is present use the variation, sensor's or WMM, to make HDT.
For the discussion: Would we trust the sensor's HDT or always make or own if WMM is enabled?

RooieDirk, 2023-06-11:
_For the discussion: Would we trust the sensor's HDT or always make or own if WMM is enabled?_ If it is magnetic I would prefer WMM, but we don't know what kind of sensor it is. It might as well be a GPS_compass with a 'true' true heading.  
For the discussion I would say: If a HDT is available use it as true. If a user wants to use WMM instead the HDT sentence should be blocked.

Hakansv, 2023-06-11:
Agree.
So this may be boiled down to add WMM as a variation source in "Communications priorities.."? If possible

closed 2026-01-03 by RooieDirk