OpenCPN/OpenCPN #3333: Transfer route to peer: inconsistent security

leamas, 2023-06-17:
**Describe the bug**
When transferring for example a route to a peer OpenCPN node and the "same" route exists at the peer, user is presented with a checkbox saying _Always replace objects from this source?_. The wording indicates that ticking this checkbox should make it possible to overwrite when making new transfers from the same machine without this question.

However, ticking this checkbox is basically a carte blanche to overwrite routes for any paired node, overwrites are not checked w r t source in any way. That is,  the actual behaviour does not match the user message.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a route in a opencpn instance.
2. Add the same route to another instance
3. Right-click on the route, choose  _Send to Peer_
4. Fill in required data, tick the "Always replace objects from this source?" checkbox
5. Initiate a new transfer, watch it overwrite the remote route without questions.
6. Initiate a new transfer of the same route from another machine, see it overwrite without questions

**Expected behavior**
Two solutions:
  - Actually check that updates comes from the same source  which did the first transfer.
  - Change the wordings in the checkbox to something like _Disable checks for uploaded duplicates_ or so.

**Additional context**

This is basically found  while reading code rather than testing. Not really sure what the best way to handle it would be. Perhaps there are other solutions.

leamas, 2023-06-27:
Thinking about this we should IMHO not make any checks for that updates comes from the originating peer. After all, the basic model is that any paired device is "trusted";  breaking this would mean entering a path to a permissions system based on peer. We don't that, for sure.

Which means that this bug could be fixed be just updating the user message.

bdbcat, 2023-06-28:
Maybe just:
_Always replace objects_

leamas, 2023-07-07:
> Maybe just:
*Always replace objects*

Yes. Should be a simple fix. Please feel free to make a patch and close this, otherwise I'll make a PR sooner or later (sailing...)

PR #3379 (2023-08-03): REST_server: Fix confusing message (#3333).
Closes: #3333
commit d59d2f8a (2023-08-03): REST_server: Fix confusing message (#3333).

Closes: #3333
closed 2023-08-03 by bdbcat