OpenCPN/OpenCPN #3341: No driver handle found for nmea2000 in release 5.8.4

douwefokkema, 2023-06-22:
Running the following code (part of AutotrackRaymarine_pi branch n2k):

bool found_handle = false;
    for (const auto& handle : GetActiveDrivers()) {
        const auto& attributes = GetAttributes(handle);
        if (attributes.find("protocol") == attributes.end())
            continue;
        wxLogMessage(wxT("handle proto %s"), attributes.at("protocol"));
        if (attributes.at("protocol") == "nmea2000") {
            m_handleN2k = handle;
            found_handle = true;
            break;
        }
    }
    if (!found_handle) wxLogMessage(wxT("nmea2000 handle not found"));

On Windows 10, with Actisense NGT-1 connected to the com port associated with the nmea2000 network, no nmea2000 handle was found. Switching to master branch of O (8d72ac) there is no problem, handle is found and nmea2000 output functions as it should.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS: [e.g. macOS]
 - Version [e.g. 22]

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: [e.g. Galaxy S20]
 - Andorid version: [e.g. 10]

**Additional context**
Add any other context about the problem here. Screenshots are usually a good way to ilustrate problems, attach them if applicable.

bdbcat, 2023-06-23:
Unable to reproduce locally with fresh build from github branch v5.8.x.
Did you build O5.8.4 locally, or use the Production Release image?

bdbcat, 2023-06-23:
Also unable to reproduce using a contrived plugin with OCPN 5.8.4 Release Image.

douwefokkema, 2023-06-23:
I can not try to reproduce, at home now. Will try again next time on board. Close for now.

closed 2023-06-23 by douwefokkema