OpenCPN/OpenCPN #3343: ais_info_gui.cpp - no diferent alert sounds triggert

Andreas-Argo, 2023-06-23:
**Describe the bug**
On AIS-Alerts you can chouse diferent soundfiles to play on alerts for CPA, DSC and SART. The Testing ot the sounds works but when it comes to real Alerts only the AISAUDIO_CPA alert and its asociated soundfile ist played. This is also so if the alert is an AISAUDIO_SART or AISAUDIO-DSC.

**To Reproduce**
Steps to reproduce the behavior:
Use and VDR -TEST file

**Expected behavior**
To play the designated soundfile for every Alert Type

**Desktop (please complete the following information if applicable):**
Raspberry Pi 4, 8GB
Opencpn 5.8.2

**Additional context**
It looks like in the file ais_info_gui.cpp at line 290 onwards
the "switch (audioType)" for the "cases" AISAUDIO_CPA, AISAUDIO_DSC and AISAUDIO_SART ar not defined.
The soundfile is then played according to what is set at "default"
If you change the "default" to an other case you get an other file played, but again for any type of alert it will be the same soundfile.

As I´m new to github y hope I explained well.
Thanks a lot for the great work
Andreas

Andreas-Argo, 2023-06-23:
I just confirmed the same behavior on my Windows 11 opencpn 5.8.4
the sound is the CPA and not the desired SART sound.
Screenshot and vdr file atached
[image]
[1-SART.txt](https://github.com/OpenCPN/OpenCPN/files/11850009/1-SART.txt)
greetings 
Andreas

TwoCanPlugIn, 2023-06-24:
Also described on the following post https://www.cruisersforum.com/forums/f134/ais-sart-alerts-275727-2.html#post3779188

Note also that when no sound card is present testing or playing a sound crashes O.

commit c52e1aab (2024-02-29): Correct AIS audio alert sound selection for DSC and SART alerts.  Closes #3343
closed 2024-02-29 by bdbcat