OpenCPN/OpenCPN #3344: Raymarine RS150 GNSS RC, YDNU-02, N2K, Dashboard - GPS Status = Bug

bc38, 2023-06-25:
Navigation based on Raymarine Axiom with external receiver Raymarine RS150, connected to CPN via NMEA 2000 USB Gateway YDNU-02.
Connection in CPN NMEA2000
Dashboard displays status as on screen
[image]
If the connection is changed to NMEA0183 then everything is fine.
[image]
Previously, I connected via YDNU-02 to Lowrance with an external Point-1 antenna and the gps status showed the correct data.

bdbcat, 2023-06-25:
Is the reported position correct?
So, is the problem only the dashboard display of GNSS status?

bc38, 2023-06-25:
> Верна ли заявленная позиция? Итак, проблема только в отображении на приборной панели статуса GNSS?

The position is stable, but there are occasional freezes, it helps to reconnect the gateway to USB

Hakansv, 2023-06-25:
Just to compare. 
On W10. Display scaling 175%
N2k by YDNU-02 connected to Simrad GS-25 (GNSS & Heading).
Here we receive PGN129029 and PGN129540. These are used by Dashboard to create GNSS Status data.
[image]

Hakansv, 2023-06-25:
Also for info:
Connection status:
[image]

bdbcat, 2025-04-26:
Closing, please re-submit with more recent OpenCPN build, if necessary.

closed 2025-04-26 by bdbcat