OpenCPN/OpenCPN #3348: OCPN doesn't send RMB sentence to bluetooth link

Rodemfr, 2023-06-28:
**Describe the bug**
Only on Android : OpenCPN does not send RMB sentence on the bluetooth-serial link.

**To Reproduce**
My configuration : 
 - OCPN receives simulated ship data from a TCP socket via WiFi (input link only)
 - OCPN is configured to output NMEA data to a bluetooth-serial NMEA device (Micronet NMEA bridge)
 - A route is active and OCPN shows BRG/XTE/etc. in the navigation console on the right of the screen
 - OCPN sends all ship sentences to bluetooth but not the navigation related sentences (RMB)
 - If I create an additional TCP output port, I see all sentences passing on it, including RMB which makes me think the problem is related to bluetooth only.

**Expected behavior**
I should receive RMB on the bluetooth link

**Screenshots**
N/A

**Desktop (please complete the following information if applicable):**
 - OS: Android
 - Version 5.6.2

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: Galaxy Tab A6
 - Andorid version: 8.1.0

**Additional context**
N/A

I would be happy to debug this myself, but I did not manage to compile OCPN for android so far. I managed to compile OCPN library with the CI scripts but not to compile the final application with QtCreator.

bdbcat, 2023-06-30:
Issue identified, will be corrected in next Android Release.
Thanks for the bug report.
Dave

bdbcat, 2023-07-08:
Corrected in OpenCPN for Android, v5.8.4[101].
Now available from Google Playstore.
Please test and close, if successful.

Rodemfr, 2023-07-09:
Tested latest beta version 5.8.4 from Play Store : it(s working fine. RMB is properly sent.
Thanks !

closed 2023-07-09 by Rodemfr