OpenCPN/OpenCPN #3354: GPX Import Track: Negative speed and track lost after restart

arkuuu, 2023-07-07:
**Describe the bug**
Importing the appended GPX file successfully shows Waypoints and Tracks in OpenCPN, but the speed in the track is negative.
After restarting OpenCPN, the tracks are gone, the waypoints are still there.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Routes&Waypoint Manager
2. Click on Tracks, and then import the appended GPX
3. See Tracks and Waypoints on map
4. See negative speed in track details window
5. Restart OpenCPN
6. Track is now gone on map and on Routes&Waypoints Manager

**Expected behavior**
Speed should be positive. Track should still be there after restart

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
 - OS: macOS 13.4.1
 - OpenCPN Version 5.6.2

**GPX-File (.txt to allow uploading it here)**
[GPX-GPX_1.1-3.7.2023, 23_37_55.txt](https://github.com/OpenCPN/OpenCPN/files/11980769/GPX-GPX_1.1-3.7.2023.23_37_55.txt)

transmitterdan, 2023-07-07:
I used the latest version of OpenCPN to test, not V5.6.2.

I imported the file and the waypoints are ordered in reverse time.  The first point in the file has a timestamp that is after the next point in the file.  OpenCPN displays each leg of the route as 0 speed because of this, but it displays the average track speed as negative because the time to complete the track is negative.  In other words, the track finished before it started.  OpenCPN does not sort the track based on time during import.  It just imports the track points in the order they appear in the GPX file.  A properly formed GPX file should have the track points in the natural time order, not reversed.

I could not reproduce the loss of track on close.  That is probably another issue.  Look at the log file to see if OpenCPN complains about anything during shutdown.  That's usually why you lose a new track on close.  It probably has nothing to do with the track itself.

arkuuu, 2023-07-07:
Thanks! That sounds so obvious now that they are simply sorted the wrong way round! Don't know why I didn't think about that. 
I'll check with Yacht Devices, the manufacturer of my NMEA router, if they can fix this, as this gpx is exported from one of their devices. 

For the other issue (imported track lost after restart) I'll try the current version of OpenCPN and check the logs and will report back.

arkuuu, 2023-07-08:
I was able to reverse the tracks with [GPSBabel](https://www.gpsbabel.org).
Now they are imported correctly and speed is shown correctly.
They are now also still there after restarting OpenCPN! Looks like the wrong order of the track messed up with it.

The command I used with GPSBabel was (reverse and only keep tracks)
```bash
gpsbabel -t -i gpx -f input.gpx -x nuketypes,routes,waypoints -x reverse -o gpx -F output.gpx
```

So, is this now a bug in OpenCPN?
I contacted the manufacturer of my device and he told me that in his opinion the current [GPX XML Schema](https://www.topografix.com/GPX/1/1/) does not enforce time order, but only a _logically connected order_ for the track segments. So he is free to provide in reverse order and thinks that the consuming software (OpenCPN in this case) should be able to handle both. I argued that time runs forwards and not backward - so a _logically connected order_ for timed events should not be reversed. He agreed and they might fix their export.
But I also agree that OpenCPN should be able to handle this. Can it be done?

nohal, 2024-02-13:
"Everything" can be done, but to be honest I'm not convinced it is worth the effort and the non-zero impact on performance in this case. I have actually never seen anything producing tracks in reverse order as they are supposed to come from a GPS and time runs forward only. If there is a real chance we could see this problem regularly, I can sure be convinced to implement such a thing, but so far consider it a simple bug on the side of the producer of the GPX file.

bdbcat, 2024-02-28:
Given the rarity of occurance, and a valid GPSBabel workaround, I'd call this NOTABUG, WONTFIX.
Closing.

closed 2024-02-28 by bdbcat