OpenCPN/OpenCPN #3355: OpenCPN 5.8.4 on Raspberry Pi 3B+ running bionic fails to start

jongough, 2023-07-09:
**Describe the bug**
Upgraded OCPN 5.6.2 to latest 5.8.4 on Raspberry Pi 3B+ using Raspbian Buster.
Start opencpn starts but does not complete, the busy mouse pointer keeps on showing.
Terminate opencpn and restart with plugins disabled and will start up. 
Restart again and it hangs with the screen showing and the busy mouse pointer.

**To Reproduce**
Steps to reproduce the behavior:
1. Install OPCN 5.8.4
2. Run OCPN from command prompt or application icon
3. OCPN starts, but does not complete.
4. Turn off OpenGL and disable it using instructions available from 'failed restart dialog' and get an error message
5. *** BUG ***
In pixman_region32_init_rect: Invalid rectangle passed
Set a breakpoint on '_pixman_log_error' to debug

**Expected behavior**
OCPN start and run normally

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information if applicable):**
 - OS:Raspbian
 - Version  10 (buster)

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: N/A
 - Andorid version: N/A

**Additional context**
Add any other context about the problem here. Screenshots are usually a good way to ilustrate problems, attach them if applicable.
[pi 3b debian 10 start issue - conf.txt](https://github.com/OpenCPN/OpenCPN/files/11994606/pi.3b.debian.10.start.issue.-.conf.txt)
[pi 3b debian 10 start issue - log.txt](https://github.com/OpenCPN/OpenCPN/files/11994607/pi.3b.debian.10.start.issue.-.log.txt)

bdbcat, 2023-07-10:
This looks like a plugin problem to me.
I first suggest that you simplify the config however possible.
1. remove/rename every plugin in ~/.local/lib/opencpn.
2. disable all comms by manual edit of config file.

If you can get a successful boot, add complexity step by step to isolate the offending element.

jongough, 2023-07-11:
I have found that there are two plugins that if not uninstalled/deleted .so files will cause the issue. This is logbookkonni and vdr. I think they are using the wrong gtk version, 2 instead of 3. With these two installed the startup of opencpn fails with no indication why. Having removed all the plugins physically and then installing them one by one I found the issue. When refreshing the catalog neither of these two show up. 

Now that we know there is an issue with them perhaps we should warn anyone doing an 'upgrade' to actually remove the files in '.local/lib/opencpn/*so' to avoid issues if opencpn will not start. Part of the problem is that opencpn needs to call each plugin in turn for icon information even if it is not installed, if there is an issue opencpn may/can/will crash and give no help. Before plugin manager when a change of this sort was made it would be identified that the plugins where not compatible and would have warned the users. Currently this does not happen.

So, should the upgrade instructions be to not only disable ALL plugins in plugin manager, but to remove any 'executable' file that may be found. These can then be installed again when the new version of opencpn is running, and at that time only compatible plugins will be shown.

jongough, 2023-07-11:
My Pi is running Buster which is considered 'old', but it still functions and it is a shame that neither LogbookKonni nor vdr have been built for that platform.

I have also 'upgraded' to focal/bullseye which is not trivial and really needs a script to help with the process.

jongough, 2023-07-11:
I would be grateful if buster armhf versions were available to match the buster OCPN 5.6.4.

What this does bring up is the need for a better way to handle plugins over an upgrade of OCPN. The current process is rather 'cross your fingers and hope for the best' and if it goes wrong, well you need to do some debugging. This is not really user friendly. 

I think there should be an option, within the OCPN plugins panel, to not load plugins at all, i.e. not even get their icons, but give a list of the currently installed and currently active plugin names (all accessible from the conf file). The user could then refresh the list of available plugins, after an upgrade, and install the ones available, and notify us, as a group, of any that are missing.

jongough, 2023-07-11:
> My Pi is running Buster which is considered 'old', but it still functions and it is a shame that neither LogbookKonni nor vdr have been built for that platform.
> 
> I have also 'upgraded' to focal/bullseye which is not trivial and really needs a script to help with the process.

The script mentioned can be found here for the Buster to Bullseye upgrade. I used it successfully on my Pi 3B after making an image copy of the sc card.
[https://forums.raspberrypi.com/viewtopic.php?t=317888](https://forums.raspberrypi.com/viewtopic.php?t=317888)

leamas, 2023-07-16:
> My Pi is running Buster which is considered 'old', but it still functions and it is a shame that neither LogbookKonni nor vdr have been built for that platform.

This is really two questions. The first is about these plugins being available at all. The general understanding is that they should be, using the already existing possibly old builds available.   Looking in ocpn-plugins.xml LogbookKonni is actually available for Debian-10/armhf/ since https://github.com/OpenCPN/plugins/commit/b2404e1ecadd8118be6ad0cc2f577800265f678a.  One possible problem is might be that the old buster version actually is built for bionic and that the code to use debian-10 plugins in this case is not in place. If so, it should be possible to tweak the bionic build to rightfully accept  debian-10 plugins using Options | Plugins | Settings | Conpatibility.

The other question is if LogbookKonni latest version is built for buster. Here, the answer basically is no, and this is on purpose. We don't update plugins for old platforms, just uses the already existing builds. Of course, this is in the end up to the plugin devs and packagers, but I think we have agreed on this as a general rule. 

The situation for VDR seems to roughly the same: It exists in ocpn-plugins.xml since https://github.com/OpenCPN/plugins/commit/b2404e1ecadd8118be6ad0cc2f577800265f678a for debian-10.

jongough, 2023-07-16:
Plugin manager should fail gracefully not lock up the system. With plugin manager the user has to upgrade the plugins after the startup, but in this case the system will only startup in fallback mode with all the plugins disabled. The user can then try to get the new versions of plugins but if they are not offered there is no way to know that they will not work. There is no message in the OCPN log, there is now dialog box saying so, just a hard lockup with the only solution being to force OCPN down. This process is not user friendly and gives no indication of what may be wrong.

The deb install method would at least warn the user when there were incompatibilities. Plugin manager methods do not. When upgrading OCPN from 5.6 to 5.8 it assumed that everyone would have OpenGL, but this capability did not come in for the Pi 3B until Bullseye and that caused issues with performance and trying to eliminate it as the root issue. I found the misbehaving plugins by disabling them all then restarting OCPN and enabling each, one at a time and restarting. All I found out was that two plugins caused the lockup but not why. For Devs this is a pain, for a standard user it is a non-starter. OCPN should not 'force' OpenGL and plugin manager should degrade in a controlled manner and provide information to users as to what is wrong.

leamas, 2023-07-17:
The thing with the 5.8 update is that it is an ABI break. Most updates are not and plugins stays compatible with the new OpenCPN version, avoiding this mess.

TBH, ABI breaking updates is an after thought, it was not part of the original design. It shows up here. OTOH, next ABI break is probably gtk4, and this will not happen any time soon. So, there is time to think about  ABI breaking updates. 

How these updates actually should be handled is not really defined, and it's thus not possible to state what the plugin manager should or should not do. The first step would be to define the user story, something like:

1. User installs new, ABI-breaking version of OpenCPN using native tools (NSIS, apt, flatpak...)
2. User starts new version
3. Now what? Alternatives  presented to user to handle incompatible plugins?
4. ...

>  OCPN should not 'force' OpenGL

If this is an issue it's IMHO a separate one.

leamas, 2023-07-17:
> Also, remember that OCPN 5.8.4 installs on armhf raspbian buster without a problem and is one of the 'upgrades' available. So, by that reasoning the plugins should also be available and rebuilt not just frozen.

This is one of thge aspects of an ABI breaking update, but does not apply to regular updates which does not break the ABI.

rgleason, 2023-07-17:
This may be a dumb question, but isn't there something we can do to fix VDR and Logbook since these are the only plugins with this known problem?

leamas, 2023-07-17:
As I understand it, there is not much you can do -- these plugins are already in the catalog for debian-10/armhf.

The problem is the upgrade procedure on buster, which seems flaky at least.  The reason is of course that we  have not spent much time on this case, for design and testing. But now 5.8 is out and the train has left the platform.

jongough, 2023-07-17:
The upgrade procedure, especially one which changes versions of critical enabling software, is not very robust. The fact that OCPN has to make an API call to each plugin that is installed, even if not activated, to ascertain its icon to use in plugin manager is going to cause issues.

A more robust way would be for a significant change like this one would be to disable all plugins automatically and make no use of the currently installed plugin binary whilst informing the user that there could be potential issues with non-upgraded plugins. The the user should be asked to update the plugin catalog and upgrade any plugin that is installed on their system, preferably with an OCPN restart after each plugin upgrade to ascertain if all is OK with their install.

One way of achieving this would be to rename the plugin binary so it cannot be run until it has been reinstalled.

This issue was raised directly as a result of an OCPN upgrade and it crashed before the user had any chance of upgrading the plugins. This was on armhf buster/bionic. There was no help or information provided to the user of what had gone wrong or how to fix it. Some plugins did have newer versions available to match OCPN others did not. It would be difficult for a non 'Dev' to fix this type of issue.

jongough, 2023-07-17:
> > OCPN should not 'force' OpenGL
> 
> If this is an issue it's IMHO a separate one.

It has been raised multiple times by multiple people on different platforms. It appears that those people are not 'normal' users. OCPN should not enforce OpenGL as it causes a similar effect to this issue, i.e. complete lock up of OCPN with no information in the opencpn.log or system logs. A not very helpful situation. 

If plugin manager needs to have the icons of plugins before they are installed/enabled then either the xml should carry the icon or plugin manager should supply a default icon until the real one can be ascertained. The plugin should on be called after it has been either enabled or the enable check box is clicked. This would make it much easier to determine which plugin has the issue. Also, better logging of the install/enable process to help identify such issues would be helpful.

jongough, 2023-07-19:
All update processes should assist the user, even ABI-breaking (really dependency breaking) updates. The process should assist the user to find the plugins that are causing the issue, not leave them with a non-working machine and no information to provide when raising issues or using the forums. In this case there was no information provided at all and the symptoms between a non-compatible plugin and OpenGL being forced were very similar. The old process would have checked and shown that the plugin installation would be broken using the inbuilt dependencies. This does not appear to happen currently. 

Not everyone is on Bullseye and it is not a straightforward process to get from buster to bullseye. It requires the use of a scripted process as some standard packages block the update process and have to be removed and then reinstalled. This is not a task for a standard user. So case not closed.

jongough, 2023-07-20:
The script I used to do the upgrade was found here:
[https://forums.raspberrypi.com/viewtopic.php?t=317888](https://forums.raspberrypi.com/viewtopic.php?t=317888)
It takes a while, but it worked. I did do a full backup of the sdcard before I started, just in case.

nohal, 2024-02-17:
Buster is years past EOL, everybody is migrating from Bullseye to Bookworm and there is no Armageddon happening. Although we still do have stuff to improve, this issue has outlived it's purpose.

closed 2024-02-17 by nohal