OpenCPN/OpenCPN #3361: AIS SART Messages - incorrect MMSI

Rasbats, 2023-07-21:
**Describe the bug**
When a SART is activated the message shows the SART MMSI with 9 digits starting with 970. 970 is the MID. The next 9 digits are the MMSI and this is what should be shown in the message.
**To Reproduce**

1. Wait for a SART to be activated
2. Message appears
3. Observe the MMSI
4. See error

**Expected behavior**
The message should show the MMSI of the ship/person in distress and not include the MID.

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
 - OS: Windows 10

Rasbats, 2023-07-21:
This happens with a feed from aisHub.net as well as a simulated alarm from ShipDriver.

Hakansv, 2023-07-21:
@Rasbats 
What I know a SART MMSI has no MID, i.e country code.
The SART MMSI is 
970XXyyyy where, 970 says is a a SART, XX is a manufacturer code and yyyy is a ID-number.
See more for example here: https://cirm.org/services/manufacturer-ids.html

Rasbats, 2023-07-22:
Thanks. I was using incorrect information and did not refer to the latest regulations:
[https://www.itu.int/rec/R-REC-M.1371-5-201402-I/en](https://www.itu.int/rec/R-REC-M.1371-5-201402-I/en)
Closing this.

closed 2023-07-22 by Rasbats