OpenCPN/OpenCPN #3362: Cell with lower details overlaps cell with more details

erik-meuwese-topicus, 2023-07-22:
**Describe the bug**
In the Netherlands there are ENC cells with different amount of details overlapping each other. To see the detailed cells you have to hide the less detailed cells. But when you hide them then other areas where there is no overlap are no longer visible. 

Lower detail cells without depth information are overlapping the detailed charts with the depth information.

**To Reproduce**
Steps to reproduce the behavior:
1. Fresh install 5.8.4
2. Options -> Charts -> Chart Downloader -> Select Catalog...
3. Select 'Netherlands Inland ENC Charts'
4. Download Wadenzee met Diepte 2023 and Nederland (excl. Zeeland) 2023
5. Go to the Waddenzee and northern provinces Friesland and Groningen
6. Example in the screenshots is Delfzijl and the Eemskanaal

**Expected behavior**
Detailed cell 1R7EMS01 with the depth information should overlap the less detailed cell of the 1R7EK000 without depth information. Not the other way around

**Screenshots**
[image]
[image]
Lower detail cells without depth information are overlapping the detailed charts with the depth information
[image]
The Eemskanaal is not visible when the detailed chart of Delfzijl (right bottom)  is visible

**Desktop**
 - OS: Windows
 - Version 10, 11

RooieDirk, 2023-11-24:
This is due to the charts used. Each chart cell does have a chart-scale, and OpenCPN does choose the chart used depending on this chart-scale. For both chartsets the chart-scale is set 1:2000, so is rather random wich chart is chown.

tip  delete  all the charts 1RWZ*.000 from the 'Nederland(exl Zeeland)' chart set, and wait for the chart supplyer (RWS) to fix it. (In due time the chart set should be changed to 'Nederland(exl Zeeland, excl Waddenzee')

leamas, 2025-07-18:
@erik-meuwese-topicus Given @RooieDirk 's explanation I tend to think that this should be closed, basically as NOTABUG. Thought on this?

RooieDirk, 2025-07-20:
Is indeed fixed by the chart supplier now.
[https://vaarweginformatie.nl/frp/main/#/page/infra_enc](url)

[image]