OpenCPN/OpenCPN #3363: Filenames and German Umlaut

nocheinpeter, 2023-07-23:
**Describe the bug**
.gpx files containing german umlaut in filename / title cannot be saved or oadad under android.
Save routes with umlaut like ä, ö or ü in filename are empty
Files saved under windows containing umlaut cannot be loaded for example from draw plugin.

**To Reproduce**
Steps to reproduce the behavior:
1. create route
2. give name containing ä, ö or ü
3. export route with a filename containing ä, ö or ü
4. file is empty

or with draw plugin
1. create area on windows
2. give title containing ä, ö or ü
3. export are with od plugin
4. import are on android with od plugin

**Expected behavior**
A clear and concise description of what you expected to happen.

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: Galaxy Tab A 2019

**Additional context**
[Test German Umlaut ä.zip](https://github.com/OpenCPN/OpenCPN/files/12138384/Test.German.Umlaut.a.zip)
[Draw Example.zip](https://github.com/OpenCPN/OpenCPN/files/12138385/Draw.Example.zip)

Apeared first with od plugin, so reported [here](https://github.com/jongough/ocpn_draw_pi/issues/530)

leamas, 2023-07-23:
The part "Expected behaviour" is missing. What are your thoughts about this?

It looks like UTF-encoded filenames is a can of worms when applied to different platforms.  Perhaps it would be better to avoid/block filenames with non-ASCII characters completely?

leamas, 2024-04-28:
At a second look, this seems to be about the Ocpn Draw (OD) plugin, and  the file should thus be filed against that plugin.

The OD plugin is maintained at  https://github.com/jongough/ocpn_draw_pi

leamas, 2024-04-28:
Filed https://github.com/jongough/ocpn_draw_pi/issues/550, closing this issue

closed 2024-04-28 by leamas

jongough, 2024-04-28:
The issue exists in OpenCPN under android and therefor it is still an issue for OpenCPN. It would appear to be an issue with Android and its handling of national characters. This issue needs to be reopened.

jongough, 2024-04-28:
Checking the code used in ODraw for android, it calls pluginmanager.cpp function PlatformFileSelectorDialog to get the file and return either wxID_OK or wxID_CANCEL with the default being wxID_CANCEL. As both OCPN and ODraw are suffering the same issue it would suggest that some analysis needs to be done on the android file handling section of OCPN.

leamas, 2024-04-29:
The issue reported here is what happens when exporting and importing to/from OD. Hence, to correct action is to transfer this bug to OD as done.

OTOH, you describe a bug in the OpenCPN plugin API. This is valid, but a different issue. This is obvious when thinking about "what is the problem", "steps to reproduce", etc.  So. filing a new issue against opencpn for this makes a lot of sense.

Note that this is the normal flow of bugs and issues: Users files issues against applications, maintainers reads this and either solves it or files new bugs upstream. The new, upstream bugs are typically  **not** the same as the original problem, they are about problems in APIs, tools and other stuff not visible to end users.

And in the end, OpenCPN is the upstream for OD.

jongough, 2024-04-30:
This issue describes what is happening. My last comment suggests where the issue is. So this issue is not complete. This issue will affect any plugin that is using the OCPN supplied file dialogs, it will not be limited to ODraw. The ODraw issue points to this one, as you suggest above it should do.

leamas, 2024-04-30:
This issue will remain closed.  Feel free to file a new one with the context you describe.

jongough, 2024-04-30:
I still do not understand why it was closed. The issue still exists. Filing a new issue provides no benefit.

jongough, 2024-04-30:
Perhaps you should raise the issue seeing as you are so keen to close this one. I have identified the area of OCPN that has the issue, but I have no way of building or testing any solution, so I cannot work on it. I have also identified that the issue is not fixed.

leamas, 2024-04-30:
Repeating the same arguments is not fruitful