OpenCPN/OpenCPN #3364: .gpx file for waypoint import not visible in file browser/explorer

codebluesailing, 2023-07-24:
The .gpx file is not visible in the pop up file browser window when attempting to import waypoints in Android 5.8.4. The file was duplicated in several directories and is not visible in any. The issue occurs in 2 instances: 1) Samsung Tablet S2 after updating to 5.8.4, and 2) a fresh install of Android 5.8.4 on a Samsung tablet S8. (Note: the .gpx file used for the S8 fresh install is the same file/name used to populate the S2 originally, and also validated with upload to Garmin and other apps)

**To Reproduce**
Steps to reproduce the behavior:
1. Open Routes/Tracks/Waypoints/Layers window, Click on the Waypoints tab
2. Click on "Import GPX" at the bottom of the window
3. Navigate to the directory containing the GPX file
4. GPX file is not visible in the directory

**Expected behavior**
GPX file should be visible to select for import

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: Galaxy Tab S8/Android version 13
 - Device: Galaxy Tab S2/Android version 7.0

**Additional context**
On the Galaxy S2, the GPX file imported as expected with a previous version (4.x ??), but the above issue appeared on the S2 after the update to 5.8.4.

bdbcat, 2023-07-24:
Lets first address import on the Android/7 device running O584.
What directory was the target GPX file saved to, and by what method?

codebluesailing, 2023-07-24:
Prior to the upgrade to 5.8.4, the file was in the tablet Download folder. The file is an output of a Garmin handheld. It was copied to my desktop, uploaded to Google Drive and downloaded to the tablet. 5.8.4 does not allow me to navigate to that folder. I copied it into the Charts {which I could access from OpenCpn file browser} using Android file utility and it is not visible there. ... same behavior in the S8 tablet in 5.8.4.

bdbcat, 2023-07-24:
Strange...
Just tested O584{102} on Android 8 phone.  Easily accessed "/storage/emulated/0/Downloads".

You should know:  On Android 12 and above, the Download directory is not directly accessible to applications like OpenCPN.  Google has implemented a policy known as "scoped storage" in Android/12 and above.  This policy essentially "sandboxes" application access to files that it owns, only.  System directories like "android" and "Download" are not directly accessible.  So the S8 tablet will never be able to access "Download" directly.  The work around is to use one of a very few file explorers (e.g. XPlore) that can access system "Download", and use that to copy GPX files to the OpenCPN private directory tree, something like 'Android/data/org.opencpn.opencpn/files/Download".  Then you may read and import the GPX  to OpenCPN as usual.

So, pardon the impertinence, but are you sure that the S2 tablet is running Android/7 ?

codebluesailing, 2023-07-24:
Ok, thanks..I'll try XPlore. Reading between the lines, I believe you are saying the native android "My Files" copy/move options wont allow the transfer "properly" to the OpenCpn directory. Even tho' I used the native capabiliity, the file was not visible in the OpenCPN directory.Yes, just double checked, the S2 uses Android 7.0. Samsung did not provide updates beyond that.Thanks so much for your help...I'll keep you posted re:XPlore.

codebluesailing, 2023-07-25:
Success! Storing the .gpx in the 'Android/data/org.opencpn.opencpn/files/' directory did the trick. Utimately I connected the S8 to my W10 desktop and used the file explorer to move the .gpx into the directory. X-Plore wasn't reading my folders/files, but probably didn't have the right settings to make it work. I did notice that for Charts, there is a utility in OpenCpn that migrates the charts from the system into this apps directory...that worked fine.Thanks again for the suggestion!!
PS--I'm going to play around with the S2 Android 7 to see what is happening. My initial guess is that 5.8.4 is limited to browsing in the "sandbox"
Cheers,Bob

codebluesailing, 2023-07-25:
Workaround needed to port .gpx file into Android OpenCpn: On Android 12 and above, the Download directory is not directly accessible to applications like OpenCPN. Google has implemented a policy known as "scoped storage" in Android/12 and above. This policy essentially "sandboxes" application access to files that it owns, only. System directories like "android" and "Download" are not directly accessible. So the S8 tablet will never be able to access "Download" directly. The work around is to use one of a very few file explorers (e.g. XPlore) that can access system "Download", and use that to copy GPX files to the OpenCPN private directory tree, something like 'Android/data/org.opencpn.opencpn/files/Download". Then you may read and import the GPX to OpenCPN as usual.

rgleason, 2023-07-26:
Thanks Alec, Have previously added a note in the Android install section. It could be expanded. I used bdbcat language.  Does anyone have text to summarize all versions of android file handling?  Just add it here please

rgleason, 2023-07-26:
Tbis should probably be changed to Discussion and "Android:" added to the title. I cant do that.

rgleason, 2023-07-26:
See https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:get_started:installation:android

rgleason, 2023-07-26:
@codebluesailing  Please feel free to revise the text and improve its applicability. Your experience with Android would be a great help to others.

codebluesailing, 2023-07-27:
suggested change in italics:
The work around to be able to read and write gpx files or charts, for example, is to use one of a very few file explorers (e.g. XPlore) that can access system “Download”. Alternatively, connect the Android device to a Windows PC with a USB cable, open the device in "This PC" folder, and use File Explorer to move the .gpx file to the OpenCpn directory from any the Android system folder or the PC. File Explorer is not constrained from moving files across the PC, Android and OpenCpn directories.