OpenCPN/OpenCPN #3365: AIS MOB SART Crashes OpenCPN on PI

rhinotronic, 2023-07-24:
**Describe the bug**
OpenCPN running on RPI crashes on receipt of an AIS SART message received by TCP or UDP network connection.
The same message sent to OpenCPN running on Windows results in the behaviour expected.

**To Reproduce**
Configure OpenCPN to receive AIS messages via network (both TCP and UDP)
Send an AIS SART message to OpenCPN using TCP or UDP
OpenCPN running on RPI crashes as soon as the message is received
This bug was encountered following a local SART activation and has been reproduced by sending AIS SART messaged generated in Python .

**Expected behavior**
AIS SART position displayed with zigzag circle around etc. as described in the product documentation.

**Screenshots**
No screenshots since OpenCPN crashes immediately
There is no entry written to the log file

**Desktop (please complete the following information if applicable):**
Raspberry Pi 4
RETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian

**Additional context**
Works perfectly on Windows version
Crash occurs irrespective of whether or not sound notifications for the message are enabled.
Testing sounds does not cause a crash
OpenGL not enabled (this negatively impacts performance)

nohal, 2024-02-17:
Can we have the Python code used to reproduce the crash or at least the NMEA messages involved?

rhinotronic, 2024-02-17:
Tested again, and the first sentence crashed OpenCPN running on PI, but the Windows version showed the AIS SART message as expected.
This test message also crashed OpenCPN on PI.  Tested by sending via UDP

SART TEST MESSAGE

***@***.***,0*17

SART ACTIVE MESSAGES

***@***.***,0*27

***@***.***,0*34

***@***.***,0*40

***@***.***,0*33

***@***.***,0*4E

***@***.***,0*0B

***@***.***,0*75

***@***.***,0*1F

***@***.***,0*64

***@***.***,0*7F

***@***.***,0*00

***@***.***,0*1E

***@***.***,0*12

***@***.***,0*70

***@***.***,0*3C

***@***.***,0*2F

***@***.***,0*31

***@***.***,0*3B

***@***.***,0*7C

***@***.***,0*04

***@***.***,0*5B

nohal, 2024-02-17:
I think this is not what you wanted to post, all the sentences seem to have been mostly eaten by some thing filtering your e-mails...

rhinotronic, 2024-02-17:
Sending again as a plain test message...

Tested again, and the first sentence crashed OpenCPN running on PI, but the
Windows version showed the AIS SART message as expected.
This test message also crashed OpenCPN on PI.  Tested by sending via UDP

SART TEST MESSAGE

***@***.***,0*17

SART ACTIVE MESSAGES

***@***.***,0*27

***@***.***,0*34

***@***.***,0*40

***@***.***,0*33

***@***.***,0*4E

***@***.***,0*0B

***@***.***,0*75

***@***.***,0*1F

***@***.***,0*64

***@***.***,0*7F

***@***.***,0*00

***@***.***,0*1E

***@***.***,0*12

***@***.***,0*70

***@***.***,0*3C

***@***.***,0*2F

***@***.***,0*31

***@***.***,0*3B

***@***.***,0*7C

***@***.***,0*04

***@***.***,0*5B

#3365)

https://github.com/OpenCPN/OpenCPN/issues/3365#issuecomment-1950167827, or
WB7AVCNFSM6AAAAAA2WE4S3WVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMYTSNJQGE3DOO
BSG4.

rhinotronic, 2024-02-17:
And again with sentences attached, in case it got mangled again.

 

Looked to be okay in my “sent items”

nohal, 2024-02-17:
And the result is exactly the same, please try to go to Github and reply using the web interface to reply. I don't know if this happens if this is done to the mail on your side or Github's , but it is done regardless of the e-mail format you use...

rhinotronic, 2024-02-17:
Tested again, and the first sentence crashed OpenCPN running on PI, but the Windows version showed the AIS SART message as expected.
This test message also crashed OpenCPN on PI.  Tested by sending via UDP

**SART TEST MESSAGE**

!AIVDM,1,1,,A,1>Nv;0?P00wW1T2KRDB00?w@2000,0*17

**SART ACTIVE MESSAGES**

!AIVDM,1,1,,A,1>M46P>P00wMbg2LW3P00?w@2000,0*27

!AIVDM,1,1,,A,1>M46P>P00wMbbFLW3200?w@2000,0*34

!AIVDM,1,1,,A,1>M46P>P00wMbUbLW2T00?w@2000,0*40

!AIVDM,1,1,,A,1>M46P>P00wMbPvLW2600?w@2000,0*33

!AIVDM,1,1,,A,1>M46P>P00wMbLBLW1`00?w@2000,0*4E

!AIVDM,1,1,,A,1>M46P>P00wMbGVLW1:00?w@2000,0*0B

!AIVDM,1,1,,A,1>M46P>P00wMbBrLW0d00?w@2000,0*75

!AIVDM,1,1,,A,1>M46P>P00wMb>>LW0>00?w@2000,0*1F

!AIVDM,1,1,,A,1>M46P>P00wMb9RLVwh00?w@2000,0*64

!AIVDM,1,1,,A,1>M46P>P00wMb4nLVwB00?w@2000,0*7F

!AIVDM,1,1,,A,1>M46P>P00wMb0:LVvl00?w@2000,0*00

!AIVDM,1,1,,A,1>M46P>P00wMasNLVvF00?w@2000,0*1E

!AIVDM,1,1,,A,1>M46P>P00wManjLVup00?w@2000,0*12

!AIVDM,1,1,,A,1>M46P>P00wMaj6LVuJ00?w@2000,0*70

!AIVDM,1,1,,A,1>M46P>P00wMaeJLVtt00?w@2000,0*3C

!AIVDM,1,1,,A,1>M46P>P00wMa`fLVtN00?w@2000,0*2F

!AIVDM,1,1,,A,1>M46P>P00wMaT2LVt000?w@2000,0*31

!AIVDM,1,1,,A,1>M46P>P00wMaOFLVsR00?w@2000,0*3B

!AIVDM,1,1,,A,1>M46P>P00wMaJbLVs400?w@2000,0*7C

!AIVDM,1,1,,A,1>M46P>P00wMaEvLVrV00?w@2000,0*04

!AIVDM,1,1,,A,1>M46P?P00wMaABLVr800?w@2000,0*5B

nohal, 2024-02-17:
Reproduced with 5.8.4 flatpak on `amd64`, but not 5.8.4 from `bookworm-backports` on a RPi (Or anything else I have currently running on my machines).

Just to confirm if my observation is right, is your installation on the Pi a flatpak or a normal package?

nohal, 2024-02-17:
The backtrace from the flatpak is not exactly revealing

Given what we are drawing the crash seems to be in https://github.com/OpenCPN/OpenCPN/blob/Release_5.8.4/src/ais.cpp#L629
But why only in flatpak?

```
Thread 1 "opencpn" received signal SIGSEGV, Segmentation fault.
0x0000555555a4a377 in ocpnDC::DrawCircle(int, int, int) ()
(gdb) bt
#0  0x0000555555a4a377 in ocpnDC::DrawCircle(int, int, int) ()
#1  0x000055555587abd7 in  ()
#2  0x0000555555880d0b in  ()
#3  0x00005555558825f5 in AISDraw(ocpnDC&, ViewPort&, ChartCanvas*) ()
#4  0x000055555590cd6d in ChartCanvas::UpdateAIS() ()
#5  0x0000555555a72150 in MyFrame::OnFrameTimer1(wxTimerEvent&) ()
```

rhinotronic, 2024-02-17:
Installed a long time ago, so the install history is gone, but Flatpack is not installed.
Looking at the sources list, the installed version may have come from ppa.launchpadcontent.net/opencpn/opencpn/ubuntu

Version = ConfigVersionString=Version 5.8.4 Build 2023-06-15

HW - Raspberry Pi 4 Model B Rev 1.2
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian

nohal, 2024-02-17:
The issue seems also gone with the flatpak from current CI builds. I will leave it open as we certainly need to retest with the normally released artifacts, but it simply seems to be gone although there were no changes I see related...

rhinotronic, 2024-02-17:
I'll remove and re-install with Flatpack ...

nohal, 2024-02-17:
Unfortunately the flatpaks from the CI builds are only available for `x86` architecture, so they can't currently be tested on the RPi. FWIW I am not able to reproduce it with the `5.8.4-3` flatpak running on Bookworm on a RPI 3

This means the only environment where I do see the crash is the  `5.8.4-3` flatpak running on my AMD workstation, everything else works fine fro me. That is a total mystery...

nohal, 2024-02-17:
OK, not that much of a mystery anymore, OpenGL must be disabled for this problem to materialize and the cause is `ocpnDC::DrawCircle` takes a wrong codepath trying to initialize GLSL shaders even in that case...

Fixed 4 months ago by https://github.com/OpenCPN/OpenCPN/commit/5bfb2c2afcb18148369a6a9ae3a175c430244dfb

nohal, 2024-02-17:
Thanks for the help figuring this out!

closed 2024-02-17 by nohal

rhinotronic, 2024-02-17:
Thanks again.