OpenCPN/OpenCPN #3367: Watchdog Depth

MikeTrader, 2023-07-27:
OpenCPN correctly parsing Depth
Watchdog not

2 years and counting

Arguably this is the most important piece of navigation data, esp at anchor!

I am willing to write a parser for OpenCPN and Watchdog

The repository for Watchdog code

MikeTrader, 2023-07-27:
https://github.com/rgleason/watchdog_pi
does not include dpt,cpp or the header file

rgleason, 2023-07-27:
Mike Trader. I would like to work with you to improve Watchdog parsing of depth alarms, possibly adding a depth below waterline parameter. 

Please Note that watchdog has NOT been updated to handle n2k and that we are missing the rudder angle alarms now, and that we have added back the pypilot alarms, such that my repos can be merged with sean's (but he has not merged) and I have no testing reports on the pypilot alarms. That is the current state of watchdog. Admittedly it needs updates.

I can maintain and fix some things but am not a programmer.  I would suggest using my github repos rgleason/watchdog_pi which is the distribution repos and is most current. I can make a new branch for you to make a PR to.

bdbcat, 2023-07-27:
Mike...
Thanks for bringing the issue to github.
github issues are regularly addressed for ongoing maintenance.  They are not lost.
This upgrade will be implemented as resources allow.
Dave

MikeTrader, 2023-07-27:
I would be happy to help, I need information tho )
I dont have any version of Visual Studio installed any longer, 
I dont know where everything is.

I spent an hour here
rgleason/watchdog_pi 
I could not find dpt,cpp (

Searching turned up nothing either

rgleason, 2023-07-27:
Mike,
In my repos. dpt.cpp and spt.hpp both exist!
You should find them under libs folder!

rgleason, 2023-07-27:
Also, I recall recently testing watchdog alarms while using & running a VDR file with a number of alerts, depth was one of them and it worked. 

What exactly is the problem? Are you actually asking for an enhancement to the plugin?  Also, if you can, upload a vdr recording that fails or has the feature so we can test it.  On the boat right now so can't work on this easily right now.

Mike there is an easy way to get setyp to compile and test. Part of it is in the Dev Manual and the other part is provided by TransmitterDan 's special github repository/branch forbOpenCpn. This may have been added as a branch to the main opencpn repos. I will try to get the links.

rgleason, 2023-07-27:
All the links to the Developer Manual are totally circular and fubar. This should be fixed asap, no wonder Mike cant find it.

rgleason, 2023-07-27:
BTW Sometime ago I tried to use the updated opencpn libs/nmea. In watchdog. It failed so I had to revert back to the library devekoped for watchdog.  It would be nice to get watchdog working with the standard library opencpn uses.

rgleason, 2023-07-27:
Got it
https://opencpn-manuals.github.io/main/opencpn-dev/windows.html

MikeTrader, 2023-07-30:
Perfect thanks, Ill take a look soon

bdbcat, 2023-07-30:
Mike...
I'm on it.  The DBT bug is real, and root cause is a simple logic error in the plugin message parser.
I have corrections in hand, along with some other long-standing bugs.
These will be published in due course, as Rick comes back online.
Thanks

rgleason, 2023-07-31:
> I would be happy to help, I need information tho ) I dont have any version of Visual Studio installed any longer, I dont know where everything is.
> 
> I spent an hour here rgleason/watchdog_pi I could not find dpt,cpp (
> 
> Searching turned up nothing either

https://github.com/rgleason/watchdog_pi/tree/master/libs/nmea0183/src

bdbcat, 2023-07-31:
Rick...
No particular reason to update the NMEA0183 libraries. The existing libs are fine, I think.  Red Herring...

rgleason, 2023-08-03:
@Hakansv    Hakan, Thanks for checking.  Don't you think we should probably add these as RFE under issues?
@MikeTrader  Will you try it and respond please?  Also what OS? Thanks.

-     Feet does not work. Edit menu is in meters, when feet is selected it goes back to meters.
-     Always shows "NA" rather than a comparison between the actual depth and the setting with a greater than or less than sign?
-     DBT Depth below transducer controls. (DPT Depth of Water is lower priority.) How about a filter to turn one or the other off so the user knows which one is being used?

closed 2023-08-03 by rgleason