OpenCPN/OpenCPN #3369: Finding Dev Manual with cell phone

rgleason, 2023-07-27:
**Describe the bug**
Android cell phones do not handle AsciiDr outline table at all well.
Therefore the Dev manual is unreachable.
I think there should be a direct link to the Dev manual in the text or some other effective way to navigate to and through the Dev manual.

Perhaps this is just a deficiency we must live with, but I am finding the links to the dev manual and plugins very circular on this cell phone.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS: [e.g. macOS]
 - Version [e.g. 22]

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: [e.g. Galaxy S20]
 - Andorid version: [e.g. 10]

**Additional context**
Add any other context about the problem here. Screenshots are usually a good way to ilustrate problems, attach them if applicable.

leamas, 2023-08-02:
In what circumstances would a developer actually need to access the Developer Manual using a phone?  Development normally takes place on a regular computer, this is true also for Android development. So: what is the use case?

BTW: Please edit the original comment, it's far from completed.

rgleason, 2023-08-03:
I wanted to provide a user-programmer who wanted to help with programming PI, with a link to instructions to debug OpenCPN in Windows. I was sailing and only had my android phone. I could not find the proper links because the left navigation does not show.

leamas, 2023-08-03:
OK, I see.

Anyway, this is about the developer manual, not core OpenCPN. For that reason I  think this bug should be closed. You might want to open a new bug against the manual at https://github.com/opencpn-manuals/development

rgleason, 2023-08-03:
Good point. Moving.

closed 2023-08-03 by rgleason