OpenCPN/OpenCPN #3372: TP frontend2 jammy builds not working

rgleason, 2023-07-31:
**Describe the bug**
User finds no TP plugins for Ubuntu Jammy.

https://www.cruisersforum.com/forums/showthread.php?p=3806960

Note that most TP FE2 plugins except Jon's are built with github com/rgleason/testplugin_pi. Version 1 0.238
**To Reproduce**
Use jammy on rpi
**Expected behavior**
All the plugins are shown and work

**Desktop (please complete the following information if applicable):**
 - Unbunu Jammy for rpi
 - opencpn 5.8.4

Is there somethinf wrong about the build itself, or the metadata file?

Is the PIM enforcing some filter so these are nor available?

leamas, 2023-07-31:
@rgleason : could you please start  opencpn using `--loglevel debug`, make a clean exit  and attach opencpn.log

bdbcat, 2023-07-31:
Rick...
Jon's latest work for Draw_pi correctly builds Jammy.
Discovered in process of building GPS-Odometer for Jammy.  The problem also appears in AIS-Radar and Statusbar.
https://www.cruisersforum.com/forums/f134/opencpn-version-5-8-4-released-276788-19.html#post3806328

It appears that there is an update to "CMake/PluginSetup.cmake" which solves the problem.

bdbcat, 2023-07-31:
I have made a PR to AIS-Radar_pi as an example of required update.
Root cause:  The metadata must contain:

`<target>ubuntu-wx32-x86_64</target>
 `
instead of

`<target>ubuntu-x86_64</target>
`

bdbcat, 2023-07-31:
Rick..
Will you make the same change to StatusBar, so we can close 
https://www.cruisersforum.com/forums/f134/opencpn-version-5-8-4-released-276788-19.html#post3806328

TwoCanPlugIn, 2023-07-31:
> iI have made a PR to AIS-Radar_pi as an example of required update.

Which repo of AIS-Radar ?

Which plugin is the "correct" version of the truth?

For some time I assumed it was Jon Gough's test plugin only to find that Rick's repo was several commits ahead so I adopted Rick's .circleci/config.yml and ci/build scripts only to find that what should have been flatpak-arm64 was spat out as flatpak-aarch64.

Appreciate the upgrade to wxWidgets 3.2 orry to say but the Plugin

bdbcat, 2023-07-31:
The AIS-Radar plugin used for Production release is:
https://github.com/rgleason/AISradar_pi

In that repo you will see the PR containing required changes for Jammy and Android, at least for AIS-Radar today.
I suspect Jon's TestPlugin is a little bit behind Jon's Draw_pi.  But, with the latest changes, Draw builds and loads on all required platforms.  If TP does not match Draw, that would be an item for Jon.

Sure, next time...
But first, lets just chop the wood and get current production plugins working consistently.
Don't forget to look for changes in cmake directory.  If your plugin is based on FE2, as most of Rick's are, then a simple diff ( I use kdiff3) between Draw_pi and your plugin will be most instructive.

rgleason, 2023-07-31:
> I have made a PR to AIS-Radar_pi as an example of required update. Root cause: For Ubuntu 22.04 (Jammy), the metadata must contain:
> 
> `<target>ubuntu-wx32-x86_64</target> ` instead of
> 
> `<target>ubuntu-x86_64</target> `

Jeez this is too bad. It is complicated but simple. We need to update all the plugins again. I will check out ais radar pr.

My next problem is to use Jon's new version or the one I understand better. I would like to review and understand the changes he has made before adopting it (adopting being my ultimate goal) but for this revision wouldn"t it be better to fix 1.0.238?

Maybe I should just build jammy with a new version so it can go to the prod repos, then push to plugins?

bdbcat, 2023-07-31:
Rick...
Please stand by a little bit.  I think there is still some trouble with AISradar.
Working it now.
We can fix 1.0.238 by updating just a couple of files, I think.  It should not be a wholesale switch to the Draw_pi mode.
Do not push Jammy yet.  It is broken.

rgleason, 2023-07-31:
Ok standing by, now ashore, prepping for renters, etc, and will have some time. Sorry I wasn't following all of this too well.
Thanks!!

bdbcat, 2023-08-01:
Rick...
AISradar_pi  PR is ready now.  Tested on several Ubuntu flavors, including Jammy.
The PR updates 3 files with a few lines of code, and adds no new files. So I think it is reasonable to consider this an update to your 1.0.238.
Recommend that you publish this path to get some initial field feedback before applying similar patches to other plugins based on this same template (e.g.statusbar_pi).

jongough, 2023-08-01:
Have you tried building with FE2 1.0.250.0? This should be the same as used for OD 1.8.28.2 . A fix for jammy.

bdbcat, 2023-08-01:
Jon...
I did the wx32 dependency load in a slightly different manner than you used for OD.  I updated the debian "control" file instead of forcing an "extra" library load of wx32...-dev package on Jammy.  Either method works, though.

Perhaps Rick could compare the two methods, and select whichever makes most sense to him.

rgleason, 2023-08-01:
Merged, ais_radar PR and watchdog PR.  Will push AISradar first. 

Have made some minor fixes to watchdog.fpb file and the menus to fix some legibility issues, seems to be working ok (added the rudder fpb section too).  and have pushed that up to test the builds.

Next thing for me to do is to fix watchdog for jammy. Have seen and looked at AISradar PR and will try to take a look at Jon's further development of TP FE2 1.0.250.0 tomorrow morning. (Thank you Jon.)  My guess is that I will need some time to see and understand Jon's commits and development. I know he's streamlined the builds with some circleci tools, but I think it will be the next cycle where I will start using Jon's updated version.

I will also update statusbar.

rgleason, 2023-08-01:
PS:  I think it is really helpful to have multiple eyes and efforts on the development of FE2, because it helps to move the code forward in an effective way and the result is a better version of FE2. Thank you both for all your efforts.

rgleason, 2023-08-01:
Alec, thanks, will do this because I need to understand how it works, but the first thing I need to do is get bdbcats PR's pushed. 

`@rgleason : could you please start opencpn using --loglevel debug, make a clean exit and attach opencpn.log`

TwoCanPlugIn, 2023-08-01:
> But first, lets just chop the wood and get current production plugins working consistently

Building plugins is like playing "Whack a Mole"

One never knows which plugin and whose repo is the correct version of the truth. AIS Radar, Draw, TestPlugin, what next?

rgleason, 2023-08-01:
Listen to me Steve. 

Just look at the dates and the PR's at Sean's site you you'll realize that he's got his mind elsewhere mostly. Additionally for a more complete picture refer to the Issues under Jon's repository for FE2.
I have maintained this for almost 7 years now as I have to have a list for reference. This is generally the current distribution and deployment of most of the TP FE2 based plugins. It is called [TP Plugins List](https://github.com/jongough/testplugin_pi/issues/177)   This list changes on a regular basis.  

Generally the testplugin version to use is Jon's most current version. Right now I am using my rgleason version 1.0.238.1 because I had to make some changes and improvements while Jon was away. I have not followed and caught up to Jon's version yet because I was then away sailing.  These changes have been posted and it is not nearly as "wackamo" as you think.

rgleason, 2023-08-01:
Dave wrote:
```
Jon...
I did the wx32 dependency load in a slightly different manner than you used for OD. I updated the debian "control" file instead of forcing an "extra" library load of wx32...-dev package on Jammy. Either method works, though.
```

I like the changes in the control file. It appears that android is building more frequently now too.  AISradar and Watchdog are working now.

I don't know if the clever renaming of metadata for jammy in cmake/pluginsetup.cmake near line 121 is easy to find when the need for Jammy disappears, but this is pretty efficient.  Jon may have some ideas about this.

Also there were some cmake/pluginpackage.cmake changes at line 94 for android.  Since android will be around for quite awhile having them here makes a good deal of sense.

rgleason, 2023-08-01:
Dave, regarding the libs/nmea0183  comparison between OpenCPN and Watchdog. There are some big differences, many files are not shared in both, and the code has improvements and differences between a number of files.  I used Winmerge on the two sources and really did not know where to start and there are so many differences it is not worth trying to resolve it right now. I believe I tried a wholesale copy of the OpenCPN version and it would not build, so I am going to leave it as it is for now.

rgleason, 2023-08-01:
PS the final Watchdog is building now and going to -prod and will be pushed shortly. 
Statusbar is next.

bdbcat, 2023-08-01:
"...when the need for Jammy disappears..."
This will not happen for some years.  As an LTS Release, Ubuntu 22.04 can be expected to be in wide use for at least 5 years.  It is just an oddball special case, caused by misalignment between debian and ubuntu release cycles.  This could happen again, though AFAICT it is not expected at least for Ubuntu 24.04 coming next year.  24.04 should be fully compliant with debian bookworm, so the plugin builds for Bookworm will work.

rgleason, 2023-08-01:
Dave,
Just found some problems with depth
1. Edit menu is in meters, when feet is selected it goes back to meters.
2. I haven't been able to get it to trigger, but perhaps I haven't found a nmea file with depth.

I am going to push this anyway as there are other improvements.

rgleason, 2023-08-01:
1. Feet does not work.  Edit menu is in meters, when feet is selected it goes back to meters.
2. Always shows "NA" rather than a comparison between the actual depth and the setting with a greater than or less than sign?
3. DBT Depth below transducer controls.   (DPT Depth of Water is lower priority.)  How about a filter to turn one or the other off so the user knows which one is being used?

[statusbar 1.0.7.0 fix jammy #940 ](https://github.com/OpenCPN/plugins/pull/940)  pr to plugins

bdbcat, 2023-08-01:
These are good topics for bug reports or RFE to Watchdog plugin..
Let's close this Jammy issue and try to stabilize the FE2 build matrix.

leamas, 2023-08-03:
> Dave, regarding the libs/nmea0183 comparison between OpenCPN and Watchdog. There are some big differences, many files are not shared in both, and the code has improvements and differences between a number of files

This is the reason there is the https://github.com/leamas/opencpn-libs/ project. In the shipdriver template this is used as a git submodule, so it's kind of automatically updated. But even if not used as a submodule it could be an alternative baseline.  

It contains nmea0183 which is used as-is in all shipdriver plugins.

rgleason, 2023-08-03:
Thanks for reminding me. May try it.