OpenCPN/OpenCPN #3374: Disabled system plugins not visible in PM

leamas, 2023-08-02:
**Describe the bug**
After  #3103,  @transmitterdan  reports that disabled system plugins are not visible in the plugin manager and thus not possible to enable. See [3103 comment](https://github.com/OpenCPN/OpenCPN/pull/3258#issuecomment-1660884364)

**To Reproduce**
Steps to reproduce the behavior:
1. Make sure at least one system plugin is disabled
2. start OpenCPN, invoke Options | Plugins
3. The disabled system plugin(s) is not visible 

**Expected behavior**
Well...

**Desktop (please complete the following information if applicable):**
 - OS:  At least windows and Linux, probably all.
 - master

leamas, 2023-08-02:
This seems to be about  where `make install` places system plugins vs where the plugin manager looks for them in the development environment. This should not affect fully installed opencpn packages. These installs in the proper location and should thus be fine.

On linux, the plugin manager looks in the directories _~/.local/lib/opencpn_ and _/usr/local/lib/opencpn_. However, with a plain `cmake ..`  the `make install` command will place the plugins in  _./lib/opencpn_ . This leads to that the plugin manager does not load the system plugins when disabled.

Then again, on my linux system this can be worked around using `-DCMAKE_BUILD_TYPE=Debug  -DCMAKE_INSTALL_PREFIX=/usr/local -DCMAKE_INSTALL_LIBDIR=lib .. `. This makes `make install` place the plugins in _/usr/local/lib/opencpn_ and everything is fine.

The locations searched for plugins can be found in opencpn.log after making a opencpn run using `--loglevel debug`.  Grep for  "PluginLoader: loading plugins from" which reveals a search path.

@transmitterdan : Thoughts?

leamas, 2023-08-02:
Windows is different. Here, it seems like the system plugins are not recognized as such. I'm on it.

leamas, 2023-08-02:
Another development issue: I have been testing with the experimental ocpn-plugins.xml attached to #3103.  This is behind some of your windows problems, I think. Will make a PR to make things work also with the current catalog. 

Also need to make a catalog PR with the new metadata...

leamas, 2023-08-02:
An updated catalog with system plugins metadata is now available, see https://github.com/OpenCPN/plugins/pull/941. 

Will keep this in beta until we have verified that this new metadata does not disturb older versions which are not prepared to see it, so to speak.

leamas, 2023-08-02:
The Linux issues are still there. However, I would not  really call it a bug, it's basically how things work. Might need a note in the manual, but besides that I don't see any major problems.

Unless there is more input I will close this bug in a few days.

transmitterdan, 2023-08-03:
The chart downloaded still not showing up in the Chart tab of Options dialog.

leamas, 2023-08-03:
That could possibly not be related to this, can it? As long as the plugin is loaded and available?

leamas, 2023-08-03:
Yes, it can. Reverting d2ce0c583d7320828d2266a9cb93ce0930f21efb seems to restore things. Ack?

leamas, 2023-08-03:
Back to square one: Unless there is more input I will close this bug in some days.

transmitterdan, 2023-08-03:
> Back to square one: Unless there is more input I will close this bug in some days.

This fixes the chart downloader tab issue.  Everything seems back to normal now.

transmitterdan, 2023-08-03:
There is still a problem but don't know if it's related to this PR.  If I run a "normal" install in Windows and try to 'reinstall' the O-Charts plugin the Charts tabs get overpopulated with Chart Downloader and O-Charts tabs. To wit:
[image]

transmitterdan, 2023-08-03:
I guess that plugin manager is initializing the plugins several times and we have no check that if a plugin has added it tab to the Options dialog then we only allow that once.

leamas, 2023-08-03:
Reinstall is basically uninstall + install. So the question is if the plugin cleans up everything as it should when uninstalled...

Another qustion is if this happens also before this large merge, I used  7e0cff73f as a point before the merge.

transmitterdan, 2023-08-03:
@leamas This issue is beyond my ability to debug.  When installing a plugin from the catalog, the new plugin is apparently loaded multiple times.  At least the loader calls ```void PluginLoader::NotifySetupOptionsPlugin(const PlugInData* pd)``` multiple times which is how the multiple tabs are instantiated.

I can't bisect to find when this came into being because debugging on Windows is too painful prior to the new local building process.  But it also happens on Linux.  Tonight I'll try to bisect on Linux and see where it started.

leamas, 2023-08-03:
I'm suspicious about o-charts: https://github.com/bdbcat/o-charts_pi/issues/40

leamas, 2023-08-04:
hm... It seems that this patch sorts things out for me on Windows. Thoughts? 
```
commit 94cfffede41ab31c2cb67fbd9e694ecc13dfc10f (HEAD -> master)
Author: Alec Leamas <leamas.alec@nowhere.net>
Date:   Fri Aug 4 16:17:48 2023 +0200

    Only run notify_setup_options() once.

diff --git a/include/plugin_loader.h b/include/plugin_loader.h
index c581bcb28..5f2a5aa7c 100644
--- a/include/plugin_loader.h
+++ b/include/plugin_loader.h
@@ -68,6 +68,7 @@ public:
 
   PlugInData();
 
+  bool m_has_setup_options;          //!< Has run NotifySetupOptionsPlugin()
   bool m_enabled;
   bool m_init_state;
   bool m_toolbox_panel;
diff --git a/src/plugin_loader.cpp b/src/plugin_loader.cpp
index 9bff0695c..bdcdad249 100644
--- a/src/plugin_loader.cpp
+++ b/src/plugin_loader.cpp
@@ -156,7 +156,8 @@ PlugInContainer::PlugInContainer()
     : PlugInData(), m_pplugin(nullptr), m_library(), m_destroy_fn(nullptr) {}
 
 PlugInData::PlugInData()
-    : m_enabled(false),
+    : m_has_setup_options(false),
+      m_enabled(false),
       m_init_state(false),
       m_toolbox_panel(false),
       m_cap_flag(0),
@@ -288,6 +289,8 @@ void PluginLoader::ShowPreferencesDialog(const PlugInData& pd,
 void PluginLoader::NotifySetupOptionsPlugin(const PlugInData* pd) {
   auto pic = GetContainer(*pd, *GetPlugInArray());
   if (!pic) return;
+  if (pic->m_has_setup_options) return;
+  pic->m_has_setup_options = true;
   if (pic->m_enabled && pic->m_init_state) {
     if (pic->m_cap_flag & INSTALLS_TOOLBOX_PAGE) {
       switch (pic->m_api_version) {

```

transmitterdan, 2023-08-04:
I'll test it. Looks like it will work.

transmitterdan, 2023-08-05:
The patch works for me.

Hakansv, 2023-08-05:
If related to this I don't know. Been out and have not followed all issues/commits. I can make an own issue if desired.

I've a O crash when the Plugins tab is redrawn while e.g. disable a plugin or update catalog master.
Both Win and RPi (Debian)
Note, FWIW; A lot AIS targets involved. (See \OpenCPN\src\options.cpp : 1630)
On RPi:
```
(opencpn:1125669): Gtk-CRITICAL **: 11:08:57.266: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar
Segmentation fault
```
On Win debug:
```
..\OpenCPN\src\options.cpp : 1633

 // with AIS it's called very often
 #if wxCHECK_VERSION(3, 0, 0)
 bool options::SendIdleEvents(wxIdleEvent& event) {
->   if (IsShown()) return wxDialog::SendIdleEvents(event);   Exception thrown at 0x66E331AF (wxmsw32u_core_vc14x.dll) in opencpn.exe: 0xC0000005: Access violation reading location 0x65819DA4.
   return false;
 }
 #endif
 
 call stack:
 
 	wxmsw32u_core_vc14x.dll!66e331af()	Unknown
 	[Frames below may be incorrect and/or missing, no symbols loaded for wxmsw32u_core_vc14x.dll]	
 	[External Code]	
>	opencpn.exe!options::SendIdleEvents(wxIdleEvent & event) Line 1633	C++
 	[External Code]	
 	opencpn.exe!PluginLoader::NotifySetupOptionsPlugin(const PlugInData * pd) Line 309	C++
 	opencpn.exe!NotifySetupOptionsPlugin(const PlugInData * pd) Line 1704	C++
 	opencpn.exe!PluginPanel::SetEnabled(bool enabled) Line 4793	C++
 	opencpn.exe!PluginPanel::SetSelected(bool selected) Line 4722	C++
 	opencpn.exe!PluginPanel::PluginPanel(wxPanel * parent, int id, const wxPoint & pos, const wxSize & size, const PlugInData plugin) Line 4507	C++
 	opencpn.exe!PluginListPanel::AddPlugin(const PlugInData & pic) Line 4122	C++
 	opencpn.exe!PluginListPanel::ReloadPluginPanels() Line 4111	C++
 	opencpn.exe!PlugInManager::HandlePluginLoaderEvents::__l2::<lambda>(wxCommandEvent & __formal) Line 1063	C++
 	opencpn.exe!wxEventFunctorFunctor<wxEventTypeTag<wxCommandEvent>,void <lambda>(wxCommandEvent &)>::operator()(wxEvtHandler * __formal, wxEvent & event) Line 548	C++
 	[External Code]	
 	opencpn.exe!MyFrame::DoOptionsDialog() Line 3970	C++
 	opencpn.exe!MyFrame::DoSettings() Line 2814	C++
 	opencpn.exe!MyFrame::OnToolLeftClick(wxCommandEvent & event) Line 2678	C++
 	[External Code]	
 	opencpn.exe!WinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, char * lpCmdLine, int nCmdShow) Line 930	C++
```

leamas, 2023-08-05:
@Hakansv : The way to check if this has something to do with your crash is making a git bisect. I used  7e0cff73f as a starting point before this big merge. If it does not happen on that commit, the merge might be involved, and a full bisect would be extremely useful. If not, it's probably something else.

BTW: That  `#if wxCHECK_VERSION(3, 0, 0)` is redundant and should be removed, we don't support anything less than wxw 3.0.0

Hakansv, 2023-08-05:
Since commit 730e7495 I can't reproduce the crash reported above.
The "if wxCHECK_VERSION(3, 0, 0)" may be obsolete but was not the actual trouble.

There's is though another issue. The installed plugin version is not correct. The version from an available update is printed instead of what's installed. I have some plugins not updated since long and all shows the same pattern. See pict:
[image]

leamas, 2023-08-07:
> Since commit https://github.com/OpenCPN/OpenCPN/commit/730e7495e6482611483e09f1cda445ac9489b4e1 I can't reproduce the crash reported above.

... which means that a bisect pinning down the exact commit causing this would be extremely helpful.

leamas, 2023-08-07:
> The version from an available update is printed instead of what's installed. 

Hm... yes, this is sort of "on purpose", but it obviously does not work with old plugins from the catalog which are are not updated. The  current alghorithm is in plugin_loader.cpp around line 161, `PluginLoader::GetPluginVersion()`. The primary goal is to show the extended  version, not just major+minor which is available in the plugin.

1. If the plugin implements API level 117 or higher, the complete version data is taken form the plugin.
2. Otherwise, if there is available metadata this is used (this is what's happening for you).
3. Otherwise, just use major+minor from plugin.

I wonder if we limiting the metadata usage to metadata where major + minor matches the plugin major + minor is safe? Need to sleep on it.

rgleason, 2023-08-09:
I havent followed this completely, but many plugins just have two parameters defined through the API.
Perhaps this is part of the problem?

```
int aisradar_pi::GetAPIVersionMajor() {
    return OCPN_API_VERSION_MAJOR;
}

int aisradar_pi::GetAPIVersionMinor() {
    return OCPN_API_VERSION_MINOR;
}

int aisradar_pi::GetPlugInVersionMajor() {
    return PLUGIN_VERSION_MAJOR;
}

int aisradar_pi::GetPlugInVersionMinor() {
    return PLUGIN_VERSION_MINOR;
}

```

leamas, 2023-08-30:
This has now morphed into a support issue. Closing

leamas, 2023-08-30:
> Should we be adding a third API parameter to all plugins?

This is about the API level. From API 1.18 there are multiple more version parts available. 

Updating to 1.18 thus requires plugin to update these new methods. For shipdriver, this is documented in https://github.com/Rasbats/shipdriver_pi/blob/master/UPDATE_PLUGIN_API.md

closed 2023-08-30 by leamas