OpenCPN/OpenCPN #3381: SignalK Listener

TwoCanPlugIn, 2023-08-04:
There are three functions defined on ocpn_plugins.h that use the new Observer class.

`extern DECL_EXP std::shared_ptr<ObservableListener> GetListener(NMEA2000Id id, wxEventType ev, wxEvtHandler *handler);`
`extern DECL_EXP std::shared_ptr<ObservableListener> GetListener(NMEA0183Id id, wxEventType ev, wxEvtHandler *handler);`
`extern DECL_EXP std::shared_ptr<ObservableListener> GetListener(SignalkId id, wxEventType ev, wxEvtHandler *handler);`

but only two functions for retrieving the payload.

`extern DECL_EXP std::vector<uint8_t> GetN2000Payload(NMEA2000Id id, ObservedEvt ev);`
`extern DECL_EXP std::string GetN0183Payload(NMEA0183Id id, ObservedEvt ev);`

Seems to be no function for retrieving the SignalK payload.

Am I overlooking something, is this an omission, or not yet implemented ?

leamas, 2023-08-04:
Indeed.  What's more is that there is no signalK driver either. We have some homework to do.

leamas, 2023-08-04:
So, just about  something like this:
```
diff --git a/include/ocpn_plugin.h b/include/ocpn_plugin.h
index c12772aed..f0192ae7d 100644
--- a/include/ocpn_plugin.h
+++ b/include/ocpn_plugin.h
@@ -1668,6 +1668,9 @@ struct SignalkId {
 extern DECL_EXP std::shared_ptr<ObservableListener> GetListener(
     SignalkId id, wxEventType ev, wxEvtHandler *handler);
 
+/** Return context JSON data for a SignalK message. */
+extern DECL_EXP std::string GetSignalkPayload(ObservedEvt ev);
+
 /** Return payload in a received n2000 message of type id in ev. */
 extern DECL_EXP std::vector<uint8_t> GetN2000Payload(NMEA2000Id id,
                                                      ObservedEvt ev);
diff --git a/src/comm_plugin_api.cpp b/src/comm_plugin_api.cpp
index 19a2c2d6c..854c88f9e 100644
--- a/src/comm_plugin_api.cpp
+++ b/src/comm_plugin_api.cpp
@@ -49,6 +49,12 @@ std::string GetN0183Payload(NMEA0183Id id, ObservedEvt ev) {
   return msg->payload;
 }
 
+std::string GetSignalkPayload(ObservedEvt ev) {
+  auto msg = UnpackEvtPointer<SignalkMsg>(ev);
+  return msg->context;
+}
+
+
 shared_ptr<ObservableListener> GetListener(NMEA2000Id id, wxEventType et,
                                            wxEvtHandler* eh) {
   return make_shared<ObservableListener>(Nmea2000Msg(id.id), eh,

```

which is easily added to master.

@TwoCanPlugIn: same question: How important is this for you right now?

TwoCanPlugIn, 2023-08-04:
@leamas Not super important.

Having hopefully fixed the circleci build issues for the Engine Dashbord plugin, in which I had implemented the NMEA 2000 listener so that it could support the NMEA 2000 support built-in to OpenCPN 5.8.x, I thought I would add support for both the NMEA 1083 and SignalK observer classes. The NMEA 183 listener works fine, but as I remarked there was no GetPayload function for SignalK.

In anycase, in retesting my code against the latest SignalK Server (v 2.1.0), I've noticed something weird in the vessel.propulsion namespace that has broken the existing SignalK parsing in the Engine Dashboard,.

So until I resolve that, it's no biggie. What's kind of funny is that no one has reported the problem, so it seems no-one is using tthe Engine Dashboard plugin against SignalK.

leamas, 2023-08-04:
@TwoCanPlugIn . OK. We will try to fix this to next release then. Thanks for reporting!

leamas, 2023-08-30:
hm... it might be better to parse the message and deliver a wxJSON object. Any listener to  this message will have to parse the json payload, and if there are multiple listeners it's better to centralize the parsing to one point.

Also, handling a wxJSOn object creates a useful typing.

antipole2, 2023-10-14:
> hm... it might be better to parse the message and deliver a wxJSON object. Any listener to this message will have to parse the json payload, and if there are multiple listeners it's better to centralize the parsing to one point.
> 
> Also, handling a wxJSOn object creates a useful typing.

For the JavaScript plugin use of this, I _strongly_ prefer to receive the JSON string. Parsing JSON in JavaScript is trivial.
````
mySignalKobject = JSON.parse(theJSONstring);
````
If it is decided to go down the wxJSON route, please provide getting the JSON string as an alternative.

leamas, 2023-10-14:
Well, parsing centrally would just mean you could do something like  `mySignalKobject = GetSignalkPayload(ev);` which is trivial as well.   And, as I said above: 
  - This adds the benefit that we only parse once even if there are multiple listeners to this data.`
  - Having a return type like `wxJSONValue`  adds useful typing compared to a plain string.

antipole2, 2023-10-14:
Yes, I can write that in C++ in the JavaScript plugin API, which gets me the object in C++.  But I would then have to construct an equivalent object in the Duktape context stack for return to the user's JavaScript script.  Not sure how that could be done but I guess it would be possible.  But it will be quite a lot of code to write and maintain.  The return type is lost to the script as JavaScript is typeless in its assignments.
I think it would be straightforward to provide access to the JSON string at least as an alternative.  For NMEA0183 the implementation is:
```
std::string GetN0183Payload(NMEA0183Id id, ObservedEvt ev) {
  auto msg = UnpackEvtPointer<Nmea0183Msg>(ev);
  return msg->payload;
}
```
Note the id is not even used.  So wondering whether this might work even as is to get the JSON string?  I am not able to try it myself.  In any event \<SignalkMsg> and \<wxJSONValue> are already defined if needed.

Maybe something like the following would be useful for those who need it not just for SignalK but for strings in general.
```
std::string GetStringPayload(ObservedEvt ev) {
  auto msg = UnpackEvtPointer<std::string>(ev);
  return msg->payload;
}
```
Or maybe it is more complicated than I naively assume.

leamas, 2023-10-14:
hm... The javascript plugin is kind of a corner case, and while we should not make things really hard for it we OTOH should not really design for it, I guess

> Not sure how that could be done but I guess it would be possible. But it will be quite a lot of code to write and maintain.

Perhaps not?
```
  wxJSONWriter writer( wxJSONWRITER_STYLED| wxJSONWRITER_WRITE_COMMENTS);
  wxString     str;
  writer.Write(root, str );
```

leamas, 2023-10-14:
>  So wondering whether this might work even as is to get the JSON string? I am not able to try it myself

The basic problem handled by the various functions in _comm_plugin_api.cpp_ is mapping between the internal message types used by core opencpn and the simplified types available in the API.  That's ther reason you cannot test this in plugin code: the needed declarations are not available.

And they should not be, they far to comple for the plugin API. Still, implementation are simple. The problem is how to design an API, actually trying to do this here.

bdbcat, 2023-10-14:
FWWI, I prefer passing a simple string in the API.
We have JSON parsing module in the new(ish) common plugin support library, so it is now possible to avoid duplication of parser code itself in the plugin, at least for simple c++ use cases.

leamas, 2023-10-14:
Yes, this avoids source duplication. But my argument is more about runtime duplication, that several plugins parses the same message.

bdbcat, 2023-10-14:
Reasonable.
Code?

leamas, 2023-10-14:
I'm on it, later today or tomorrow

leamas, 2023-10-14:
Two open issues:
- Since we dont include the wxJSON headers in ocpn_plugin.h we cannot return a wxJSONValue. Needs some thought.
- Error handling: Possible errors needs to be incoded in a fixed json message structure.

leamas, 2023-10-14:
Filed  #3460. For now, please consider this as a sketch. All sorts of input is more than welcome!

antipole2, 2023-10-15:
What about sending SignalK data?  I had assumed we could use WriteCommDriver as the comment says it can be used "on bus protocols other than NMEA2000'"
But if we go down the route of getting a wxJSONValue, should we also be writing a wxJSON Value for consistency?

Slightly off topic...  the three getters all include an id as well as the ev but none appear to use the id.  Is the id no longer needed?  Do we need to supply a meaningful one?

leamas, 2023-10-15:
> What about sending SignalK data? 

I hav'nt really considered this. However, the main argument for  parsing centrally is about not parsing the same message multiple times, and this does not apply when sending. So in this case I don't see using wxJSON instead of a string as meaningful.

>  the three getters all include an id as well as the ev but none appear to use the id.

In the SignalK case we transmit a complete state, so the id is indeed not used here. It could be used in the future if we are are able to send the actual state change instead of the complete state.

In the others the id is kept as a possibility to double-check that the received data matches what's expected. It should be correct as long as we listen to the  expected data, but FWIW my gut feeling is that this does not hurt.

TwoCanPlugIn, 2023-10-17:
I haven't followed the recent discussions thus far.

Simple question, from a plugin developer perspective, what difference, if any will there be in receiving a SignalK payload using (I guess, as proposed) 
`std::string GetStringPayload(ObservedEvt ev)`

versus how we do it today using the OCPN plugin messaging ?

In all honesty, other than only receiving a NMEA sentence with the requested NMEA183 sentence ID, NMEA 183 listeners don't add any further value; the plugin still has to parse the sentence.

Re, writing SignalK, just curious what the use case is? It's not as if a client will expect to receive an unsolicited SignalK message without having previously subscribed or issued a query, and it's not as if OpenCPN provides all of the Service Discovery and Web server functions normally associated with SignalK?

leamas, 2023-10-17:
The proposed way to receive SignalK  data is rather something like

    auto ptr = GetSignalkPayload(ObservedEvt ev);
    auto msg = UnpackEvtPtr<wxJSONValue>(ptr);

The msg structure is  a pointer to a json map containing something like this (pasted from ocpn_plugin.h in #3460), the context stuff is to be added.
```
 *  - "Data": the parsed json message
 *  - "ErrorCount": int, the number of parsing errors
 *  - "WarningCount": int, the number of parsing warnings
 *  - "Errors": list of error messages i. e., strings.
 *  - "Warnings": list of error warnings i. e., strings.
```

The two-step procedure is about avoiding referencing wxJSONValue in ocpn_plugin.h. This header is designed to be self-contained and does not include other ones like json_val.h. We are basically using a ref-counted void*.

However, this is till a WIP. To repeat, the overall benefit compared to just using the string is to avoid that multiple plugins parses the same message.

leamas, 2023-10-17:
For better or worse this path has a side effect: it is not possible to access the signalK json data using indexes like in `msg["data"]`. This is about general C++ semantics: the [] operator is not const since it will create data in the dictionary if it does not exist. 

Using for for example `msg.ItemAt("data")` works fine, returning an undefined value if  the index does not exist. Overall, I think these semantics are good since they allow us to use read-only data. If there is a need to modify it one need to make a deep copy, though. I consider this a corner case.

leamas, 2023-10-17:
Closed by #3460. Please file new issues if need be.

closed 2023-10-17 by leamas

leamas, 2023-10-17:
A new chaper is to be added to the manual at https://opencpn-manuals.github.io/main/opencpn-dev/plugin-messaging.html.  Should be rebuilt tomorrow.