OpenCPN/OpenCPN #3387: Git master: Win; No SignalK data received

Hakansv, 2023-08-06:
On Windows:
SignalK data is streaming and O starts to connect but no data is received/parsed.
In debug I can see the web socket thread is started but I can't find further errors. (Lack of knowledge)

I checked out commit a0ab5e427fb and the Signal K connection is working as usual.

When checkout commit e3a2185b330 no data is received.

Hakansv, 2023-08-07:
More info..
This was added in commit 932814c2c65.
```
+        // We cannot use select() on sockets with fd >= 1024.
+        if (sockfd > 500){
+          closesocket(sockfd);
+          sockfd = INVALID_SOCKET;
+          sleep(5);
+          freeaddrinfo(result);
+          return sockfd;
+        }
```
To just try to locate where the SK connection is closed I looked there. In the following I'm not sure what I'm doing but it's an observation:
The "sockfd" is every time > 500 so the socket is closed. (Debug values: 1816, 1928, 2316, 1928, 1192, 1928)
For a test I changed "if (sockfd > 500){" to "if (sockfd > 2400){"
Then the SK connection is working as usual!

leamas, 2023-08-07:
Seems to be about POSIX/linux vs Windows/WInAPI.  

Linux select manpage: https://man7.org/linux/man-pages/man2/select.2.html
WinAPI select docs: https://learn.microsoft.com/en-us/windows/win32/api/winsock2/nf-winsock2-select

Key excerpt from linux manpage:

```
 WARNING:  select()  can  monitor only file descriptors numbers that are less than FD_SETSIZE (1024)—an
 unreasonably low limit for many modern applications—and this limitation will not change.   All  modern
 applications should instead use poll(2) or epoll(7), which do not suffer this limitation.
```

There is no sign  of any limitation on fds in the windows case. As for linux/MacOS easywsclient should really use poll(2) instead. I have done this work long time ago on lirc, it was sort of interesting...

The dirty fix would be to only make this check on POSIX platforms

bdbcat, 2023-08-07:
"The dirty fix would be to only make this check on POSIX platforms"
I think that is the practical workaround.  That patch was added specifically for Android, which has some hard-to-debug occasional runaway of socket creation.  With the patch in place. the 5 second spin allows the kernel to catch up with socket deletion and re-use.
Lets do that now, (#ifndef  _WIN32) and move to implement using poll(2) as a TODO.

Hakansv, 2023-08-07:
The #ifndef _WIN32 .. is what I'm using as a "dirty fix" at home now. Why not implement that now, to avoid further tester's issues, and TODO a more sophisticated solution as next step?

bdbcat, 2023-08-07:
Yep, I'll do that now

closed 2023-08-07 by Hakansv

leamas, 2023-08-07:
> and move to implement using poll(2) as a TODO.

Not a TODO in the sense we should just moidify our lib.  If we should stick to easywsclient  this would be about filing a bug upstream, and possibly make a PR against them.