OpenCPN/OpenCPN #3391: Data overflow handling causes  missed fixes

leamas, 2023-08-07:
**Describe the bug**
If the rate of incoming data (GPS fixes) is higher the output channels system stops displayin positions.

This is on a raspberry combined with a Waveshare GPS hat. Basic configuration:
1. One input serial port ttyS0 from the GPS hat, 9600 baud
2. One output channel for NMEA instruments, 4800 baud
3. One output UDP channel over the network, most likely irrelevant.

**To Reproduce**
Steps to reproduce the behavior:
1. Start system above
4. On a native package, after less than a minute the NMEA instruments stops displaying data. Using flatpak, positions and speed starts sort of "come and go" but is basically unusable after  a somewhat  longer time (5 minutes?)

**Expected behavior**
Data should be displayed on the NMEA instruments and/or a relevant error message should be displayed

**Additional context**
Problem can be worked around using output filtering so that only relevant data is forwarded to the slow 4800 baud NMEA channel.

Although this configuration might be considered somewhat odd, increased use of NMEA2000 might lead to similar situations if trying to forward data from NMEA2000 inputs to NMEA0183 output channels.

The correct actions here would be to buffer data in a better way. This would make it possible to detect the situation  and provide a meaningful error message. It could also make it possible to use the correct strategy to just ditch data which "didn't make it" to the output channel and use the latest instead.

bdbcat, 2023-08-07:
First effort should be to define what we want to do when the input "bucket" overflows due to slow flow on the output.
It is a fundamental question of policy.  No amount of buffering will help until we decide what incoming data to throw away when the buffer nears a high-water mark.
First thought:  Discard any data that does not contain position or velocity information, and allow the buffer to drain down to a safe "low-water" mark.  Then re-enable all data.  Rinse and repeat.
Other ideas?

bdbcat, 2023-08-07:
High-water mark:  80% buffer size.
Low-water mark:  20% buffer size.

bdbcat, 2023-08-07:
Simple observation:
A 10 Hz GPS receiver data flow will likely overrun a 4800 baud connection.  So discarding anything that does not contain position/velocity (ie. discard everything except RMC message) probably will not help in this case.

TwoCanPlugIn, 2023-08-07:
Note that most NMEA 2000 devices have data rates far in excess of NMEA 183 devices  operating at 4800 baud.

It's common to have NMEA 2000 devices operating at 10Hz, whereas most NMEA 183 devices operate at 1Hz.

Simple approach is to only send every tenth converted sentence or to use a 1Hz timer to construct and send the sentence.

Hakansv, 2023-08-07:
Steven's note does comply with IEC 61162-1 using RS232 @ 4800 and 1 Hz frequency.
Also IEC 61162-2 deals with 1 Hz data update but baud rate 38 400.
```
targets/second = baud rate / bit per sentence * targets per sentence
38 400 /[22+90/6*4)*10] = 46 sentences / second (4 targets per sentence) = 187 targets / second
```
So OCPN may not make efforts to support devices not following IEC 61162-1  @4800?
I've seen new GNSS receivers (puck) instead using baud rate 115 000. Probably to been able to use the extensive GSV handling when multiple GNSS system are covered.

leamas, 2023-08-07:
I think simple buffer alghorithms will work better than schemes based on for example just forwarding every tenth message or so. Simplicity wins in the long run

> High-water mark: 80% buffer size.
> Low-water mark: 20% buffer size.

Another idea: lets say that next message to send is GGA. We could then check buffer if there is more GGA messages buffered. If so, we have a overrun condition.  The overrurn condition is cleared by removing all buffered data up to the last buffered GGA, after which  this last GGA message is sent. 

There is nothing special with the GGA message here, the same alghoritm would be repeated for any message about to be sent. The advantage of this is that it goes straight onto the core task: always send the last message available, there is no point sending outdated data.

bdbcat, 2023-08-07:
Not exactly simple, since we will need to create a map of received messages, and keep running track of how many of each message is in the buffer.  Or scan the complete buffer on each input message, looking for stale messages to discard.
A nice little coding problem, good interview question. :)

leamas, 2023-08-07:
Well, the alghoritm is not that complicated, assuming messages inserted at back and consumed (send) from front.
1. Look at front to hget next message. Let's again asume it's GGA.
2. Search for GGA from back to front, stop at first match
3. If this match is the same as the one at 1. there is no overrun. Send the item and done
4. If these are different, send the item in 2. and scrub all GGA messages from buffer.

All this could be done using the heavily optimized standard library. However, it will require a mutex to handle concurrency.

bdbcat, 2023-08-07:
What library would you suggest?

leamas, 2023-08-07:
Just std::vector, std::find, etc.

bdbcat, 2023-08-07:
OK. Sounds like a reasonable approach.  Mostly like my second option of dealing with the duplicates on receipt, scrubbing and dropping as necessary.  Maybe this algorithm should kick in when a high water mark is reached, since this bug has not been  widely reported.
As time permits...

Rodemfr, 2024-01-02:
In aeronautics we see similar issues in planes networks where you can have data produced (far) faster than they are consumed (e.g. inertial unit). To solve these cases, we distinguish two types of messages : *sampled* messages and *queued* messages.
Sampled messages are the ones for which a new version of the message makes the previous one obsolete (e.g. position).
Queued messages are the ones which must not be discarded because they are part of a wider transmission (e.g. flight plan) for which loosing a message is an error.

If we try to apply this concept to NMEA0183, messages like GGA/RMC are definitely *sampled* messages and only the latest version is interesting us.

Extending leamas' idea above would lead to something like that : 
- Offer the possibility for a user to list the NMEA "sampled" sentences for which only the latest version is interesting (GGA, RMC, etc.)
- When these sentences are pushed to the buffer : check if an older version exists. If yes, overwrite it. Else, add the sentence to the buffer.

This way, handling 10Hz GNSS GGA/RMC/GLL is not an issue, we just have to tag them as "sampled" sentences, or something like that.

Note that it is not solving my problem (#3572) which still need a buffer increase.

leamas, 2024-01-02:
Is there any NMEA0183 message which should/could be considered "queued"? My thinking so far has been that all of them are "sampled".

Rodemfr, 2024-01-02:
I don't know every NMEA message but I think also that most of them are "sampled" if not all. However their importance is unequal. Loosing a GGA message sent at 10Hz is not a big deal while loosing VDM/VDO is much more problematic since you might endanger your ability to detect collisions for the next minutes. You can also miss static vessel description which are not sent often.
So that's why the idea of having some sort of intermediate filtering in which you declare that some high frequency sentences can be limited to the latest received one in the queue is much more flexible than an ON/OFF filtering. It is also not too complex to understand by the user.

leamas, 2024-01-02:
If we just follow  "my" simple algorithm  it will automatically filter anything which is repeated at a high rate in an aggressive way. 

I have been thinking exactly about the scenario you describe; in it's most simple form it is about one message which comes at a very  high rate and another, important one which comes just occasionally. The algorithm should handle this just fine without any other filtering.  

The basic limitation is the number of message types  vs the buffer size. As long as the buffer can keep  one (two?) messages of each kind no message types  should be completely lost.

And if this limitation is indeed an issue, I guess current filtering works just fine.

That said, we need to create good unit tests for this to iron out the details. Actually, the unit tests should perhaps be the first part of an implementation. Test driven development could shine in a case like this..

Rodemfr, 2024-01-03:
[This](https://github.com/OpenCPN/OpenCPN/issues/3391#issuecomment-1668545328) algorithm is perfect for GGA/GLL, just keeping the last sentence, but for AIVDM/AIVDO this is much more problematic because we don't know the nature and importance of the AIS message inside the payload. If we filter VDM the same way than GGA, we might loose important other VDM messages in case of burst reception of AIS data.

All of this make me think about another possible way to filter NMEA data : a per sentence rate limiter.
You give a time window (e.g; 1 second) and a maximum number of sentence you allow in this window. Every sentence exceeding its maximum rate would be dropped.

By default, you disable the filtering for all sentences. But then you give the possibility to define and enable it for each sentence : 
- For GGA coming at 10Hz, we set 1 sentence max per second so that it can be output on a 4800baud link.
- For VDM/VDO possibly coming by bursts : 30 sentences per 30s -> this is the same average output rate than for GGA but allows bursts of sentences.
- Etc.

The algorithm is not too complex to implement : Each time a new sentence arrives, you "just" have to count how many of these sentences have been received within the specified time window and you drop the proper sentence in case of overflow.

The problem is probably to present such a configuration in a understandable manner to the end user. Maybe we could show a checkbox with something like "Auto Rate Limiter" ? OCPN would define the filtering window for the most known sentences, taking into account the target bitrate.

leamas, 2024-01-03:
> The problem is probably to present such a configuration in a understandable manner to the end user

Indeed, the idea was to make it easier for the end user, not worse.

We should probably focus on defining  some realistic test cases first. It's hard to evaluate anything before we can agree on those.

TwoCanPlugIn, 2024-01-08:
Why are we contemplating sending AIS traffic over a low speed serial port ? I believe this is against the standard and I very much doubt that old equipment that only supports 4800 baud is capable of handling VDM/VDO messages.

The NMEA 0183-HS standard (38400 baud) was specifically designed to handle the higher data rates required by AIS and Radar (ARPA). 

As I stated before, if OpenCPN is sending NMEA 0183 data over a low speed serial port, then it should adhere to the NMEA 0183/ IEC 61162-1 standard and transmit the sentences at the specified rates, typically 1Hz.

And if OpenCPN cannot handle the higher data rates for incoming data, such as NMEA 2000 messages, and you have to implement filtering, then you have a serious latency problem.

nohal, 2024-01-08:
If all we talk about here is the output connection and it's behavior, then increasing the FIFO buffer size of the output connection seems like an absolute no brainer to me to cover the burst scenarios and will simply handle everything where data discarding is not yet required.
The size of the buffer should probably reasonably cover about a second of data at link speed (Which is something over baudrate/8/82 messages for NMEA 0183 given not all messages are the maximum size defined by the standard), if much bigger, the latency will become unacceptable.
Discarding data starts to be required when the link is not being able to transfer everything it should even when fully saturated (=when the FIFO buffer is full). For that many possible implementations are possible of course, but simply letting pass every Nth sentence of the ones coming at high rate while increasing N until the pressure on the FIFO buffer drops to reasonable might do.

nohal, 2024-01-09:
I have implemented a prototype of the filter in https://github.com/nohal/OpenCPN/commit/6baa975ca70d63eeb8348bfb173178ca52f5085c

On heavily overloaded outputs the biggest problem in the current codebase actually is `CommDriverN0183Serial::SendMessage` which tries to resend the message in case of a full buffer (Which of course is full all the time).
My PoC seems to work quite reasonably, but is not magic, when the input:output ratio grows to something like 10:1 it simply hurts a lot and throttling each message individually taking into account what it is would work better. The size of the FIFO buffer matters a lot at low output speeds.

Rodemfr, 2024-01-09:
Sizing the output buffer with the bitrate is indeed solving all the burst issues.

An example of NMEA bursts : I use [MicronetToNMEA](http://www.github.com/Rodemfr/MicronetToNMEA) as a bridge between my TackTick instruments and my tablet with OpenCPN. By nature Micronet network updates all data at once every second. You then have a burst of NMEA0183 sentence which overflows the current 20 sentence buffer.

TwoCanPlugIn, 2024-01-13:
A 4800 baud serial port can handle about 480 characters per second. Given the maximum size of a NMEA 0183 sentence is 82 characters and the NMEA 0183 standard stipulates an update rate of 1Hz, that means the maximum number of sentences that can be sent in one second is 6.

Now let's add a real use case. My old Robertson Autopilot has wind and chartplotter interfaces. When the autopilot is operating in Wind mode, it expects to receive a MWV every second. Similarly if operating in Nav mode, it expects to receive a XTE sentence every second. If it times out receiving either of these, it displays an alert and usually stops following the commanded direction. 

Another use case could be an older NMEA 0183 instrument display. Let's say it is configured to display depth data. Again it will expect to receive a DPT sentence every second. When it times out receiving the expected sentence it will probably alternate between display a depth value and '---'.

So if your FIFO output buffer exceeds the maximum number of sentences that can be sent at 1Hz for a given baud rate, it will never empty without the target device timing out on it's expected messages.

The approach you're taking appears elegant but I personally think it is just plain wrong.

nohal, 2024-01-13:
Plain wrong is feeding an output with more data than it can transfer, nothing else.

What can be done about it without being aware what the desired behavior should be is the question.

There are several approaches varying between universal, which is what I outlined, which needs zero configuration and produces (ideally) uniform delays proportional to the overload ratio, which of course may exceed what the consumer on the other side of the overloaded link expects and you call "plain wrong" and one that is aware of the use case (of which you name multiple) and acts accordingly, which of course needs configuration.

Limiting to 1 Hz, as elegant as it might seem and as insufficient as it is if there are more than X (~6*82 or ~15 average in my data samples) different sentences per second and as hard to actually do for example for AIS data or XDR sentences from multiple sensors (Which on the other hand are the usual sources of the overload, besides the data bursts originating from batch data conversion), is simply not the only real use case and I personally think pretty much exactly as "plain wrong" as doing anything else that was not explicitly requested by the owner of the system.

The reality is that NMEA0183 sentences in average are nowhere near the 82 byte limit in any system I have ever seen (The averages actually are ~30 in all the data samples I have available). From my testing it is also absolutely clear that for the bursting scenario the FIFO buffer somewhat bigger than what the calculation based on the average message size suggests for one second simply performs better, at a price of occasionally having to drop messages  (but does it really matter in a scenario where by definition any algorithm is dropping data?)

leamas, 2024-01-13:
One problem with a simple "discard overflow" strategy is that if the load is a mixture of say one message which it comes "too much"  and another which comes seldom. If we just discard the things which we have no space for the risk is that the "seldom" message completely disappears.

My usecase, the initial report here is such a load. This is the reason I suggest the algorithm above. This is still simple in the sense that it does not need any configuration. This should really be a requirement on any solution, users could always use filtering if configuration should be needed.

As @nohal noted, the algorithm used for trimming the buffer and the buffer size are more or less independent.

TwoCanPlugIn, 2024-01-13:
I think your specific use case is non-standard. (of course you could have configured your Waveshare GPS Hat to output at 1Hz instead of 10Hz), in which case you probably would never have discovered this "anomaly"

Keep in mind that the NMEA 0183 standard is one talker/multiple listeners over a 4800 baud connection. A talker only sends the data that it is designed to do (Position, Depth, Wind, Boat Speed, Autopilot commands etc.) and at a rate of 1Hz.

When you say 

> Plain wrong is feeding an output with more data than it can transfer, nothing else.

it ignores the fact that the NMEA standard is designed  to prevent this.

For the higher data rates and unpredictability of AIS, that's why the NMEA 0183 HS standard at 38400 baud was introduced.

Here is another use case: VHF DSC radio. It expects to only receive RMC, GGA or GLL sentences,  Why on earth would you overload the 4800 baud connection with extraneous data at the risk of causing the radio to lose its current position.

Ideally, OpenCPN's NMEA 0183 output should be filtered, and sentences sent at 1Hz as per the standard. And perhaps for a serial connection configured at 4800 baud, OpenCPN does not send AIS or ARPA messages.
 
Ultimately, you're writing the code and it's your software, so we'll just have to agree to disagree.

nohal, 2024-01-14:
That the standard is designed this way is sure nice. But by pure existence of this issue we see that the world does not care about it much. And to be honest I probably yet have to see a product actually enforcing this aspect of the standard in the multiplexer scenario, which is what we do.

Of course limiting data to 1Hz on connections configured for very low speeds would be one of pretty sane ways to try to approach the filtering problem. Still not solving the problem where the amount of sentences present in the data stream is not able to pass even at 1 Hz, but OK.

Do we really want to know that the device on the other side of the link is a radio and apply some sort of filtering automatically? I do not think so, we provide filtering features the user can use to do it.

Of course backing your arguments with actual code proving they work would be nice, but whatever, I will probably write some more "my code" for "our software" and see how inferior you will  consider it then.

leamas, 2024-01-24:
Added  line buffer implementing my ideas [above](https://github.com/OpenCPN/OpenCPN/issues/3391#issuecomment-1668545328) in https://github.com/leamas/OpenCPN/tree/buffer-3391. This is orthogonal to @nohal's  work so far focused on adapting the buffer size. 

The unit tests are reasonably comprehensive. Performance seems ok -- the worst  case insert is about 10 usec on a Rpi 4

Assigning bug to Pavel

leamas, 2024-04-11:
Not really, for two reasons.

The first is that configuring filters makes the setup much more complicated. Actually it's probably a long way for most users to understand that filtering is needed to handle a basically non-functional system (it took some time for me).

The second is that the algorithm here is much more clever than a dumb filter. It passes all data through as long as it works, only limiting data which comes too often.

Hakansv, 2024-04-11:
Using a serial output to an AP you would need a filter to restrict the TX to $ECxxx and maybe some wind-data if supported by the AP. The native e.g. GPRMB/GPRMC is not what you need. So here is the filter not dumb but necessary.

leamas, 2024-04-11:
>  So here is the filter not dumb but necessary.

"dumb" and "necessary"  are not mutually exclusive

However, without filtering the unwary user just faces a system which does not work with a behaviour as of the start. We can and should add filtering of messages which should not be forwarded on top of this, this logic is not touched. But  it should be on top of something which does not show the behaviour which is the root of this issue.

Furthermore, when you add NMEA2000 feeds into the equation it's perhaps not enough to filter out the relevant messages.  Such a feed could swamp a slow connection also with "correct" data..

Hakansv, 2024-04-11:
If anyone tries to choke a 4800 output by e.g. tx-all why not simply message the user when there's a buffer overflow and ask him to restrict the data output using a filter on "NMEA 0183 output to COM X"
Some logic to detect what kind of messages to let through would fail? If I set up a tx to an AP how would that logic make a clever selection to feed the AP with the correct RMC for example.

leamas, 2024-04-11:
>  restrict the data output using a filter on "NMEA 0183 output to COM X

This is probably harder implementation-wise. And a filter just wont work if a NMEA2000 device sends too much data where we need to use some if it, but bot all.

I have been looking into this, overruns are interesting feedback even handled by buffering. Could probably be added.

But the core question is why you feel so strong about this, that it shouldn't be done?

rgleason, 2024-04-11:
N2000 converted to N0183 is a danger without filtering or Alec's and Pavel's ideas.
How many new users will configure this way?  My opinion: Not many, but it can happen.
Perhaps there should be a message as Hakan suggests.

leamas, 2024-04-11:
> The "normal" use for an N0183 output to a serial port using baude 4800 is to send data to an AP.
If we implement something that tries to take away the need for a filter no user will get that to work.
This connection will use only N0183 and not N2k.

I don't see the problem here. This PR should do the right thing, handling overruns if they happen and otherwise basically do nothing.

Are you under the impression that this PR in some way limits the way to manually create filters?  It does not.  If a user defines working filtering there will be no overruns and the buffering thus forwards all messages. That is, with a working filter setup everything works as before the PR

Hakansv, 2024-04-11:
Good, thanks.
My discussion about what's happening if a user set up an output creating a buffer overload. Do we have the opportunity to message that in some way with this PR? That will be an advantage I suppose?

leamas, 2024-04-11:
I think this might be the conclusion. That is, for each message where we detect overruns we should provide some feedback, at least in the log. Creating something using pop-ups might be tricky since it can become too noisy.

Perhaps a single pop-up message "There are overruns, please check the log" ?

Hakansv, 2024-04-11:
A single pop-up indication a data connection trouble in some way may be very handy. "overrun" may not be understood?

leamas, 2024-04-11:
Well, the wording was just a sketch. I agree that this is not end-user language.

TwoCanPlugIn, 2024-04-12:
I think the word you are looking for is "overflow".

An alternative could be "error" but that might frighten users.

Nonetheless as Hakan and myself have argued, it is an elegant solution to a problem that is far more easily solved simply by the use of filters.

If the NMEA 0183 output is to an Autopilot, depending on the brand/mode and the mode of operation it may only need RMB, APB, XTE sentences. (In fact my old Robertson AP200 only uses XTE sentences in Nav mode)

More importantly, assuming this solution is in the current builds, it does not work. 

In my testing, for input, O receives position, speed, wind updates at 10Hz. Output is configured at 4800 baud. A route is activated. The algorithm does not seem to work as you have designed, as the RMB, APB and XTE sentences are not reliably sent. (As evidenced by being coloured RED in the debug Window and not being received in a Serial Port monitor)

Output from Putty
[image]

NMEA Debug Window
[image]

On behalf of our users, please stop this madness. The advice they will have to be given, "use filters" will still be necessary even with your fancy algorithms and buffers.

TwoCanPlugIn, 2024-04-12:
I should add another scenario, a connection to a VHF DSC radio, where it receives GGA, GLL or RMC sentences from which to derive its current position, and where it sends and receives DSC sentences. Designed with NMEA standards in mind, it is expecting to receive these at 1Hz frequencies. 

Important to note that none of the multiplexer/gateway vendors implement anything that you are attempting. For example as the following table shows, by default position sentences are sent at 1Hz, a per the NMEA standard for 4800 baud connections.

[image]

Just because your RPi GNSS Hat is sending data at a non-standard rate, doesn't mean that you should ignore the standard and output the data at the same rate.

leamas, 2024-04-12:
We now have a good setup to test various buffer strategies. The buffer interface is `CommOutQueue` defined [here](https://github.com/leamas/OpenCPN/blob/pavel/buffer-3391/model/include/model/comm_out_queue.h).  To swap in a new buffer implementing this is more or less a oneliner.

Patches welcome...

leamas, 2024-04-13:
Hm.. After a nights sleep I begin to think more about @TwoCanPlugIn  idea to limit the rate. Since it is about complying to standards it's never wrong to do so. Furthermore, it would actually be quite simple to modify current buffer so it  forwards messages at a higher rate the 1 Hz.

There would the still be corner cases where we run into overruns for other reasons (communications problems, a **real lot** of message tyeps, data rates slower then 4800 bps, should the happen.  But that would jsut be about keeping current code.

We should log when we limit the rate, but not user dialog for that -- it's just normal.  If there are overruns anyway, this should lead to a GUI message, though.

I will try to implement this.

leamas, 2024-04-15:
@TwoCanPlugIn : have implemented rate limits i. e, messages which comes faster than a given rate are dropped. Although not that bad code wise I'm not convinced this is the right thing to do.

Consider the two drivers n0183_net and n0183_serial things are quite different. The network driver has basically no problems with output bandwidth while the serial certainly has.

Furthermore, say that we start to drop messages with less than 900ms "gap" between each. This means that we will only transmit this message as 1.8Hz, which probably  is worse than 0.9Hz. For this reason a gap triggering should probably be lower, say 0.7 or 0.6 Hz. But how useful is this then, really?

Also, limiting the rate might lead to all sorts of problems for "bursty" data with limited overall rate.

OTOH, limiting the buffer size to say 10 messages also means a time limit if the output channel is constrained. That is, if we start to throw outdated messages when we have 10  or 15 messages in buffer this is the right thing to do because it means that the oldest messages are about a second old using a 30 byte/message estimate. It also has the nice benefit that it does nothing at all if nothing is buffered i. e., the output channel can keep up with the input.

For these reasons I think we should configure the filter with a rate limit of 0 ms, effectively disabling this.

A possible refinement could be to define the buffer size in bits rather than messages. If we impose a max buffer of 4800 bits it would mean that we drop messages older than one second on a standard 4800 bps connection. This is well defined and easy to implement. It would basically define a well-defined limit on how old a message could be before it's dropped rather than forwarded.

Note that the "size" here is for each message type, the overall buffer size is basically unlimited, depends on the number of message types.

Thoughts?

TwoCanPlugIn, 2024-04-15:
I agree with your initial comment, in that I don't think an incoming rate filter is of benefit. The scenario I had in mind and where I think this becomes a problem is where there are multiple devices connected to a multiplexer (eg. depth, wind, speed, gps, heading) connected to the PC with either USB, WiFI or Ethernet and one slow speed serial output port connected between the PC and an autopilot (or PC and VHF DSC). All of the incoming sentences should be sent at the 1Hz rate as per the standard, simply because, irrespective of the multiplexer, that is the rate at which the actual sensor transmits. 

If the user has not configured an outgoing filter then DBT, MWV, VHW, GGA, GLL, GSV, RMC as well as autopilot APB, RMB and XTE sentences are all going to be transmitted over the serial port, and possibly overflow the port. The risk is dropping the sentences necessary for the autopilot. Add to that scenario, if an AIS transceiver is present, we may also get bursts of AIS traffic. 

I don't think any of the multiplexer/gateway vendors perform any magic. For example all of the NMEA 2000 gateway vendors only generate and send position sentences (GGA, GLL, RMC etc) at 1Hz even though the NMEA 2000 PGN's (eg pgn 129025 & 26) are sent at 10Hz. From reading the literature for the multiplexers, it doesn't seem that they do any fancy buffering. 

I think that is difficult to decide on whether to use a buffering technique based on incoming rate, inter-sentence gap, or bit lengths..

If you are to think of discarding sentences, if the user has an active route, once the buffer overflow is detected, then perhaps other sentences should be dropped at the expense of ensuring APB, RMB & XTE sentences are sent. However what if the outgoing connection is to a VHF DSC radio ? It only accepts position sentences, it doesn't care about depth, speed, wind, heading etc.

I think at the very least there should be some way of alerting the user that a slow speed out put port is overflowing. The guidance should simply be for the user to select the appropriate filter, eg. only allow autopilot sentences to be transmitted out of the serial port.

I think we're trying to implement something quite difficult, which could be solved simply by better end-user education.

[image]

leamas, 2024-04-15:
I have added a pop up dialog when overruns are detected.  

As for the rest: yes, it's hard and eventually it's better if user configures some kind of filtering. But until that point we must still  behave reasonably consistent to avoid the really confusing problems described in the top comment. That behaviour is just unacceptable.

The filter does both these things: discovers the overruns and handles them in a way which keeps the system running in a reasonably consistent way. 

Also, the discussions here basically boils down to disable the rate  limit and set the buffer limit to 10 or 15 messages corresponding to about a second of data. Older data than so is discarded.  Overall, I think this means the buffering does the right thing.

PR #3818 (2024-04-18): Serial driver: Fix high load problems
Fixes problems described in #3815 and  #3391:

  - Make sure all data is actually written
  - If overflowing, apply a sane strategy rather than just silently dropping messages at random.
  - Remove the static variable which makes it impossible to have more than one instance.

Closes: #3815
Closes: #3391

Current state: Works for me, including the test case described in #3391
closed 2024-04-18 by leamas