OpenCPN/OpenCPN #3393: Upload to Garmin GPS no longer working on 5.8.4

mcdonaldajr, 2023-08-08:
**Describe the bug**
Since updating to 5.8.4 on my Linux Mint machine I can no longer upload routes to my Garmin 557. I get the error message Error on Route Upload. Garmin GPS not connected, when I select a route and press Send to GPS button in Route Manager. Prior to pressing the button, I am successfully receiving GPS data. GPS 557 communication protocol is set to Garmin Mode, and OpenCPN connection is set to "use Garmin Mode".  Also "Use Garmin mode for uploads" is selected. NMEA Debug Window shows NMEA sentences being received, and stops when attempting to send data to GPS, and does not restart until I restart OpenCPN. If I use OpenCPN 5.4.2. on my Windows Laptop, it uploads the route successfully. I can also use GPSbabel on this Linux Mint machine to send routes to my GPS, so I am confident it is not a hardware problem. I would like to revert to OpenCPN 5.4.2. on my Linux Mint machine to completely prove that it is version 5.8.4 causing the issue, but don't know how to downgrade OpenCPN.

I have ensured that the permission are set correctly: 
$ ls -l /dev/ttyUSB0
crw-rw-rw- 1 root dialout 188, 0 Aug  8 13:26 /dev/ttyUSB0

and that I am in the dialout group:
$ groups
anthony adm dialout cdrom sudo dip plugdev lpadmin sambashare

The USB to Serial product is an Actisense USG-1 NMEA Serial Gateway

Extract from log below after attempting to Send to GPS twice.

13:10:12.066 MESSAGE glChartCanvas.cpp:1009 OpenGL-> Framebuffer OK, size = 632
13:10:12.072 MESSAGE glChartCanvas.cpp:554 BuildFBO 3
13:10:12.072 MESSAGE glChartCanvas.cpp:1009 OpenGL-> Framebuffer OK, size = 640
13:10:19.250 MESSAGE comm_bridge.cpp:186    ***GPS Watchdog timeout at Lat:56.2118   Lon: -5.55733
13:10:26.256 MESSAGE ocpn_frame.cpp:5641 LOGBOOK:  2023-08-08 12:10:26 UTC  DR Lat   56.21182 Lon   -5.55733
13:10:33.871 MESSAGE comm_drv_factory.cpp:54 MakeCommDriver: Serial:/dev/ttyUSB0
13:10:34.512   ERROR gps_wx_logging.h:14 GPS_Packet_Read: Bad count
13:10:34.512 MESSAGE gps_wx_logging.h:30 Unit:	GPSMAP 55are Version 6.8Ietail v2022.0 2elsh Scrn, 480x64er Unknown
ID:	1665
Version:	6.80
13:10:34.692 MESSAGE gps_wx_logging.h:30 A001 protocol not supported
13:10:34.693   ERROR gps_wx_logging.h:14 INIT: No table entry for ID 1665

13:10:36.006   ERROR gps_wx_logging.h:14 GPS_Packet_Read: Timeout.  No data received.
13:10:36.014 MESSAGE comm_n0183_output.cpp:439 Garmin GPS could not be initialized on port: /dev/ttyUSB0 Error Code is -1
 LastGarminError is: 
13:11:11.049 MESSAGE comm_drv_n0183_serial.cpp:261 Closing NMEA Driver Serial:/dev/ttyUSB0
13:11:11.049 MESSAGE comm_drv_n0183_serial.cpp:278 Stopping Secondary Thread
13:11:12.049 MESSAGE comm_drv_n0183_serial.cpp:290 Stopped in 1 sec.
13:11:12.049 MESSAGE comm_drv_factory.cpp:54 MakeCommDriver: Serial:/dev/ttyUSB0
13:11:14.003   ERROR gps_wx_logging.h:14 GPS_Packet_Read: Timeout.  No data received.
13:11:14.003 MESSAGE gps_wx_logging.h:30 Unit:	GPSMAP 557, UK/IRE Software Version 6.80
ID:	1665
Version:	6.80
13:11:14.183 MESSAGE gps_wx_logging.h:30 A001 protocol not supported
13:11:14.183   ERROR gps_wx_logging.h:14 INIT: No table entry for ID 1665

13:11:16.002   ERROR gps_wx_logging.h:14 GPS_Packet_Read: Timeout.  No data received.
13:11:16.010 MESSAGE comm_n0183_output.cpp:439 Garmin GPS could not be initialized on port: /dev/ttyUSB0 Error Code is -1
 LastGarminError is: 
13:11:17.010 MESSAGE comm_drv_factory.cpp:54 MakeCommDriver: Serial:/dev/ttyUSB0

**Desktop (please complete the following information if applicable):**
System:
  Kernel: 5.15.0-78-generic x86_64 bits: 64 compiler: gcc v: 11.3.0 Desktop: Cinnamon 5.6.8
    tk: GTK 3.24.33 wm: muffin dm: LightDM Distro: Linux Mint 21.1 Vera base: Ubuntu 22.04 jammy

leamas, 2024-02-07:
This seems to be a duplicate of #3496 (or rather the other way around).  This is fixed in current master, adding label

rgleason, 2024-02-08:
@mcdonaldajr  Perhaps the original poster OP should close this now. Please.

closed 2024-02-09 by mcdonaldajr