OpenCPN/OpenCPN #3398: Unable to make a list of routes (PlugIn_Route_Ex)

Rasbats, 2023-08-14:
**Describe the bug**

The API does not have access to a list of RouteEx, unlike the waypoints in a RouteEx. Have tried to make my own route list but failed several times.

Cannot make a list using std::list either. Got involved with unique pointers in the definition of` GetRouteEx_Plugin()` .

```
 m_bUsingFollow = false;
 PlugIn_Route_Ex thisRoute;
 list<PlugIn_Route_Ex*> mylist;
 
 wxArrayString myRoutes = GetRouteGUIDArray();

 for (int r = 0; r < myRoutes.size(); r++) {
 
 	thisRoute = *GetRouteEx_Plugin(myRoutes[r]);
 	wxMessageBox(thisRoute.m_NameString);
 	mylist.push_back(&thisRoute);
 }
```
 `GetRouteGUIDArray()` works fine. Should a list of _PlugIn_Route_Ex_ be defined in the API to work with `GetRouteEx_Plugin`?

leamas, 2023-08-14:
This is probably about ` unique_ptr<>` not being assignable. After all, if a variable could be assigned with the value of  such a pointer, the first pointer would no be unique any more.  For that reason, a `unique_ptr`  value must be transferred using `std::move`  which transfers the pointer from one variable to another. 

I have not tested anything, but this code compiles when pasted into the ShipDriver constructor:
```
std::vector<std::unique_ptr<PlugIn_Route_Ex>> routes;
auto uids = GetRouteGUIDArray();
for (size_t i = 0; i < uids.size(); i++) {
   routes.push_back(std::move(GetRouteEx_Plugin(uids[i])));
}
```

 Thanks for locating `GetRouteEx_Plugin()`, I was looking for something like that without success

Rasbats, 2023-08-14:
Ok ... we have the routes but how to use the route names to populate a dialog? `unique_ptr<>` again?

leamas, 2023-08-14:
Isn't it just to dereference like any pointer? `wxString name = routes[0]->m_NameString` , etc. Or?

leamas, 2023-08-14:
IN any case this is not a bug, more like a discussion . I know, partly my bad. Converting.