OpenCPN/OpenCPN #3400: Info windows have no title bar in night mode

rahra, 2023-08-18:
**Describe the bug**
The information windows such as the object query or the AIS target query have no window border and title bar if the night mode is active. Because of this the windows cannot be moved or resized.

**To Reproduce**
1. Switch to night mode with F5.
2. Open a window by double clicking somewhere on the map or an AIS target.
3. Window opens without border and title bar.

**Expected behavior**
It should have a title bar as it does have in day mode.

**Screenshots**
The following screenshots show the difference between day and night mode.

[image]
[image]

**Desktop (please complete the following information if applicable):**
This happens at least on a standard Ubuntu 22.04.3 LTS as well as Ubuntu 23.04.
In both cases OpenCPN is installed from the official OpenCPN PPA, currently at version 5.8.4.

rahra, 2023-08-24:
There is something odd with the night mode anyway. I brought up some problems in the past and it is discussed  in this issue: https://github.com/OpenCPN/OpenCPN/issues/2416

Further down in that issue I posted various screenshots which also show the behavior of this issue. So probably it is somehow related to each other.

transmitterdan, 2023-08-31:
This does not seem to happen on Windows:

[image]

bdbcat, 2023-08-31:
This is done intentionally on some dialog displays.
We use standard window controls provided by the operation system.  The system APIs give us very little ability to style the window decorations on most platforms.  So, for instance, we cannot change the color of window border reliably.

cf. AISTargetQueryDialog.cpp:189
Read the comment there for further discussion.

rahra, 2023-09-01:
> This is done intentionally on some dialog displays.
> We use standard window controls provided by the operation system. The system APIs give us very little ability to style the window decorations on most platforms. So, for instance, we cannot change the color of window border reliably.

This issue is not about the color of the window decoration. It is that the window decoration is missing (see screenshot).

bdbcat, 2023-09-01:
The available window decoration includes a pure white 2 pixel border around the entire window.
This destroys night vision completely.

nohal, 2024-02-14:
I would say the dark mode support on Linux has become pretty decent so we can either remove the decorations hack completely or at least make it optional and disabled by default - it seems not right as this issue shows.

rahra, 2024-02-15:
I can confirm that dark/light switching is just a single click on a widget in the task bar and almost all applications adapt immediately. Even OpenCPN (I tried), but as mentioned in this thread there are some oddities.

commit a4acf607 (2024-02-16): Make the dialog window behavior consistent. Closes #3400

- The confusing behavior disabling window decorations actually never worked
  on anything but some Linux flavors, never on Windows, which is
  the only platform where this is a real problem
- Nowadays dark theme support and switching between themes is well
  supported on Linux and macOS, with Windows coming in wx 3.3.
  If we want to implement what we tried with this, we need custom
  decoration-less windows that work on all platforms instead of hacks
  with wxDialog/wxFrame.
closed 2024-02-16 by nohal