OpenCPN/OpenCPN #3404: RFE : add read "raw n2k" YDEN & YDWG data from network

captcha1, 2023-08-21:
**Is your feature request related to a problem? Please describe.**

It would be good to get all the NMEA 2000 data (eg : position, COG/SOG, wind, AIS) into OpenCPN over the (Wi-Fi) network.

**Describe the solution you'd like**

Please read "raw NMEA 2000" from Yacht Devices YDEN & YDWG devices, as described in https://yachtd.com/downloads/ydwg02.pdf , APPENDIX E. "Format of Messages in RAW Mode" : Messages sent from Device to PC have the following form: hh:mm:ss.ddd D msgid b0 b1 b2 b3 b4 b5 b6 b7<CR><LF>

I can loan someone a YDEN device if it helps.

**Describe alternatives you've considered**

IEC 61162-450 and / or 61162-460 would be nice as well (but I've never seen this "in the wild").

rgleason, 2023-08-21:
I believe the design decisions around this subject were carefully considered and the programmers had good reason for the decisions made. You are asking for redesign.

rgleason, 2023-08-22:
I believe Pavel will be closing this with "not implementing" comment.

bdbcat, 2023-08-22:
Rick...
This is on the roadmap for next major OCPN release.

rgleason, 2023-08-22:
Thanks. Dave