OpenCPN/OpenCPN #3406: Headless use case: Activating route does not work

leamas, 2023-08-22:
**Is your feature request related to a problem? Please describe.**
In the headless scenario described in  #3346 it is not possible to activate a route directly from the same remote PC which is used to create a new route.

**Describe the solution you'd like**
Two parts. The first would be a simple checkbox in the _Transfer Route to Peer_   dialog which allowed for activating the transferred route directly.

The other part would be a remote interface which allowed to list and manipulate routes. This would include Activate, Deactivate and Reverse, the most common actions.  This interface could perhaps be exposed to user in a plugin, most users will definitely not need it.

tkurki, 2023-08-28:
In addition you'd probably want stuff like setting and advancing the next point in route, go to waypoint and go to "cursor"/arbitrary point.

The operations that you describe sound a lot like the Course API that we recently added to Signal K v2: 
- OpenAPI description: https://demo.signalk.org/doc/openapi/?urls.primaryName=course
- Documentation: https://github.com/SignalK/signalk-server/blob/master/WORKING_WITH_COURSE_API.md

This is what we came up with, but I'd be really happy to hear comments. If somebody wants to try it out I can provide a Docker image with some prepopulated data & Freeboard as the UI and/or a demo system accessible over the net.

Afai can understand OpenCPN also identifies routes with UUIDs, so adding this API as is or just aligning things for potential future cooperation would be pretty straightforward.

leamas, 2024-01-16:
I did the first part here, activation. This was somewhat constrained by the existing API, and I focused to get the internal structure  better for now, expanding the current api which is based on parameters rather than json.

But for the second part, a more general list and modify routes, this API looks nice and could probably be used. I'm interested in doing it, but will probably not be able to do it in the 5.9/5.10 cycle.

leamas, 2024-01-22:
closing this after  #3584. There is a obvious place for a more general REST interface, but this is better handled in a separate bug.

closed 2024-01-22 by leamas