OpenCPN/OpenCPN #3409: Headless use case: Externally triggerred exit needed

leamas, 2023-08-22:
**Is your feature request related to a problem? Please describe.**
In the headless use case #3346, the natural way to shut down the system is (would be) to just push a button. However, even if picking up the button is trivial, there is no way inform OpenCPN it should  exit. This leads to the "last start seemed..." dialog when starting, which is a real headache in the headless case

**Describe the solution you'd like**
There should be a remote API making it possible to issue a command like `opencpn --remote --exit` which initiates an ordered  shutdown.

**Describe alternatives you've considered**
The simplest alternative would be to expand the current command line interface which uses sockets to communicate with an existing opencpn instance. However, this is less attractive due to #3349 which basically means it does not work for Flatpak.

A solution to #3349 will probably pull in a dbus interface. Adding a "Exit" function to such an interface would be trivial

tkurki, 2023-08-29:
It seems overly simplistic to concentrate on shutdown. How would you start the headless process? In what UI would the end user send the shutdown message from? Each os has its own way yo manage services. On Linux systemd. Why not think about headless O as yet another daemon?

tkurki, 2023-08-29:
Ah just saw https://github.com/OpenCPN/OpenCPN/discussions/3346#discussioncomment-6609559

So the real problem is that there is no clean shutdown mechanism from out of process.

leamas, 2023-08-29:
> . How would you start the headless process?

Automatically while booting

> In what UI would the end user send the shutdown message from

By pressing a physical button.

> Why not think about headless O as yet another daemon?

Headless or not, OpenCPN is still a GUI foreground program and as such  not really a good candidate for a daemon --  the daemon concept is geared towards things which runs in the background with no or little UI.

Using a systemd service is basically a band aid until we have fixed #3349 -- after that we could just use the normal freedesktop autostart  mechanisms by dropping opencpn.desktop into _~/.config/autostart_

> So the real problem is that there is no clean shutdown mechanism from out of process.

That's one of the problems and also to the very problem tracked by this bug.

> just saw https://github.com/OpenCPN/OpenCPN/discussions/3346#discussioncomment-6609559

However, that comment is about another issue tracked by #3349.

My current thinking is that both of these could be solved by adding a Dbus interface, I'm exploring this path.

PR #3502 (2023-12-20): New local communications framework
A stab to formalize the command line API exposed by a OpenCPN client. This introduces new options to 
  - Terminate a running opencpn instance.
  - Open a GPX waypoint/routes file in a running instance
  - Print the local REST server endpoint like http://localhost:8000

Internally, the API is formalized using interfaces and base classes. This part also covers the single instance checking.  Communincation backends exists for both the "old" style wxWidgets  socket communication and, for Linux, dbus. The main reason for using dbus is that it works also on Flatpa [... cut]
closed 2023-12-20 by bdbcat