OpenCPN/OpenCPN #3412: OpenCPN looses active track when crashing

rahra, 2023-08-24:
**Description**
OpenCPN occasionally crashes (rarely) thereby loosing the active track. It's possible to restart it directly afterwards but it's bad that the track data is lost. Probably the track data should be stored in more robust manner so that OpenCPN can recover it after an unexpected crash.
This issue seems to be similar or probably related to https://github.com/OpenCPN/OpenCPN/issues/3316 but I'm not sure.

I'm on a trip since 4 months and made more the 6000 nm so far, heavily using OpenCPN, and it happened about 3 or 4 times so far. The last time was yesterday and I found the following segfault in `dmesg`:

```
opencpn[39538]: segfault at 5a38306a ip 00007f773691a4e1 sp 00007ffe23fa8790 error 4 in libwx_baseu-3.2.so.0.2.0[7f7736874000+13c000] likely on CPU 1 (core 1, socket 0)
Code: 4c 01 e7 8b 77 20 83 ce 01 e8 7b bb f5 ff 48 83 c4 08 4c 89 e0 5d 41 5c c3 f3 0f 1e fa 41 56 31 d2 41 55 41 54 55 48 89 f5 53 <4c> 8b 2f 48 89 fb 4c 8b 77 08 48 8b 06 48 89 ef 31 f6 4c 89 e9 4d
```

**To Reproduce**
Unfortunately I could not yet find out what causes it.
Yesterday I tried to fix the odd colors in night mode and after that I rotated to the colors with F5 and at some point it crashed. But that could have been just an unrelated coincidence since I tried to fix night mode colors several times in the past and I always rotate through the colors with F5 every night without a crash of OpenCPN. At one occasion it even crashed by itself, it didn't touch it at all.

**Expected behavior**
Of course it shall not crash. But even then it should be able to recover the data from the latest track which was active while crashing.

**Desktop (please complete the following information if applicable):**
 - Ubuntu 22.04.3 LTS
 - OpenCPN 5.8.4 installed from official PPA.

leamas, 2023-10-24:
This is also discussed in https://github.com/OpenCPN/OpenCPN/discussions/3346#discussioncomment-6759601. 

TL;DR This is not only about when OpenCPN crashes.

rahra, 2023-10-25:
Are you sure?
I'm talking about the tracking and your issue is about the currently active route.

leamas, 2023-10-25:
sorry, you are right. It's related, but not the same

nohal, 2024-02-14:
The recovery from crashes should be more robust in the current master. I am still completely unsure how this can actually happen at all as the changes file is open in append mode. Anyway, leaving all these issues open until we have some more testing input with the new code.

rahra, 2024-02-15:
I just found out that I probably have a backup of the corrupt files. If yes, I'll check and analyze it.

leamas, 2025-05-03:
Supposedly fixed in 84f2b70bf. Marking as Done pending verification