OpenCPN/OpenCPN #3415: Navobj back-up: add monthly and weekly copy

bcn2, 2023-08-29:
**Is your feature request related to a problem? Please describe.**
We see from time to time - quite frequent - users that did loose their tracks, routes etc. - the navobj file - and did "eat up" their five revolving back-up copies.

**Describe the solution you'd like**
As one might not be aware inmediately that one is loosing that data, and trying again and again restarting OpenCPN, why not providing some longer term copy? One month and one week dated for example. A safety cushion.

antipole2, 2023-10-13:
Or recover from your regular backups?

bdbcat, 2023-10-13:
We are planning this enhancement for the next OCPN version. (automatic monthly/weekly backups, safely stored as read-only archive.)

commit a3d01fb0 (2024-02-13): Implement monthly and weekly rotating backups. Closes #3415
closed 2024-02-13 by nohal