OpenCPN/OpenCPN #3420: O loses track when crossing dateline (180E/W)

LeaseOnLife, 2023-09-04:
We crossed longitude 180 from east to west, in Fiji. The active track (bright red) continued across the 180 degrees longitude. 
But when stopping and starting O,  the newer portion, since crossing 180 degrees longitude  (in our case the western portion) of the track was gone/lost, all the way to current position. This happened on 2 systems: Both are Debian bookworm, latest git-master, on a X64 laptop as well as on a PI4

Not sure how to reproduce, we don't have a VDR running all the time. Maybe simulating something with the shipdriver plugin will show a similar behavior.

rgleason, 2023-09-04:
See these Weather_Routing  bug Issues which are very similar.

 Bug 04 Routing at Date Line West from 180 is frozen #79 
https://github.com/rgleason/weather_routing_pi/issues/79

Bug 03 Entering position between 0 and 1 degree W gives false position. #99
https://github.com/rgleason/weather_routing_pi/issues/99

Perhaps related?

LeaseOnLife, 2024-05-23:
to revive this: we crossed the dateline again, west to east this time. I started a VDR just before, and also watched the track closely, to see what is happening:  While the active track continues to be drawn, past the dateline, there are no new track points added when looking at the track via the route/track/waypint manager. VDR log attached.
[vdr-dateline-log-RMC-only.log](https://github.com/OpenCPN/OpenCPN/files/15411408/vdr-dateline-log-RMC-only.log)

bdbcat, 2024-06-14:
Fixed.
Close?

LeaseOnLife, 2024-06-14:
no, unfortunately not, pulled latest from git and tested with the VDR provided. active track track-points are not added past IDL.

LeaseOnLife, 2024-06-14:
scratch that... had an ssh window open and built on the PI but tested on a previous version on the laptop. standby for test results

LeaseOnLife, 2024-06-14:
works, fixed, thanks! closed

closed 2024-06-14 by LeaseOnLife