OpenCPN/OpenCPN #3421: A route crossing the dateline (180E/W) is missing some lines between route points

LeaseOnLife, 2023-09-04:
To reproduce:

1. zoom into around Fiji, including some area west and east of loggitude 180.
2. draw a route with some waypoint west of 180 , then cross to east of 180 and draw a view more waypoints
3. move the chart to the left , so longitude 180 is on the left half of the screen
4. route sections west of 180 show no connection between route points, see screen shot
5. move the chart a bit to the right, 180 longitude to the east, 
6. the lines connecting the route points show up again

[image]

rgleason, 2023-09-04:
See these Weater_Routing  bug Issues which are very similar.

 Bug 04 Routing at Date Line West from 180 is frozen #79 
https://github.com/rgleason/weather_routing_pi/issues/79

Bug 03 Entering position between 0 and 1 degree W gives false position. #99
https://github.com/rgleason/weather_routing_pi/issues/99

bdbcat, 2024-06-14:
Fix.  
Close?

rgleason, 2024-06-14:
@bdbcat  Do you think this will correct Weather_routing issue?
https://github.com/rgleason/weather_routing_pi/issues/79

bdbcat, 2024-06-14:
Maybe.
Only thing to do is try it.

LeaseOnLife, 2024-06-14:
[image]
a quick test of weather routing worked both ways across the IDL. but I have no idea wether that was fixed with this commit, or was working before already.

rgleason, 2024-06-15:
I believe it has been fixed by this change as WR has not had a PR for this specific bug.