OpenCPN/OpenCPN #3422: CPN crash when fixing ENC in the screen area.

bc38, 2023-09-04:
Problem reproduction
- panning the map screen to an area with no ENC coverage
- CPN reload
- enable vessel auto-centering, return to vessel position in coverage area enc
- cpn crash

Any approach or panning into the ENC coverage area will crash.

Solved by deleting all SENCs and rebuilding the map data base.

Even after removing SENC and rebuilding the map database, all ENC settings are reset to default.
The problem is constant not only on my ENC cards but also reproduced on NOAA cards, tested on CPN release and master versions.

Crash report dump
[CrashReports.zip](https://github.com/OpenCPN/OpenCPN/files/12509988/CrashReports.zip)
Video
https://www.dropbox.com/scl/fi/pb6txz3p98gjpn6s4joqq/2023-09-04-140552.mp4?rlkey=dg89akrjfk829m61i1ia04p29&dl=0

transmitterdan, 2023-09-05:
Can you also post the opencpn.log file associated with the crash.  I tried to reproduce on Windows but could not.  Also, it might help to post opencpn.conf as well.

When reporting an issue please follow the bug reporting template.  We don't know what OS and version of O you are using.

bc38, 2023-09-05:
> Can you also post the opencpn.log file associated with the crash. I tried to reproduce on Windows but could not. Also, it might help to post opencpn.conf as well.
> 
> When reporting an issue please follow the bug reporting template. We don't know what OS and version of O you are using.

OS Win11 x64
Video nvidia geforce GTX 1070ti, last driver
CPN 5.8.4 / 5.9

Log and Config
[opencpn.zip](https://github.com/OpenCPN/OpenCPN/files/12523076/opencpn.zip)

nohal, 2024-02-17:
So far not reproducible for me on Linux and macOS.

transmitterdan, 2024-02-17:
Could not reproduce on Windows.

Thinking about what might be happening...

Suppose an ENC chart is going through the initial SENC conversion because scrolling across a large area triggers the SENC process.  Then if one were to close O maybe it does not properly complete the SENC process or leaves some cruft.  Then when restarting O and jumping to the ship's location O encounters the cruft left by incomplete SENC processing.  Would be hard to reproduce this.

bdbcat, 2024-02-28:
bc38...
Some recent changes relating to problems of this sort were made in OCPN 5.9.0, on github.
If possible, could you try to reproduce with current master version?

bc38, 2024-02-29:
> bc38... Some recent changes relating to problems of this sort were made in OCPN 5.9.0, on github. If possible, could you try to reproduce with current master version?

Dave, I'm already testing the new changes to the S-57 and it looks good and stable.
I checked by adding a new map catalog, deleted the SENC files and prepared them again, and rebuilt the entire map database.
There were no failures! Also, ENC settings were not reset to default. because There were no malfunctions during ENC preparation.
Thank you!

bdbcat, 2024-02-29:
bc38...
Thanks for the test and report.  Good news.
Please close this issue now.  We can re-open if the problem re-appears.

closed 2024-02-29 by bc38