OpenCPN/OpenCPN #3431: Dashboard Plugin - Day/Night Mode seems totally broken

TwoCanPlugIn, 2023-09-12:
I was intrigued by this [post](https://www.cruisersforum.com/forums/f121/windows-software-for-sailing-instruments-279565.html)

Not sure if this is a bug in OpenCPN, Dashboard Plugin, , wxWidgets or whatever.

On Windows, in the Dashboard Preferences dialog, under the Appearance tab, selecting the font colour has no effect. In fact the font colour setting is not persisted. On the other hand, the font selection does work and is persisted.  

On Linux, there seems to be no option to select the font colour (At least on my VM running Ubuntu 20.04 LTS)

Finally, the Dashboard Title Bar, does not follow the Windows Themes. Admittedly it used to be easier aeons ago to change the font, background for every Windows UI element (title bar, borders, scroll bars, window background etc.), but is still possible to at least change the title bar colour.

While the main Window of OpenCPN does follow the WIndows Title Bar theme, the Dashboard does not.

The expected behaviour should be that the Dashboard font colours are obeyed and persisted, and the title bar should follow the Title Bar theme. 

I would argue that the Day/Night mode and colour selection should be simplified, When in Night Mode, the background should be black or grey, and the font colours are suitably contrasted (eg. White, Red, Light Green or Light Blue). That is the approach that commercial instrument vendors have adopted.

[image]
[image]
[image]

rgleason, 2023-09-12:
Maybe this "feature" could be used for positive improvements to Day/night?  Is that the point?

TwoCanPlugIn, 2023-09-12:
I personally wouldn't describe fixing a bug as a "feature". In this case (at least on Windows), changing the Dashboard Font colours seem not to work. 

A bug is something that doesn't work as intended. No different to [this bug](https://github.com/OpenCPN/OpenCPN/issues/3343).

mpuz, 2023-10-17:
Ubuntu 22 dark theme - UI is totally broken with no font color cotrol
[image]

nohal, 2024-02-17:
Colors for fonts are now persisted on Windows, which is the only platform where they are configurable by the system font selection dialog.
There is nothing I am aware of we can do about the color of the title bars for the dockable AUI frames, that is managed by the wxAUIManager and the visual distinction is a feature as I understand it.
The color scheme of the dashboard is now also pretty much fully configurable and seems to me diming quite fine.

That is about as much as we can do about the bug reported. Further improvements and behavioural changes to the dashboard need to be handles as separate feature request issues or via a pull request.