OpenCPN/OpenCPN #3435: Reset Chart settings when adding a new map directory

bc38, 2023-09-19:
When adding a new map catalog to an existing Chart database, the Chart settings are reset to default.
Reproduced on #3422

CPN 5.8.4
Win11

nohal, 2024-02-17:
I can't reproduce this, can you describe the exact steps?

bc38, 2024-02-17:
> I can't reproduce this, can you describe the exact steps?

Video
https://www.dropbox.com/scl/fi/pb6txz3p98gjpn6s4joqq/2023-09-04-140552.mp4?rlkey=dg89akrjfk829m61i1ia04p29&dl=0

nohal, 2024-02-17:
That seems to show #3422 and I suppose that this issue is related to the very end of it, but I honestly still don't understand what exactly "adding a new map catalog to an existing Chart database" is here and how to reproduce this issue.

bc38, 2024-02-17:
The essence of the problem is that when you add a new map catalog to an existing one. ENC display settings are reset to default.
Tomorrow I'll check it on the latest CPN build and on new video hardware

Hakansv, 2024-02-17:
I've also sometimes during past years seen some "Vector chart Display" settings being reset to default.
But I've never been able to reproduce it in a traceable way. This may be a hint?
bc38: I also don't in details understand "_add a new map catalog to an existing one_"
Could you describe this in step by step with some path examples and what's in the "existing one" before.

closed 2024-03-11 by bc38