OpenCPN/OpenCPN #3436: O 5.9... TTM (AIS) target "rollover" causes OCPN to hang -"no responce"

Hakansv, 2023-09-19:
**Describe the bug**
A mouse rollover action on a TTM target, in order to show the "AIS info block" or right click menu, creates O to hang and not responding.
Using O 5.8.4 the rollover action is working as expected. See pict 1
New or old versions of radar_pi to create a TTM target makes no difference.

**To Reproduce**
1. Install radar plugin if not already present. 
2. Use radar type "Emulator"
3. Be sure to receive nav-data like position, heading. (And AIS targets?)
4. Right click, select: "Show radar" and click "Start/Stop radar" to get status "Transmitting"
5. Zoom until the emulated targets form a circle of red squares.
6.  In the neighborhood of one radar target right click and select "Acquire radar target"
7. Wait until the TTM MARPA target, a green circle, is present and stable. 
8. Use the mouse to rollover the target in order to show the AIS info or right click menu.
9. O is hanged.

**Expected behavior**
On rollover a info block as of picture 1 or a right click menu.

**Desktop (please complete the following information if applicable):**
 - OS: Win10 and Rpi Debian Bullseye.
 - OCPN version 5.9.. from present Git master branch.
 - Radar plugin version 5.5.0 or from present radar_pi Git master is the same

**More**
I've tried to run Debug but no clue or break. It just hangs.

**Screenshot**
Pict 1:
[image]

Hakansv, 2023-09-19:
While debug I found a endless loop here:
ais_target_data.cpp : 1023
```
for (int i = 0; s_mmsi.length() - 3; i++) {
      nMID = wxAtoi(s_mmsi.Mid(i, 3));
      if (IsValidMID(nMID)) break;
    }
```
The "s_mmsi.length() - 3" should have stopped at "7" since size() is 9. (I tried .size() but no difference)
But it counts "i" up forever???
Some c++ I don't understand, obviously. Hmm

bdbcat, 2023-09-19:
This is bad code.
Try this:

    for (int i = 0;  i < s_mmsi.length() - 3; i++) {
          nMID = wxAtoi(s_mmsi.Mid(i, 3));
          if (IsValidMID(nMID)) break;
        }

bdbcat, 2023-09-19:
Has this code snip changed from O584 to master?

Hakansv, 2023-09-19:
I < s_mmsi.length() is of course correct. I was blind for the typo. Thanks.
Yes it's changed. It's meant to sort out misused MMSI for e.g. fishing buoys but still detect a SAR.
PR to come.

closed 2023-09-19 by Hakansv