OpenCPN/OpenCPN #3437: SIGSEGV at RenderGLLs

lipixx, 2023-09-20:
**Describe the bug**
When zooming in the map having CM93 vector maps installed, OpenCPN is sluggish and finally segfaults.

**To Reproduce**
Steps to reproduce the behavior:
1. Start OpenCPN
2. Zoom in.

**Expected behavior**
Not to crash.

**Screenshots**
(gdb) bt
#0  0x00000000400b0b5a in  ()
#1  0x00007fffba48e4c0 in  () at /lib64/libnvidia-glcore.so.535.54.03
#2  0x00007fffba0f3042 in  () at /lib64/libnvidia-glcore.so.535.54.03
#3  0x0000555555dd26c1 in s52plib::RenderGLLS(_ObjRazRules*, _Rules*, ViewPort*) [clone .part.0] [clone .isra.0] ()
#4  0x0000555555de1008 in s52plib::DoRenderObject(wxDC*, _ObjRazRules*, ViewPort*) [clone .part.0] [clone .isra.0] ()
#5  0x0000555555d98229 in s57chart::DoRenderRegionViewOnGL(wxGLContext const&, ViewPort const&, OCPNRegion const&, LLRegion const&, bool) [clone .constprop.0] ()
#6  0x0000555555c8d360 in cm93compchart::DoRenderRegionViewOnGL(wxGLContext const&, ViewPort const&, OCPNRegion const&, LLRegion const&) ()
#7  0x0000555555bc5b1c in glChartCanvas::RenderQuiltViewGL(ViewPort&, OCPNRegion const&) ()
#8  0x0000555555bc60ca in glChartCanvas::RenderCharts(ocpnDC&, OCPNRegion const&) ()
#9  0x0000555555bc8065 in glChartCanvas::Render() ()
#10 0x0000555555bb910e in glChartCanvas::OnPaint(wxPaintEvent&) ()
#11 0x00007ffff6f8fc5e in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) (entry=<optimized out>, handler=handler@entry=0x5555564ede70, event=...)
    at ../src/common/event.cpp:1390
#12 0x00007ffff6f91b73 in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) (this=<optimized out>, event=..., self=self@entry=0x5555564ede70) at ../src/common/event.cpp:996
#13 0x00007ffff6f91e69 in wxEvtHandler::TryHereOnly(wxEvent&) (this=this@entry=0x5555564ede70, event=...) at ../src/common/event.cpp:1587
#14 0x00007ffff6f91eff in wxEvtHandler::TryBeforeAndHere(wxEvent&) (event=..., this=0x5555564ede70) at ../include/wx/event.h:3692
#15 wxEvtHandler::ProcessEventLocally(wxEvent&) (this=0x5555564ede70, event=...) at ../src/common/event.cpp:1520
#16 0x00007ffff6f91fe2 in wxEvtHandler::ProcessEvent(wxEvent&) (this=0x5555564ede70, event=...) at ../src/common/event.cpp:1493
#17 0x00007ffff6f8feb3 in wxEvtHandler::SafelyProcessEvent(wxEvent&) (this=<optimized out>, event=<optimized out>) at ../src/common/event.cpp:1611
#18 0x00007ffff74cb4b3 in wxWindow::GTKSendPaintEvents(_cairo*) (this=this@entry=0x5555564ede70, cr=0x5555569a5920) at ../src/gtk/window.cpp:4289
#19 0x00007ffff7caf96c in wxGLCanvas::OnInternalIdle() (this=0x5555564ede70) at ../src/gtk/glcanvas.cpp:316
#20 0x00007ffff76349c0 in wxWindowBase::SendIdleEvents(wxIdleEvent&) (this=0x5555564ede70, event=...) at ../src/common/wincmn.cpp:2821
#21 0x00007ffff76349f8 in wxWindowBase::SendIdleEvents(wxIdleEvent&) (this=<optimized out>, event=...) at ../src/common/wincmn.cpp:2837
#22 0x00007ffff76349f8 in wxWindowBase::SendIdleEvents(wxIdleEvent&) (this=<optimized out>, event=...) at ../src/common/wincmn.cpp:2837
#23 0x00007ffff750144a in wxFrame::SendIdleEvents(wxIdleEvent&) (this=0x55555648fb10, event=...) at ../src/gtk/frame.cpp:182
#24 0x00007ffff753b885 in wxAppBase::ProcessIdle() (this=<optimized out>) at ../src/common/appcmn.cpp:375
#25 0x00007ffff748b4e8 in wxApp::DoIdle() (this=0x555556142c90) at ../src/gtk/app.cpp:159
#26 0x00007ffff748b637 in wxapp_idle_callback(gpointer) () at ../src/gtk/app.cpp:107
#27 0x00007ffff521439d in g_idle_dispatch (source=0x55556f826cf0, callback=0x7ffff748b620 <wxapp_idle_callback(gpointer)>, user_data=0x0) at ../glib/gmain.c:6163
#28 0x00007ffff521839c in g_main_dispatch (context=0x555556173ad0) at ../glib/gmain.c:3460
#29 g_main_context_dispatch (context=0x555556173ad0) at ../glib/gmain.c:4200
#30 0x00007ffff5276438 in g_main_context_iterate.isra.0 (context=0x555556173ad0, block=1, dispatch=1, self=<optimized out>) at ../glib/gmain.c:4276
#31 0x00007ffff521799f in g_main_loop_run (loop=0x555556398410) at ../glib/gmain.c:4479
#32 0x00007ffff5606975 in gtk_main () at ../gtk/gtkmain.c:1329
#33 0x00007ffff74a11f5 in wxGUIEventLoop::DoRun() (this=0x555556b5fae0) at ../src/gtk/evtloop.cpp:65
#34 0x00007ffff6eb9c7a in wxEventLoopBase::Run() (this=0x555556b5fae0) at ../src/common/evtloopcmn.cpp:78
#35 0x00007ffff6e9551d in wxAppConsoleBase::MainLoop() (this=0x555556142c90) at ../src/common/appbase.cpp:334
#36 0x00007ffff6eec832 in wxEntry(int&, wchar_t**) (argc=<optimized out>, argv=<optimized out>) at ../src/common/init.cpp:506
#37 0x0000555555788e1c in main ()

**Desktop (please complete the following information if applicable):**
 - OS:
 
Linux llit 6.3.8-200.fc38.x86_64 #1 SMP PREEMPT_DYNAMIC Thu Jun 15 02:15:40 UTC 2023 x86_64 GNU/Linux
Fedora release 38 (Thirty Eight)

]# nvidia-smi
Wed Sep 20 18:51:32 2023       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.54.03              Driver Version: 535.54.03    CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce GT 1030         Off | 00000000:07:00.0  On |                  N/A |
| 35%   41C    P0              N/A /  30W |    749MiB /  2048MiB |      4%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         

**Additional context**
It may be an Nvidia issue. Recent driver versions broke compatibility with previous versions and maybe OpenCPN is compiled against older nvml.h versions? May not be the case.

leamas, 2023-10-24:
hm... Have you tried using the nouveau driver instead? That should show if it's about nvidia's driver.

nohal, 2024-02-25:
@lipixx can you please try to enable OpenGL and then start OpenCPN from the terminal with `mesa_glthread=false opencpn`?