OpenCPN/OpenCPN #3438: Bug: Plugin Version Numbering (small bug)

rgleason, 2023-09-22:
Please see the report in Watchdog

https://github.com/rgleason/watchdog_pi/issues/70#issuecomment-1731215269

leamas, 2023-09-22:
This seems to be a bug in the plugin catalog eventually caused by a faulty PR to opencpn/plugins. Core OpenCPN does nothing wrong here, it just reflects the version as defined in the catalog.

In other words, this should probably just be closed as NOTABUG.

One a sidenote getting  the verssion from the catalog is sort of a hack made necessary since we did not have complete version data available in the binary plugin API. From API  1.18 there are methods in the plugin API which returns complete  version data, obsoleting the version in the catalog. That should make things more robust in the long run.

rgleason, 2023-09-22:
NOTABUG   We need to move the plugin API up to 118, but what are the pitfalls of doing that?

closed 2023-09-22 by rgleason