OpenCPN/OpenCPN #3440: How to build master branch code for Android  APK

Quentin-xue, 2023-09-27:
Code ： opencpn master branch
Hardware: RK3568 which supports buildroot and Android 12
1：Buildroot 
      compile successfully  and run stably ,only thing neet to imporve is enable GPU accelerate and now using mesa3d-21.1.8 which is software OpenGL functions,my colleagues work on this
[image]
[image]

2：Android 12 
I work weeks on How to build an Android APK but still don't understand steps  clearly,Anyone can provide tips or commands or steps to compile it?  thanks in advance.

3: Some questions
A：Which wxWidgets version do we need (above 3.2)? I find /opencpn/OpenCPN/cache/OCPNAndroidCoreBuildSupport/wxWidgets/include/wx$ cat version.h |grep wxVERSION_STRING   
#define wxVERSION_STRING   wxT("wxWidgets 3.1.0")

B：Anything is not correct in my compile command 
`COMPDIR="/home/corey/opencpn/andorid/ndk/21.3.6528147"
cmake  \
  -D_wx_selected_config=androideabi-qt-arm64 \
  -DwxQt_Build=build_android_release_64_static_O3 \
  -DQt_Build=build_arm64/qtbase \
  -DCMAKE_AR=$COMPDIR/toolchains/llvm/prebuilt/linux-x86_64/bin/aarch64-linux-android-ar \
  -DCMAKE_CXX_COMPILER=$COMPDIR/toolchains/llvm/prebuilt/linux-x86_64/bin/aarch64-linux-android21-clang++ \
  -DCMAKE_C_COMPILER=$COMPDIR/toolchains/llvm/prebuilt/linux-x86_64/bin/aarch64-linux-android21-clang \
  -DOCPN_Android_Common=OCPNAndroidCommon-master \
  -DCMAKE_TOOLCHAIN_FILE=../buildandroid/build_android.cmake \
  -DOCPN_TARGET_TUPLE="Android-arm64" \
  -Dtool_base="/home/corey/opencpn/andorid/ndk/21.3.6528147/toolchains/llvm/prebuilt/linux-x86_64" \
  -DCMAKE_INSTALL_PREFIX=/  ..`

C：We need all the depend libs like：HarfBuzz,pango,BZip2,lz4 right？those from CMakeLists.txt

Quentin-xue, 2023-10-18:
[image]
[image]
[image]

Quentin-xue, 2023-10-18:
[image]

Quentin-xue, 2023-10-19:
Now I think it is very easy to build Android APK，but I will continue to check whether everything is running correctly

Quentin-xue, 2023-10-19:
thanks all the contributors

closed 2023-10-19 by Quentin-xue

nohal, 2023-10-19:
Did you need any patches to make it build in your environment?

Quentin-xue, 2023-10-19:
@nohal NO

My steps as following if it is not correct please correct it 

1: Download OCPNAndroidCoreBuildSupport.zip and makesure _master_base is set correctly  please  check file 
`OpenCPN/libs$ cat AndroidLibs.cmake `

2: Compile opencpn ，the libgorp.so we need will be created

```
ARM32
cmake \
  -DCMAKE_BUILD_TYPE=Release \
  -DOCPN_TARGET_TUPLE:STRING="Android-armhf;16;armhf" \
  -Dtool_base="/home/corey/opencpn/andorid/ndk/21.3.6528147/toolchains/llvm/prebuilt/linux-x86_64"\
  -DCMAKE_INSTALL_PREFIX=./install   \
  ..

ARM64
cmake \
  -DCMAKE_BUILD_TYPE=Release \
  -DOCPN_TARGET_TUPLE:STRING="Android-arm64;16;arm64" \
  -Dtool_base="/home/corey/opencpn/andorid/ndk/21.3.6528147/toolchains/llvm/prebuilt/linux-x86_64"\
  -DCMAKE_INSTALL_PREFIX=./install   \
  ..
```
3：Download OpenCPN-Android from 
` https://github.com/bdbcat/OpenCPN-Android`
modify  file ,change it to  your OCPNAndroidCoreBuildSupport location
```
 OpenCPN-Android$ git diff app/build.gradle
        //def Qt_Base = "/home/dsr/Projects/qt5"
    def Qt_Base = "/home/suiyuan626/share/openCPN/Android/OCPNAndroidCoreBuildSupport/qt5"

    //def Qt_Build = "build_arm32_debug_19"
    def Qt_Build = "build_arm32_19_O3"

    def Qt_Build_ARM64 = "build_arm64_O3"
    def Qt_Build_x86 = "build_x86"

    def OCPN_Base = "/home/suiyuan626/share/openCPN/Android/OpenCPN"

    //def OCPN_Build = "build_android_32"
    def OCPN_Build = "build32bit"

    //def OCPN_Build_64 = "build_android_64"
    def OCPN_Build_64 = "build64bit"

    def OCPN_Build_x86 = "build_android_x86_O3"
```

4: Configure QT creator, create a helloword example to confirm all the setting is correct and can generate APK
```
suiyuan626@ubuntu:~/share/qt/example/build-First_hello-Android_Qt_6_5_3_Clang_arm64_v8a-Debug$ ll
total 1756
drwxrwxr-x  9 suiyuan626 suiyuan626    4096 Oct 17 14:31 ./
drwxrwxr-x 20 suiyuan626 suiyuan626    4096 Oct 18 09:09 ../
drwxrwxr-x  7 suiyuan626 suiyuan626    4096 Oct 17 14:28 android-build/
-rw-rw-r--  1 suiyuan626 suiyuan626    1296 Oct 17 08:35 android-First_hello-deployment-settings.json
-rw-rw-r--  1 suiyuan626 suiyuan626   47050 Oct 17 14:31 build.ninja
drwxrwxr-x  3 suiyuan626 suiyuan626    4096 Oct 17 08:35 .cmake/
-rw-rw-r--  1 suiyuan626 suiyuan626   56357 Oct 17 14:28 CMakeCache.txt
-rw-rw-r--  1 suiyuan626 suiyuan626   56357 Oct 17 14:31 CMakeCache.txt.prev
drwxrwxr-x  8 suiyuan626 suiyuan626    4096 Oct 17 14:31 CMakeFiles/
-rw-rw-r--  1 suiyuan626 suiyuan626    2839 Oct 17 08:35 cmake_install.cmake
drwxrwxr-x  4 suiyuan626 suiyuan626    4096 Oct 17 14:28 First_hello_autogen/
-rwxrwxr-x  1 suiyuan626 suiyuan626 1539312 Oct 17 14:28 libFirst_hello_arm64-v8a.so*
-rw-rw-r--  1 suiyuan626 suiyuan626   39812 Oct 17 14:31 .ninja_deps
-rw-rw-r--  1 suiyuan626 suiyuan626    1183 Oct 17 14:31 .ninja_log
drwxrwxr-x  2 suiyuan626 suiyuan626    4096 Oct 17 14:31 .qt/
[... 55 lines cut ...]
-rw-r--r-- 1 suiyuan626 suiyuan626  6360 Oct 18 19:20 AndroidManifest.xml
drwxrwxr-x 5 suiyuan626 suiyuan626  4096 Oct 18 19:20 app/
-rwxrwxrwx 1 suiyuan626 suiyuan626  1135 Oct 18 19:20 build.gradle*
drwxrwxr-x 7 suiyuan626 suiyuan626  4096 Oct 18 19:21 .gradle/
drwxrwxr-x 3 suiyuan626 suiyuan626  4096 Oct 18 19:20 gradle/
-rw-rw-r-- 1 suiyuan626 suiyuan626   969 Oct 18 19:20 gradle.properties
-rw-rw-r-- 1 suiyuan626 suiyuan626   784 Oct 18 19:20 gradle.properties~
-rwxrwxr-x 1 suiyuan626 suiyuan626  4971 Oct 18 19:20 gradlew*
-rw-rw-r-- 1 suiyuan626 suiyuan626  2404 Oct 18 19:20 gradlew.bat
drwxrwxr-x 6 suiyuan626 suiyuan626  4096 Oct 18 19:20 libs/
-rw-rw-r-- 1 suiyuan626 suiyuan626 18092 Oct 18 19:20 LICENSE
drwxrwxr-x 4 suiyuan626 suiyuan626  4096 Oct 18 19:20 lobsterpicker/
-rw-rw-r-- 1 suiyuan626 suiyuan626    91 Oct 18 19:20 local.properties
drwxrwxr-x 3 suiyuan626 suiyuan626  4096 Oct 18 19:20 res/
-rw-rw-r-- 1 suiyuan626 suiyuan626   203 Oct 18 19:20 settings.gradle

suiyuan626@ubuntu:~/share/openCPN/Android/OpenCPN/build-opencpn.simple-Android_for_armeabi_v7a_arm64_v8a_x86_x86_64_Clang_Qt_5_14_2_for_Android-Debug/android-build/app/build/outputs/apk/debug$ ll
total 61604
drwxrwxr-x 2 suiyuan626 suiyuan626     4096 Oct 18 19:21  ./
drwxrwxr-x 3 suiyuan626 suiyuan626     4096 Oct 18 19:21  ../
-rw-rw-r-- 1 suiyuan626 suiyuan626 63069374 Oct 18 19:21 'org.opencpn.opencpn-v104(5.8.4)-debug.apk'
-rw-rw-r-- 1 suiyuan626 suiyuan626      352 Oct 18 19:21  output-metadata.json
suiyuan626@ubuntu:~/share/openCPN/Android/OpenCPN/build-opencpn.simple-Android_for_armeabi_v7a_arm64_v8a_x86_x86_64_Clang_Qt_5_14_2_for_Android-Debug/android-build/app/build/outputs/apk/debug$

```

bdbcat, 2023-10-19:
This looks and sounds about right.
In production we use Android Studio instead of QTCreator.  This allows us a tighter integration with Playstore tools and workflow.
Good Job!